ArduCAM/Arduino #1: ArduCam Mini not working

deltaGPhys, 2015-06-19:
There seems to be less documentation for the mini board than the others; I can't get it to work (OV 5642). I get just garbage back to the serial monitor. Also, since there's no SD port on the mini shield, how will I save pictures?

Any help is appreciated!

Thanks,
jg

ArduCAM, 2015-06-19:
Please find more documents from link: http://www.arducam.com/arducam-mini-released/ 
You can connect extra SD card reader or transfer the captured image to host app like this: www.arducam.com/downloads/app/ArduCAM_v1.0.exe
Regards
Lee

ArduCAM, 2015-06-20:
Dear Sir,
Thanks for your question.

Please read the software application note for detail:
http://www.arducam.com/downloads/shields/ArduCAM_Camera_Shield_Software_Application_Note.pdf 

Please try this sketch to test the ArduCAM mini  5MP  https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_OV5642_Video_Streaming 
And use the host application console from, and save pictures on your PC.
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_host_v1.0.exe 

Any question else,please feel free to let us know.
Best Regards,
Lee 
From: deltaGPhys
Date: 2015-06-19 22:57
To: ArduCAM/Arduino
Subject: [Arduino] ArduCam Mini not working (#1)
There seems to be less documentation for the mini board than the others; I can't get it to work (OV 5642). I get just garbage back to the serial monitor. Also, since there's no SD port on the mini shield, how will I save pictures?
Any help is appreciated!
Thanks,
jg

deltaGPhys, 2015-06-21:
Thanks for the replies - I'll dig into this today and let you know!

deltaGPhys, 2015-06-21:
I have installed the program and am getting an SPI interface error when I try to open the port on the software. Judging by the pinouts, it seems like I should just plug the Mini shield into the Arduino, digital pins 10-13,gnd, etc. - all of the way to the end of that section. The shield that I got is set up differently from the one in the picture (it's a continuous row of 8 pins, so it doesn't look like there will be any jumpering over to the ICSP section. Let me know if this isn't set up correctly.

Thanks for the help,
jg

ArduCAM, 2015-06-21:
Sorry there is a mistake in the demo code, it use pin9 as cs instead of pin10
// set pin 10 as the slave select for the digital pot:
const int CS = 9;
Would you please let me know which type of arduino board you are using, different arduino boards has different SPI pin out, it is more reliable to use ICSP pins because it is the same SPI port between arduino boards

deltaGPhys, 2015-06-22:
It's an UNO R3. I've changed the pin in the sketch and reuploaded, but I'm still getting an SPI error from the application. It's hard to see the errors, BTW< because the windows application has a fixed height, which is higher than my screen. I'm using a tablet PC, so I rotate the screen to see it :)

deltaGPhys, 2015-06-22:
Sometimes, after I open the port, I get "ArduCam start!", no error, but no pics. I see the RX LED lighting on the Arduino when I press capture, but it never transmits.

deltaGPhys, 2015-06-25:
Any ideas on this? I'm going to have to return the camera if I can't get it to work - it's not doing anything for me at this point.

ArduCAM, 2015-06-25:
Dear Sir,
It seems that there is some thing wrong with the wiring.
Would you please take a photo to show me how you do the wiring.
You have to make the I2C and SPI working correctly before taking any pictures.
 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: deltaGPhys
Date: 2015-06-25 22:57
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
Any ideas on this? I'm going to have to return the camera if I can't get it to work - it's not doing anything for me at this point.

deltaGPhys, 2015-06-26:
Interesting news: Wiring it like this:
[image]
...does not work, though all of the pins are correctly placed, I think.

Wiring it like this:
[image]
[image]
[image]
...does. I'm using the ICSP pins for the SPI pins, but the same pin 10 and SCL/SDA pins. 

Shouldn't those be exactly equivalent setups?

I'm glad that I'm getting something out of it, though! Is there a pre-made sketch for sending photos to an SD card (such as OpenLog?).

Thanks,
jg

deltaGPhys, 2015-06-26:
Aha: all the pins are exactly correct... except for the AREF, which should have the +5V pin there.

deltaGPhys, 2015-06-26:
Now that the camera works fine attached to the computer, my unsolved question is how to save those photos to an SD card (OpenLog is the card breakout that I have). I assume that I'll be able to figure out having the sketch take photos with large delays between, though if there's a good power-saving way to just take one photo a minute or so, I'm all ears.

ArduCAM, 2015-06-29:
Hi again,
Yes, you are right, and finally you resolved the problem by yourself. You are genius!
The Arduino board VREF is not 5V VCC you have to use external power.
 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: deltaGPhys
Date: 2015-06-27 07:06
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
Now that the camera works fine attached to the computer, my unsolved question is how to save those photos to an SD card (OpenLog is the card breakout that I have). I assume that I'll be able to figure out having the sketch take photos with large delays between, though if there's a good power-saving way to just take one photo a minute or so, I'm all ears.

deltaGPhys, 2015-06-29:
My remaining issue is how to store images to OpenLog or similar SD card shield/breakout. Is there example code for that?

Thanks,
jg

-- 
Joshua Gates 
Physics Faculty 
Tatnall School (Wilmington DE) 
JHU Center for Talented Youth

ArduCAM, 2015-06-29:
please have a look at this example:
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/REVC/ArduCAM_OV2640_Digital_Camera/ArduCAM_OV2640_Digital_Camera.ino 
SD card cs pin should go Pin9
Best Regards,
Lee
 
-----------------------------------------------------
 
From: deltaGPhys
Date: 2015-06-29 23:08
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
My remaining issue is how to store images to OpenLog or similar SD card shield/breakout. Is there example code for that?

Thanks,
jg

deltaGPhys, 2015-06-29:
I'll check it out - thanks!

jg

-- 
Joshua Gates 
Physics Faculty 
Tatnall School (Wilmington DE) 
JHU Center for Talented Youth

 https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/REVC/ArduCAM_OV2640_Digital_Camera/ArduCAM_OV2640_Digital_Camera.ino

geolog125, 2015-07-07:
Dear Lee, hello!
I have encountered same problem! Arducam mini occupies same pins on Arduino Uno, so no possibilities to save any pictures to SD card.
No pins to connect SD card besides Arducam mini.
SD card data logging shield has a conflict with Arducam mini.

In Internet and both you web site I didn't find any solution.
Please, give us CLEAR answer of how to connect both Arducam mini and  SD card module.

I maybe order for portion of these Arducam modules, if It can be resolved.

ArduCAM, 2015-07-07:
Dear Sir,
Please connect other free pin to ArduCAM mini CS pin and change the arduino example sketch accordingly.
Would you please send me a wiring diagram you currently do.
 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: geolog125
Date: 2015-07-07 08:13
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
Dear Lee, hello!
I have encountered same problem! Arducam mini occupies same pins on Arduino Uno, so no possibilities to save any pictures to SD card.
No pins to connect SD card besides Arducam mini.
SD card data logging shield has a conflict with Arducam mini.
In Internet and both you web site I didn't find any solution.
Please, give us CLEAR answer of how to connect both Arducam mini and SD card module.
I maybe order for portion of these Arducam modules, if It can be resolved.

geolog125, 2015-07-07:
Dear Lee, thanks for an answer. 
I connect Arducam mini to Arduino by this way.
[image]
It gives me opportunity to read data from PC i appropriate software.
But I need to save data to SD card.
[image]

But pins 11,12,13 are occupied. How to connect in that case?

geolog125, 2015-07-07:
Also, I can connect my microsd card via ICSP and all goes well.
But when I tried to add ArducamMini to that connection, I got "SPI interface error" message in Arducam sketch.
What to do?
[image]

ArduCAM, 2015-07-07:
Dear Sir,
The two Chipselect signal from ArduCAM mini and SD card reader should be only one active at a given time.
Can you send me your test code?
 
Best Regards,
Lee
 
-----------------------------------------------------
 
From: geolog125
Date: 2015-07-07 10:19
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
Also, I can connect my microsd card via ICSP and all goes well.
But when I tried to add ArducamMini to that connection, I got "SPI interface error" message in Arducam sketch.
What to do?

geolog125, 2015-07-07:
Dear friend, when I updated Arducam classes from your github,
I tried these connection:

[image]

But, alas, your sample brought a random errors like: "Arducam is not detected", "SD card error", and all spectrum of mistakes. One moment I even captured a picture with mess data.
 After all, I got code here: http://forum.arduino.cc/index.php?topic=331067.msg2298891#msg2298891

It worked for me!
But image quality is following (that's little keyboard lamp):

[image]

How to make quality better of manage it?

ArduCAM, 2015-07-07:
Hi again,
To fix the RGB color issue, please read the post here: http://www.arducam.com/rgb565-format-issues/Best Regards,
Lee
 
-----------------------------------------------------
 
From: geolog125
Date: 2015-07-07 11:13
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
Dear friend, when I updated Arducam classes from your github,
I tried these connection:But, alas, your sample brought a random errors like: "Arducam is not detected", "SD card error", and all spectrum of mistakes. One moment I even captured a picture with mess data.
After all, I got code here: http://forum.arduino.cc/index.php?topic=331067.msg2298891#msg2298891
It worked for me!
But image quality is following (that's little keyboard lamp):
How to make quality better of manage it?

geolog125, 2015-07-07:
I 'll try. 
Why your code for digital cam (https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/REVC/ArduCAM_OV2640_Digital_Camera/ArduCAM_OV2640_Digital_Camera.ino) is not running with my scheme, generates an errors?

geolog125, 2015-07-07:
After applying these http://www.arducam.com/rgb565-format-issues/
I obtain stranger picture:
[image]

Any solutions?

ArduCAM, 2015-07-07:
Things become more and more complex.
Please try to read the document here: http://www.arducam.com/downloads/shields/ArduCAM_Camera_Shield_Software_Application_Note.pdf 
And use the host application from: https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_host_v1.0.exe 
Make sure if you use the example sketches in mini folder https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_OV2640_Video_Streaming 
Before you changing any code by yourself, please try the steps above.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: geolog125
Date: 2015-07-07 11:23
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
I 'll try. 
Why your code for digital cam (https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/REVC/ArduCAM_OV2640_Digital_Camera/ArduCAM_OV2640_Digital_Camera.ino) is not running with my scheme, generates an errors?

geolog125, 2015-07-07:
Dear Lee, thanks for answer.
But maybe there are any solutions for SD card image write for Arducam mini?
I did image capture with your PC software it is OK, but I need just save images into SD card with Arducam mini. 
In your examples Arducam mini can only be used as a  PC camera. 
In my projects **I need autonomous capture  by Arduino.** The samples you have shown did not coincide, because it is PC image capture.
What I have to fix to obtain normal image?
[image]
[image]

```javascript
//most promising, i think. saves nonsense data, though.

//http://forum.arduino.cc/index.php?topic=285778.0
// inspired by http://sauerwine.blogspot.fr/2013/07/an-arduino-time-lapse-camera-using.html
// and http://www.arducam.com/how-arducam-use-a-external-trigger-from-a-sensor/

#include <UTFT_SPI.h>
#include <SD.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
#include <avr/pgmspace.h>
#include <ov2640_regs.h>

#define BMPIMAGEOFFSET 54
#define SD_CS 9
const int CS1 = 10;

// Constants that define the format of the picture taken
const int bmp_mode = 0;
const int mode = bmp_mode;

const char bmp_header[54] PROGMEM =
{
      0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x36, 0x00, 0x00, 0x00, 0x28, 0x00,
      0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
// Instanciation of the class ArduCAM
ArduCAM myCAM1(OV2640, CS1);
static int k = 0;

// Used in takePicture() to transfer the buffer of the ArduCAM into a BMP file
// inspired by http://sauerwine.blogspot.fr/2013/07/an-arduino-time-lapse-camera-using.html
void writeBMP(File outFile){
 
  //char VH, VL;
  char VH, VL, buf[256];
  uint8_t temp,temp_last;
  int i, j, posn, nextNum;
 
 /*
    //Write the BMP header
  for( i = 0; i < 54; i++)
  {
    char ch = pgm_read_byte(&bmp_header[i]);
    outFile.write((uint8_t*)&ch,1);
  }

  //Read the first dummy byte from FIFO
  temp = myCAM1.read_fifo();
  //Read 320x240x2 byte from FIFO
  for(i = 0; i < 240; i++)
    for(j = 0; j < 320; j++)
    {
      VH = myCAM1.read_fifo();
      VL = myCAM1.read_fifo();
      //RGB565 to RGB555 Conversion
[... 149 lines cut ...]
  }
  else {
  Serial.println("SPI All Good");
  } 
 
  // change MCU mode (?)
  myCAM1.write_reg(ARDUCHIP_MODE, 0x00);
  myCAM1.InitCAM();
 
  // SD card initialisation
  if (!SD.begin(SD_CS))
  {
    while (1);		//If failed, stop here
    Serial.println("SD Card Error");
  }
  else
 {   Serial.println("SD Card detected!");
}}

void loop(){
		takePicture();
		delay(20000);
	
}
```

ArduCAM, 2015-07-07:
Dear Sir,
Thanks for your information.
I bought a SD card module exactly the same as yours, and found the same problem.
I have figured out that there is design fault on SD card module.
The SD moudule use a SN74LVC125A as a level shifter to buffer the 5V and 3.3V signals.
And I found that all the OE signals are connected to GND directly, so all the output pin Y are always driver the SPI signals like MISO, MOSI, SCLK.
There is no problem for MOSI and SCLK which are output signal from processor. But the MISO are shared by all SPI devices on the bus, it should be tri-state when the device is inactive.
But the SD card reader alway drive the MISO with a fixed logic '0' or '1' when it is inactive. For this kind of SD reader, it can only monopolize the SPI bus, it can not work with other SPI devices.
 
We have another product which is Rev.C arducam, which as build in SD card reader and LCD screen and the ArduCAM circuit like the ArduCAM mini.
So don't doubt there is such problem with ArduCAM shield, we have fully tested it.
http://www.arducam.com/arducam-shiled-rev-c-released/ 

Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: geolog125
Date: 2015-07-07 16:45
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
Dear Lee, thanks for answer.
But maybe there are any solutions for SD card image write for Arducam mini?
I did image capture with your PC software it is OK, but I need just save images into SD card with Arducam mini. 
In your examples Arducam mini can only be used as a PC camera. 
In my projects I need autonomous capture by Arduino. The samples you have shown did not coincide, because it is PC image capture.
What I have to fix to obtain normal image?

//most promising, i think. saves nonsense data, though.
//http://forum.arduino.cc/index.php?topic=285778.0
// inspired by http://sauerwine.blogspot.fr/2013/07/an-arduino-time-lapse-camera-using.html
// and http://www.arducam.com/how-arducam-use-a-external-trigger-from-a-sensor/
#include <UTFT_SPI.h>
#include <SD.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
#include <avr/pgmspace.h>
#include <ov2640_regs.h>
#define BMPIMAGEOFFSET 54
#define SD_CS 9
const int CS1 = 10;
// Constants that define the format of the picture taken
const int bmp_mode = 0;
const int mode = bmp_mode;
const char bmp_header[54] PROGMEM =
{
      0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x36, 0x00, 0x00, 0x00, 0x28, 0x00,
      0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
      0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
// Instanciation of the class ArduCAM
ArduCAM myCAM1(OV2640, CS1);
static int k = 0;
// Used in takePicture() to transfer the buffer of the ArduCAM into a BMP file
// inspired by http://sauerwine.blogspot.fr/2013/07/an-arduino-time-lapse-camera-using.html
void writeBMP(File outFile){
  //char VH, VL;
  char VH, VL, buf[256];
  uint8_t temp,temp_last;
  int i, j, posn, nextNum;
 /*
    //Write the BMP header
  for( i = 0; i < 54; i++)
  {
[... 138 lines cut ...]
  temp = myCAM1.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
  {
    Serial.println("SPI interface Error!");
    while(1);
  }
  else {
  Serial.println("SPI All Good");
  } 
  // change MCU mode (?)
  myCAM1.write_reg(ARDUCHIP_MODE, 0x00);
  myCAM1.InitCAM();
  // SD card initialisation
  if (!SD.begin(SD_CS))
  {
    while (1);      //If failed, stop here
    Serial.println("SD Card Error");
  }
  else
 {   Serial.println("SD Card detected!");
}}
void loop(){
        takePicture();
        delay(20000);
}

geolog125, 2015-07-07:
Dear Lee, hi again!
I got another SD card module of another customer (not micro SD). 
So, the is been changed slightly, and green fields disappeared.
But there is another problems not related to SD card's conflict.
1) Magenta and blue spots on light areas. Can they be removed?;
[image]

2) The only resolution is 320x240. How to change (enlarge) it?
I need distinct Arducam mini because of its size and form factor.
I need just little camera guided by Arduino.

ArduCAM, 2015-07-08:
Dear Sir,
I'm glad to hear that you are making progress.
But I am on a business trip these days and will be back at the weekend.
So I would like you to test the SD card saving function with the following examples, although they are designed for Rev.C arducam, but it can work for ArduCAM mini board:
This example is for BMP file saving, 320x240 bmp image supported only,
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/REVC/ArduCAM_OV2640_Camera_Playback 
This example is for JPEG file saving, and save 1600x1200 JPEG images,
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/REVC/ArduCAM_OV2640_Digital_Camera 

 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: geolog125
Date: 2015-07-08 07:50
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
Dear Lee, hi again!
I got another SD card module of another customer (not micro SD). 
So, the is been changed slightly, and green fields disappeared.
But there is another problems not related to SD card's conflict.
1) Magenta and blue spots on light areas. Can they be removed?;
2) The only resolution is 320x240. How to change (enlarge) it?
I need distinct Arducam mini because of its size and form factor.
I need just little camera guided by Arduino.

geolog125, 2015-07-08:
Dear Lee, thank you very much!
After the modification of code you supplied above all started to work as need!
The reason of malfunction with my prior attempts to use it was bad microsd reader.
With this reader everything is OK.
[image]
Thank you again for your consult!

ArduCAM, 2015-07-10:
Dear Sir,
Thanks for your news.
Would you please describe how do you resolve your problems in the forum, it will help others to use ArduCAM more easily. 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: geolog125
Date: 2015-07-08 14:15
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCam Mini not working (#1)
Dear Lee, thank you very much!
After the modification of code you supplied above all started to work as need!
The reason of malfunction with my prior attempts to use it was bad microsd reader.
With this reader everything is OK.

Thank you again for your consult!

closed 2015-07-21 by ArduCAM

zeidooo, 2015-07-29:
Dear Lee,
i'm triing to interface the Arducam OV2640 with Beagle Bone Black and i'm using your code in Github but everytime it show me the same error "SPI interface ERROR!"
any information will be helpful for me 
Thank you

rei-vilo, 2015-08-07:
Same for me! `SPI interface ERROR!`

What are the exact settings for SPI, which mode?

shravanib, 2015-12-16:
Dear Sir,
I purchase arducam OV5642 last week.
Since the time I connected it, I keep on getting the "SPI Interface Error" evrytime I run the Host application for PC console.
I have not been able to see any image whatsoever.

I am new to arduino and thus wanted to know what does the error exactly mean? What is the actual problem? Is something wrong with the Arducam or the code or the arduino board itself (Uno R3).
I have tried all the connections and solutions you provided previously in the forum with no luck.

I would be highly obliged if you could please help me solve this problem.
Thank you.

capblack, 2016-02-01:
Dear Lee, I have the same problem  whit OV5642

SPI.begin(); 
64   //Check if the ArduCAM SPI bus is OK 
65   myCAM.write_reg(ARDUCHIP_TEST1, 0x55); 
66   temp = myCAM.read_reg(ARDUCHIP_TEST1); 
67   Serial.println(temp);        <<<<<<========== here temp return 0x85 "SPI Interface Error"
68   if (temp != 0x55)                                                                                        
69   { 
70     Serial.println("SPI interface Error!"); 
71     //while(1); 
72   } 

       with all my boards , any help ?
       Thank you.

ArduCAM, 2016-02-02:
Hi capblack, please try the camera on Arduino UNO R3 first before moving to other platform to make sure hte hardware is working.
BTW: which kind of board you are using right now?

capblack, 2016-02-02:
Dear Lee, thanks for you fast response , but yesterday started to work ok, right now I'm using a Mega 2560 and I would like to work with nodeMCU, I have also tried and received the same error "SPI Interface Error", the difference started to run I think it is in the ArduCAM_v3.3.1 not sure.
my intention really is to try to run this camera with pic32mz or  stm32f4 and try to get 60fps 1280x720 , I do not know if the FPGA allows do this
Thank you.

ArduCAM, 2016-02-03:
Hi Capblack, are you sure ArduCAM-Mini-5MP works fine with Mega2560? Did you try the workaround hieudini suggested?
The ArduCAM is designed for high resolution still image not for video. Due to the limitation of the SPI speed, you can only get low resolution low frame rate video.
Let me know if you need more help.

capblack, 2016-02-04:
Hi Mr Lee, yes I'm sure , I implement this example and now work fine
http://www.arducam.com/how-to-connect-arducam-shield-to-mega-2560/#more-509
Yes I can see the SPI is very slow, maybe I move to OV5642 Camera Module ?
can this module bring 60 fps on 720p ?
Thank you.

ArduCAM, 2016-02-05:
Hi capblack, the OV5642 datasheet says it supports 60fps at 720p, but I'm afraid that it is not easy to make it work on that setting.
would you please let me know more about your application, maybe I can give you more suggestions.
We will be off for Chinese New Year holiday from Feb. 5th to Feb. 16th, so my reply will be delay, please understand.

capblack, 2016-02-10:
Hi Mr Lee, yes is not intended for Arduino ,could be on pic32mz or STM32F4. the idea is to use it to FPV, so I need a high fps and low latency essentially is the most important.
on the other hand I have development a high-speed link RF, that's why I brought her camera simple.
thanks for your kindness

ArduCAM, 2016-02-11:
Hi capblack. Yes I understand. But as you can see the 5MP full frame output is only 15fps. If you need 720p with 60fps, you have to tune the register in the sensor which is not easy though.
I'm interested in your high-speed RF link, what kind of solution do you want to use?

capblack, 2016-02-11:
Hi Mr Lee,I have developed a link on standard 5.8ghz video transmitter and receiver, similar to Manchester encoding and reached 0.5 Mb / s enough to carry  720p at 60fps I think . the point is I do not have much experience in digital video cameras and that's why I'm trying, and I started with ArduCam Mini  found it more simple, I had no account the limitation of spi speed , but anyway I'm looking for a fast camera that can give me raw data, in the easiest way :)
Thanks

ArduCAM, 2016-02-14:
I'm afraid you can not use raw data which is much bigger size. In your application you can only use MJPEG or H.264 stream to transfer the realtime video, MJPEG is eaiser compared to H.264.
The OV5642 camera support MJPEG output, but as I said eariler to get 60fps@720p output from OV5642 is not easy, we have to try.

capblack, 2016-02-14:
Hi Mr Lee, when I referred to raw data, I meant hypothetically go obtaining data byte by byte each photo and then process it in a shift register and transmit them, then the receiver was in charge of rebuilding the data, surely a compressed format such as MJPEG facilitates work,, but as you say the main thing is to make the camera to operate in that resolution.

andrawes, 2016-03-01:
Dear ArduCam, 

My new camera so far has been only giving me the SPI interface error message so far. When I see what "temp" returns, it seems that it returns 0.
Thank you for your help, 
Andy

ArduCAM, 2016-03-01:
@andrawes, would you please let me know what kind of ArduCAM you have and what kind of Arduino board  you are using, what is the wiring between them?

AHVMF, 2016-03-23:
Dear @ArduCAM, 

I am trying to get a 4 camera burst for OV2640 using the OV5642 code. I tried making changes to the number of frames and also to the resolution size. Although the cameras are being detected, i am not getting any image. And the size i am receiving from the host application is 0. 

May i please know what are the changes i need to make in the OV5642 code to get 4 OV2640 to work in the burst mode.

Thank you so much!!

TimoteoMendes, 2016-05-03:
hello all,
recently i bought a arduCAM 4 cameras adapter board and a arducam mini 5MP.
connect all together using an arduino uno but always have the same problem with all exemples.
First i have to include the:
#include <ov5642_regs.h>
because is not on the examples and change the baud rate for serial 115200.
i allwasy receave a nicer reply that all is ok but stops on take picture.
Finally i even used the code on this page from the geolog125 but have the same problem.
After adapted it to use my ov5642.
always receave this:
ArduCAM Start!
SPI All Good
SD Card detected!
Taking picture...
Init ? OK
Waiting for capture...

anyhelp is apreciated.
If this is an hardware problem i have to return the camera.
Thanks,
Timóteo

tomburtonwood, 2016-06-10:
Hello, 

Is there a drawing, schematic or photograph showing how to connect the 4 camera module to the Raspberry Pi - as shown in this image ( http://www.arducam.com/arducam-porting-raspberry-pi/arducam-pizero-3/ ) would appreciate some guidance on this - i just bought the module and would love to get up and running asap.
T

dmainmon, 2017-02-19:
[https://youtu.be/XqfSz1jZ2JA](https://youtu.be/XqfSz1jZ2JA)

ArduCAM Mini with ESP8266-12E. Working example and source code.

merimem, 2017-04-05:
Hello,

I just bought an ARDUCAM PiNOIR ov5647 with CS mount and I will use it for the raspberry pi 3. The problem is that I cannot find the documentation of the camera. There is no code available for it. Please can any one help me??

diegoto, 2017-05-04:
Hello deltaGPhys, did you manage to activate your camera ov 5642 and SD? I have the same problem and I am not able to solve it``

diegoto, 2017-05-08:
Hi capblack, I am trying to connect my OV 5642 camera, and I am not able to make it work. I have not taken any photos yet. I would like to get in touch with you to resolve any doubts.

a greeting

UCTRONICS, 2017-05-09:
@tomburtonwood
Hi,
Which demo do you use? Maybe it's because the code exist some bugs.
Please tell us and we will check it and give you a reply as soon as quickly.

Let us know if you need more help.
Regards,
ArduCAM support team.

capblack, 2017-05-09:
Hi diegoto , I have set up this camera without any problem, with arduino mega and esp8266 and one, with the examples mentioned here, they all work without any problem, try to verify the connections, and also you can check here

SPI.begin();
 64 //Check if the ArduCAM SPI bus is OK
 65 myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
 66 temp = myCAM.read_reg(ARDUCHIP_TEST1);
 67 Serial.println(temp); <<<<<<========== here temp return 0x85 "SPI Interface Error"
 68 if (temp != 0x55)
 69 {
 70 Serial.println("SPI interface Error!"); <<<=== If you get this , there is a hard problem
 71 //while(1);
 72 }

annienair, 2017-06-14:
Hi,
I am using an arducam shield v2 on arduino uno and I keep getting the SPI interface error for any code that I try to execute.
The myCAM.read_reg(ARDUCHIP_TEST1) function returns 0 always for SPI_CS 10 in the following code.
//Check if the ArduCAM SPI bus is OK
 myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
 temp = myCAM.read_reg(ARDUCHIP_TEST1);
Serial.println(temp);
 if (temp != 0x55)
 {
Serial.println("SPI interface Error!");
}
}

VIVEKSUTAR0709, 2017-11-14:
Hello Sir/Madam

I have a problem in Websocket Camera application
It shows me Websocket Connected while clicking on Connect (file attached)

but when I press Send Button it won't show anything..

please help mi with this.
[image]

jay07bm26, 2017-12-02:
hello? 
I have ov2640 camera module but for video streaming over wifi its very slow, i get only 1 fps,
i cannot understand whether ov2640 is not capable of video streaming or should i move for another camera module like ov5642?
can you suggest which camera is better for video stream over wifi

[7 more replies]