ArduCAM/Arduino #3: ArduCAM_Mini_OV2640_Yun_WebServer — IP Address Bug

rei-vilo, 2015-08-08:
There a bug affecting the IP address in the ** ArduCAM_Mini_OV2640_Yun_WebServer** example. 

By default, the last character is `#0A`, which makes the picture in the web page blank.

Just add `if (ip[m] == 0x0A) ip[m] = 0x00;` to line *ca.* 130

```
  while (q.available())
  {
    ip[m] = q.read();
    if (ip[m] == 0x0A) ip[m] = 0x00;
    m += 1;
  }
```

With this fix, the image is displayed correctly.

ArduCAM, 2015-08-09:
Thanks for your comment. I do not have such problem with IP address parse.

[image]
[image]

rei-vilo, 2015-08-09:
My configuration includes

* Arduino Yun with YunSys v1.5.3
* Arduino IDE 1.6.5

What's yours?

ArduCAM, 2015-08-10:
Firmware Version	OpenWRTYun Attitude Adjustment 1 / LuCI 0.11 Branch (0.11+svn10537)
Kernel Version	3.3.8
Arduino IDE 1.5.8

closed 2015-08-20 by ArduCAM