ArduCAM/Arduino #4: ESP8266 Arduino Core

sh0d5n, 2015-08-31:
I've been trying to get an ArduCAM mini to work with a LoLin ESP8266 ESP-12 using the Arduino core and Arduino IDE.

It works perfectly on my UNO.

There was a compile issue with ArduCAM.h as it does not define a section for "ESP8266". Having fixed (bodged) that I think my inserted section has issues because the following line in ArduCAM.cpp causes the ESP to crash and restart.

207 	//Must initialize the Bus default status 
208 	sbi(P_CS, B_CS);

I don't suppose you've experiment with this setup at all? I'm afraid my personal knowledge of Arduino code and boards is not very good at the moment and I'm not really sure where to start debugging it. I'm a c# developer and trying to learn.

ArduCAM, 2015-09-02:
Dear Sir,
Thanks for your question.
We also plan to use our ArduCAM with ESP8266 wifi module, so we can work together on this.
Would you please send your code, then I can debug here? 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: sh0d5n
Date: 2015-08-31 18:04
To: ArduCAM/Arduino
Subject: [Arduino] ESP8266 Arduino Core (#4)
I've been trying to get an ArduCAM mini to work with a LoLin ESP8266 ESP-12 using the Arduino core and Arduino IDE.
It works perfectly on my UNO.
There was a compile issue with ArduCAM.h as it does not define a section for "ESP8266". Having fixed (bodged) that I think my inserted section has issues because the following line in ArduCAM.cpp causes the ESP to crash and restart.
207 //Must initialize the Bus default status 
208 sbi(P_CS, B_CS);
I don't suppose you've experiment with this setup at all? I'm afraid my personal knowledge of Arduino code and boards is not very good at the moment and I'm not really sure where to start debugging it. I'm a c# developer and trying to learn.

sh0d5n, 2015-09-02:
Thanks I would really appreciate that. My end goal (along with some sensors) is to create a wifi camera (timelapse upload over wifi) and as the ESP8266 costs about £2 it seems a better bet that an Arduino + £15 of wifi shield! I looked at an Arduino + ESP8266 but its difficult to justify the Arduino at all as the ESP8266 can be programmed via Arduino code.

I am using one of these ESP8266 v12's: http://www.wemos.cc/wiki/Hardware/Pin
And the Arduino core here: https://github.com/esp8266/Arduino

I am also using the latest version of the core and your code.

My Arduino IDE board is set to "Generic ESP8266 Module" and my upload and serial speed is 115200 on COM 8. Arduino IDE programmer defaults to "AVRISP mkII".

I am testing with the "ArduCAM > Mini > ArduCAM Mini OV2640 Video Streaming" example. The only change I have made is to set "const int CS" to "15" as indicated by this post https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/using-the-arduino-addon . (In fact I've tried lots of different pins and am mainly now testing without the ArduCAM attached as I currently appear to have a software issue not a problem with the ArduCAM).

I added the following section to ArduCAM.h but my knowledge currently doesn't go far enough to do anything other than a copy and paste of the "AVR" code:
                 #if defined (ESP8266)
                 #define cbi(reg, bitmask) *reg &= ~bitmask
                 #define sbi(reg, bitmask) *reg |= bitmask
                 #define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
                 #define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);

                 #define cport(port, data) port &= data
                 #define sport(port, data) port |= data

                 #define swap(type, i, j) {type t = i; i = j; j = t;}

                 #define fontbyte(x) pgm_read_byte(&cfont.font[x])  

                 #define regtype volatile uint32_t
                 #define regsize uint8_t

                 #endif

It now compiles and uploads to the ESP8266.

Brute force (removing code until it works!) means I know my first issue is this line in ArduCAM.cpp:

                 Line 207: //Must initialize the Bus default status 
                 Line 208: sbi(P_CS, B_CS);

Which causes the following looped (constantly restarts) error in the Arduino IDE serial window:

                 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

                 load 0x4010f000, len 1264, room 16 
                 tail 0
                 chksum 0x42
                 csum 0x42
                 ~ld

Obviously my new section in the header file is wrong!

Thanks again!!

ArduCAM, 2015-09-03:
Dear Sir,
Thanks for your detail information.
I'm quite new to the ESP8266, but it looks great. We can use the ESP8266 SOC as the Arduino board and connect to my camera directly.
So I think it is very worthwhile to port my library to ESP8266.
I will do some experiments on ESP8266 these days. I will let you update if I made some progress.
 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: sh0d5n
Date: 2015-09-03 04:32
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ESP8266 Arduino Core (#4)
Thanks I would really appreciate that. My end goal (along with some sensors) is to create a wifi camera (timelapse upload over wifi) and as the ESP8266 costs about £2 it seems a better bet that an Arduino + £15 of wifi shield! I looked at an Arduino + ESP8266 but its difficult to justify the Arduino at all as the ESP8266 can be programmed via Arduino code.
I am using one of these ESP8266 v12's: http://www.wemos.cc/wiki/Hardware/Pin
And the Arduino core here: https://github.com/esp8266/Arduino
I am also using the latest version of the core and your code.
My Arduino IDE board is set to "Generic ESP8266 Module" and my upload and serial speed is 115200 on COM 8. Arduino IDE programmer defaults to "AVRISP mkII".
I am testing with the "ArduCAM > Mini > ArduCAM Mini OV2640 Video Streaming" example. The only change I have made is to set "const int CS" to "15" as indicated by this post https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/using-the-arduino-addon . (In fact I've tried lots of different pins and am mainly now testing without the ArduCAM attached as I currently appear to have a software issue not a problem with the ArduCAM).
I added the following section to ArduCAM.h but my knowledge currently doesn't go far enough to do anything other than a copy and paste of the "AVR" code:
#if defined (ESP8266)
#define cbi(reg, bitmask) *reg &= ~bitmask
#define sbi(reg, bitmask) *reg |= bitmask
#define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
#define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);
             #define cport(port, data) port &= data
             #define sport(port, data) port |= data
             #define swap(type, i, j) {type t = i; i = j; j = t;}
             #define fontbyte(x) pgm_read_byte(&cfont.font[x])  
             #define regtype volatile uint32_t
             #define regsize uint8_t
             #endif

It now compiles and uploads to the ESP8266.
Brute force (removing code until it works!) means I know my first issue is this line in ArduCAM.cpp:
             Line 207: //Must initialize the Bus default status 
             Line 208: sbi(P_CS, B_CS);

Which causes the following looped (constantly restarts) error in the Arduino IDE serial window:
             ets Jan  8 2013,rst cause:2, boot mode:(3,6)
             load 0x4010f000, len 1264, room 16 
             tail 0
             chksum 0x42
             csum 0x42
             ~ld

Obviously my new section in the header file is wrong!
Thanks again!!

sh0d5n, 2015-09-03:
Thanks.

Please let me know if I can do anything for you, testing etc...

Thanks again

ArduCAM, 2015-09-11:
Hi Again,
Sorry for my late reply.
Yes, you are right, it is lost definition of Macros for ESP8266 in the ArduCAM.h head file.
I did the same thing like you did to add support for ESP8266, I found that I need to get the register physical address of the CS pin in my ArduCAM constructor .
ArduCAM::ArduCAM(byte model,int CS) 
{ 
P_CS = portOutputRegister(digitalPinToPort(CS)); 
B_CS = digitalPinToBitMask(CS);
But I found the ESP8266 core doesn't implement the function like portOutputRegister and  digitalPinToBitMask, I have to use digitalWrite to replace all the sbi and sbi macro. 
Do you have any ideas how to implement the portOutputRegister and  digitalPinToBitMask functions in order to keep compatibility?

Best Regards,
Lee
 
-----------------------------------------------------
 
From: admin@arducam.com
Date: 2015-09-03 22:31
To: ArduCAM/Arduino
Subject: Re: Re: [Arduino] ESP8266 Arduino Core (#4)
Dear Sir,
Thanks for your detail information.
I'm quite new to the ESP8266, but it looks great. We can use the ESP8266 SOC as the Arduino board and connect to my camera directly.
So I think it is very worthwhile to port my library to ESP8266.
I will do some experiments on ESP8266 these days. I will let you update if I made some progress.
 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: sh0d5n
Date: 2015-09-03 04:32
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ESP8266 Arduino Core (#4)
Thanks I would really appreciate that. My end goal (along with some sensors) is to create a wifi camera (timelapse upload over wifi) and as the ESP8266 costs about £2 it seems a better bet that an Arduino + £15 of wifi shield! I looked at an Arduino + ESP8266 but its difficult to justify the Arduino at all as the ESP8266 can be programmed via Arduino code.
I am using one of these ESP8266 v12's: http://www.wemos.cc/wiki/Hardware/Pin
And the Arduino core here: https://github.com/esp8266/Arduino
I am also using the latest version of the core and your code.
My Arduino IDE board is set to "Generic ESP8266 Module" and my upload and serial speed is 115200 on COM 8. Arduino IDE programmer defaults to "AVRISP mkII".
I am testing with the "ArduCAM > Mini > ArduCAM Mini OV2640 Video Streaming" example. The only change I have made is to set "const int CS" to "15" as indicated by this post https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/using-the-arduino-addon . (In fact I've tried lots of different pins and am mainly now testing without the ArduCAM attached as I currently appear to have a software issue not a problem with the ArduCAM).
I added the following section to ArduCAM.h but my knowledge currently doesn't go far enough to do anything other than a copy and paste of the "AVR" code:
#if defined (ESP8266)
#define cbi(reg, bitmask) *reg &= ~bitmask
#define sbi(reg, bitmask) *reg |= bitmask
#define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
#define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);
             #define cport(port, data) port &= data
             #define sport(port, data) port |= data
             #define swap(type, i, j) {type t = i; i = j; j = t;}
             #define fontbyte(x) pgm_read_byte(&cfont.font[x])  
             #define regtype volatile uint32_t
             #define regsize uint8_t
             #endif

It now compiles and uploads to the ESP8266.
Brute force (removing code until it works!) means I know my first issue is this line in ArduCAM.cpp:
             Line 207: //Must initialize the Bus default status 
             Line 208: sbi(P_CS, B_CS);

Which causes the following looped (constantly restarts) error in the Arduino IDE serial window:
             ets Jan  8 2013,rst cause:2, boot mode:(3,6)
             load 0x4010f000, len 1264, room 16 
             tail 0
             chksum 0x42
             csum 0x42
             ~ld

Obviously my new section in the header file is wrong!
Thanks again!!

sh0d5n, 2015-09-11:
I appreciate you looking.

Sorry no idea, beyond my understanding at the moment :(

sh0d5n, 2015-09-11:
Does this changeset help at all from the esp8266 core site? I note that in the OneWire library they are replacing the mentioned method calls with alternatives...
https://github.com/esp8266/Arduino/pull/160/files#diff-f787a2ccd4d80dcbd997fda587e7d441

In the commits there is one which says "replease digitalPinToPort, digitalPinToBitMask, portOutputRegister, p…" https://github.com/Links2004/Arduino/commit/f1ae1ba2671ea8beeebda2f22eb7ec869caae01b

ArduCAM, 2015-09-14:
Thanks for your information.
I will try to figure it out and let you update. 
If you find something new, please also let me know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: sh0d5n
Date: 2015-09-12 05:51
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ESP8266 Arduino Core (#4)
Does this changeset help at all from the esp8266 core site? I note that in the OneWire library they are replacing the mentioned method calls with alternatives...
https://github.com/esp8266/Arduino/pull/160/files#diff-f787a2ccd4d80dcbd997fda587e7d441
In the commits there is one which says "replease digitalPinToPort, digitalPinToBitMask, portOutputRegister, p…" Links2004/Arduino@f1ae1ba

ArduCAM, 2015-09-19:
Dear Sir,
Yes, you are right they have resolved the problem of digitalPinToPort, digitalPinToBitMask, portOutputRegister porting.
I made some mistake in my previous email.
I think it is no problem to use ArduCAM with the ESP8266 Arduino board.
I will try it out ASAP.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: admin@arducam.com
Date: 2015-09-14 09:25
To: ArduCAM/Arduino
Subject: Re: Re: [Arduino] ESP8266 Arduino Core (#4)
 
Thanks for your information.
I will try to figure it out and let you update. 
If you find something new, please also let me know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: sh0d5n
Date: 2015-09-12 05:51
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ESP8266 Arduino Core (#4)
Does this changeset help at all from the esp8266 core site? I note that in the OneWire library they are replacing the mentioned method calls with alternatives...
https://github.com/esp8266/Arduino/pull/160/files#diff-f787a2ccd4d80dcbd997fda587e7d441
In the commits there is one which says "replease digitalPinToPort, digitalPinToBitMask, portOutputRegister, p…" Links2004/Arduino@f1ae1ba

ArduCAM, 2015-09-20:
Dear Sir,
I have made our ArduCAM-mini work on ESP8266.
I made some changes to ArduCAM.cpp source file.
change the sbi and cbi with simple digitalWrite function.
I will update my library ASAP.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: admin@arducam.com
Date: 2015-09-19 21:23
To: ArduCAM/Arduino
Subject: Re: Re: [Arduino] ESP8266 Arduino Core (#4)
Dear Sir,
Yes, you are right they have resolved the problem of digitalPinToPort, digitalPinToBitMask, portOutputRegister porting.
I made some mistake in my previous email.
I think it is no problem to use ArduCAM with the ESP8266 Arduino board.
I will try it out ASAP.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: admin@arducam.com
Date: 2015-09-14 09:25
To: ArduCAM/Arduino
Subject: Re: Re: [Arduino] ESP8266 Arduino Core (#4)
 
Thanks for your information.
I will try to figure it out and let you update. 
If you find something new, please also let me know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: sh0d5n
Date: 2015-09-12 05:51
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ESP8266 Arduino Core (#4)
Does this changeset help at all from the esp8266 core site? I note that in the OneWire library they are replacing the mentioned method calls with alternatives...
https://github.com/esp8266/Arduino/pull/160/files#diff-f787a2ccd4d80dcbd997fda587e7d441
In the commits there is one which says "replease digitalPinToPort, digitalPinToBitMask, portOutputRegister, p…" Links2004/Arduino@f1ae1ba

sh0d5n, 2015-09-20:
Thanks Lee, I really appreciate how quickly you've worked on this. Brilliant!

ArduCAM, 2015-09-20:
Hi sh05n
I have update the Arduino libraries, now it support the ESP8266 processor.
The examples are in the /example/ESP8266 folder.
Let me know if you still have any issue with this update.
Lee

closed 2015-09-20 by ArduCAM

ArduCAM, 2015-09-20:
Dear Sir,
I have update my Arduino libraries for ESP8266-12 board. Please try it out and let me know if you still have any issues.
Now one thing I need you help.
I want to use ESP8266 to take photos and create a webserver, user can start capture and review the image on the webpage.
Even more if possible, the ESP8266 can stream a live MJPG video to the webpage.
Can you help me how to do that on both ESP8266 and html programming?
Thanks in advance.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: sh0d5n
Date: 2015-09-20 15:38
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ESP8266 Arduino Core (#4)
Thanks Lee, I really appreciate how quickly you've worked on this. Brilliant!

sh0d5n, 2015-09-21:
Sorry, I've only just gotten a chance to try (new baby) and am having an issue but suspect it is a crossed wire...

With reference to this image (http://www.wemos.cc/wiki/Hardware/Pin) and your wiring guide (http://www.arducam.com/arducam-mini-released/) I've wired it...
CS -> CMD (CS)
MOSI -> SD1 (MOSI)
MISO-> SD0 (MISO)
SCK -> CLK (SCLK)
GND -> GND
5V -> VIN 5V (VIN)
SDA -> D5 (GPIO14)
SCL -> D6 (GPIO12)

I suspect I've connect SDA and SCL wrong? I'm using the example "ArduCAM_Mini_OV2640_Video_Streaming" in the ESP8266 and have changed "Serial.Begin" to 115200.

I get the following errors:

SPI1 interface Error!
Can't find OV2640 module!

I've looked at images on webpages but haven't tried it yet. As a first step I was looking at example Arduino code which serves an image from an SD card to a web page. When googling for a reference just now I found this one...
http://blog.startingelectronics.com/arduino-web-server-tutorial/ 
"◾Part 11: Arduino SD Card Web Server – Displaying Images – a web page that contains an image is hosted by the Arduino web server."

Thanks again for your help

ArduCAM, 2015-09-22:
Dear Sir,
Thanks for your reply.
The wiring is wrong, I use the following connection.

CS -> D0(GPIO16)
MOSI -> D7(GPIO13)
MISO-> D6(GPIO12)
SCK -> D5 (GPIO14)
GND -> GND
5V -> VIN 5V (VIN)
SDA -> D2 (GPIO4)
SCL -> D1 (GPIO5)

Thanks for your link of arduino web server tutorial, but I would like to transfer the image/video to webpage without to store the image to SD card first, and transfer it directly to web page. Do you know how to do that.

Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: sh0d5n
Date: 2015-09-22 02:41
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ESP8266 Arduino Core (#4)
Sorry, I've only just gotten a chance to try (new baby) and am having an issue but suspect it is a crossed wire...
With reference to this image (http://www.wemos.cc/wiki/Hardware/Pin) and your wiring guide (http://www.arducam.com/arducam-mini-released/) I've wired it...
CS -> CMD (CS)
MOSI -> SD1 (MOSI)
MISO-> SD0 (MISO)
SCK -> CLK (SCLK)
GND -> GND
5V -> VIN 5V (VIN)
SDA -> D5 (GPIO14)
SCL -> D6 (GPIO12)
I suspect I've connect SDA and SCL wrong? I'm using the example "ArduCAM_Mini_OV2640_Video_Streaming" in the ESP8266 and have changed "Serial.Begin" to 115200.
I get the following errors:
SPI1 interface Error!
Can't find OV2640 module!
I've looked at images on webpages but haven't tried it yet. As a first step I was looking at example Arduino code which serves an image from an SD card to a web page. When googling for a reference just now I found this one...
http://blog.startingelectronics.com/arduino-web-server-tutorial/ 
"◾Part 11: Arduino SD Card Web Server – Displaying Images – a web page that contains an image is hosted by the Arduino web server."
Thanks again for your help

sh0d5n, 2015-09-22:
Sorry I'm still doing something wrong :(

I've connected as you have suggested and also tried some other variations... no luck.
"SPI1 interface Error!
 Can't find OV2640 module!"

I have retested the arducam module on my UNO and it works ok still.

Sorry no, I don't at the moment

ArduCAM, 2015-09-22:
I use Generic ESP8266. 
Please double check the pin definition on both hardware and sketch, maybe I got some thing wrong.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: sh0d5n
Date: 2015-09-22 09:56
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ESP8266 Arduino Core (#4)
Sorry I'm still doing something wrong :(
I've connected as you have suggested and also tried some other variations... no luck.
"SPI1 interface Error!
Can't find OV2640 module!"
I have retested the arducam module on my UNO and it works ok still.
Sorry no, I don't at the moment

sh0d5n, 2015-09-22:
Thanks Lee. On closer inspection it appears my ESP8266 development board has a blob of solder between the SDA and SCL pins. I'm trying to sort it and will let you know.

Thanks again

dmolner, 2015-12-11:
I'm very interested in this. 
Could you tell me if this works with Yawcam?

sh0d5n, 2015-12-12:
PS. sorry I never said, this works great. The GPIO's were wired differently on my development board which was why it didn't originally work.

dmolner, 2015-12-12:
Thank you Sh0d5n
Could you send a link for see the picture or you don't upload files to Internet?
I think this is a good system for a Weather Cam.

tiller123, 2016-01-29:
Hi!

I just came across your posts and find them very interesting.  Can you please clarify which camera is being discussed?  I know the OV7670 camera comes in several versions: 1) no FIFO, 2) FIFO with 8 output pins and parallel readout, 3) SPI output and maybe 4) serial TTL output.  Which one of thee are you discussing?

I am interested in version 2) FIFO and 8 pins to read out a byte at a time.  Is the ArduCAM library applicable to this camera?  What is the best pins connection for the ESP8266-12 with this camera?

Thank you

ArduCAM, 2016-02-01:
Hi Sir,
I'm discuss the 3rd scenario in your list, SPI camera interface.
But you can also find the FIFO version from us: http://www.uctronics.com/ov7670-0-3-megapixel-1-6-inch-cmos-sensor-camera-module-fifo.html 
Here is the demo code, but it is not designed fro ESP8266, it uses too much pins.
https://github.com/ArduCAM/Arduino/tree/master/OV7670FIFO 
So we recommend to use our ArduCAM-Mini to connect to ESP8266 like this:
http://www.arducam.com/arducam-supports-esp8266-arduino-board-wifi-websocket-camera-demo/ 
Let me know if you have more questions.
Regards,
Lee

daclay, 2016-03-03:
Good Day,

I am trying to make the ESP8266 ArduCam_Mini_OV2640_websocket_server example program work.

I am running Arduino version 1.6.7 software, Firefox web browser, Windows 7, and an Adafruit Huzzah 8266 board. The web page is stored on the PC hard drive, and opened as a file rather than being loaded from another server.

When I start the 8266, I get the following messages: Ov2640 detected, Connecting to "Testing_SSID", WiFi connected, Server started, and the IP address of the 8266.

I open the web page with Firefox, change the IP address box, and click connect. The ESP8266 prints Analyzing request headers, and 15 "Got Line:"  messages and hangs. About 20 seconds after clicking on "Connect" with Firefox, I get a Web Socket Error message on the PC, and about 30 seconds after that, the ESP8266 prints "Websocket established".

Any suggestions? Did the newer version of the Arduino software break the example, or is my method of loading the web page at fault?

Thanks,

Best Regards,

Dave
#

ArduCAM, 2016-03-08:
@Dave, sorry for my late reply. Websocket will be a little bit tricky. There is debug information can be printed in the serial monitor,  you can try that first. We have another demo for ESP8266 which is based on http protocol, you can check it from: https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture

daclay, 2016-03-08:
Hello Lee,

Thanks for your response, no problem with your response time!

I already tried the other program while waiting for this response. It works fine with two changes. There are two places which define the ESP output buffer size of 4096 which need to be changed to 2048 because of a limitation of the ESP8266. 

I was interested in the websocket version for fastest frame rate, but based the testing with the capture program, the ESP input and output data rates are a stronger influence, so making the websocket program / web page work is no longer an issue for me.

Thanks again for your help. You have a very good product!

Best Regards,
Dave

dmolner, 2016-03-15:
Hello,

I test a OV2640 with ESP8266 but every test I have same answer:
ArduCAM Start!
Can't find OV2640 module!

I tested wires and solders.
I tested with two Lolin Wemos anb one Wemos D1 mini

I need help for solve this problem.

Best regards,
David.
[image]

ArduCAM, 2016-03-15:
David, Please modify the code like this
if ( ( vid == 0x26 ) && ( ( pid == 0x41 ) || ( pid == 0x42 ) ) ) {
    Serial.println("OV2640 detected.");
  }else{
    Serial.println("Can't find OV2640 module!");
    while(1);
  }
Wish you success.

dmolner, 2016-03-15:
Great! Now work and I can connect to ESP8266
But how I see the camera?
Can I connect with explorer to camera?
The IP for Arducam+ESP8266 is 192.168.1.143

ArduCAM, 2016-03-17:
Please use example https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture 
and modify the ESP8266 core library with following changes from folder ..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0 which normally located on C driver.
//--------------------------------------------------
[WiFiClient.h]
#define WIFICLIENT_MAX_PACKET_SIZE 2048
 
[ClientContext.h]
add the line 
_pcb->snd_buf = TCP_SND_BUF;
before
size_t room = tcp_sndbuf(_pcb);
 
[lwipopts.h]
#define TCP_MSS                         2048

//--------------------------------------------------
Then use the html page to control the camera:
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture/html/video.html

dmolner, 2016-03-17:
Perfect!
But...

I see correct video if I put 192.168.1.143/stream in web browser.
I see picture with 192.168.1.143/capture
But not working video.html page, I think I need install anything else in my PC.
I change the IP for adress my cam but not work.

Is possible send capture to ftp server?
I bought too a 5MP Arducam could this work same?
Best regards,
David

[image]

dmolner, 2016-03-18:
Good news.
I tried some things and finally works with Ispy program.
I can see and capture pictures.
I need now adjust the focus
I need send the image to an FTP server with the ESP8266 for make an autonomus webcam.

[image]

ArduCAM, 2016-03-18:
@dmolner, congratulations. What do you mean by lspy program?
You can use the html page to control the camera and tune the focus:
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture/html/video.html
We don't have FTP server demo right now.

dmolner, 2016-03-18:
Hello, 
Ispy is a webcam program. https://www.ispyconnect.com
HTML page video.html not work in my PC

ArduCAM, 2016-03-22:
@dmolner, 
Please use example https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture 
and modify the ESP8266 core library with following changes from folder ..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0 which normally located on C driver.
//--------------------------------------------------
[WiFiClient.h]
#define WIFICLIENT_MAX_PACKET_SIZE 2048
 
[ClientContext.h]
add the line 
_pcb->snd_buf = TCP_SND_BUF;
before
size_t room = tcp_sndbuf(_pcb);
 
[lwipopts.h]
#define TCP_MSS                         2048

//--------------------------------------------------
Then use the html page to control the camera:
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture/html/video.html 

admin@arducam.com

msurguy, 2016-03-24:
Thanks for posting here, @ArduCAM! This is a great reference. 

Which other ArduCam cameras will work with ESP8266? I think if there is more than one, marketing them on your website towards ESP8266 could get you a lot of orders :)

dmolner, 2016-03-24:
Hello @ArduCAM, 
The Html pages not work on my PC. 
Could you make a zip with all necessary files? 

Best regards. 
David

ArduCAM, 2016-03-28:
@dmolner, David you have to download all the files together with the html, it should work. https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture/html

ArduCAM, 2016-03-28:
@Maksim, thanks for your question and suggestion, the ArduCAM mini 2mp and 5mp both work with esp8266. we need your involvement to make it better place for camera applications.

hjsalom, 2016-04-20:
Arducam is a 5 volts DC device while ESP8266-12 is a 3.3 volts DC device so... how are you connecting them without damaging the module ?

ArduCAM, 2016-04-27:
@hjsalom  You can power the ArduCAM with 3.3V and then connect to ESP8266 directly.