ArduCAM/Arduino #5: ArduCAM Mini 5MP rev B high res JPEG issue

DVuckovc, 2015-09-02:
Hi,

Great job on making a nice camera for an Arduino.

I have an issue when reading the pictures that have a larger resolution than 640 x 480. Namely the picture has an appropriate file size, however, when displayed half of the picture is missing. 

I am using Arduino MEGA 2560 and this sketch to test the ArduCAM mini  5MP:
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_OV5642_Video_Streaming

I am using the host application console for the test
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_host_v1.0.exe

Here is the screenshot of the program. It can be clearly seen that lower part of the picture is missing

[image]

I get a similar result when I read the data over UART as well.
Is there an issue with the image interpretation or is there something else wrong

Thank you in advance
Dusan

ArduCAM, 2015-09-03:
With my experience, the missing parts on the picture comes from the missing data when read from the serial port. The host should read the serial port as fast as it can be, or else the missing data happens.
Some of the computer has too much load, the host application is freeze at some time, so the data will be lost at that time.

ArduCAM, 2015-09-03:
Dear Sir,
Thanks for your question.
With my experience, the missing parts on the picture comes from the missing data when read from the serial port. The host should read the serial port as fast as it can be, or else the missing data happens.
Some of the computer has too much load, the host application is freeze at some time, so the data will be lost at that time. 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: DVuckovc
Date: 2015-09-02 23:26
To: ArduCAM/Arduino
Subject: [Arduino] ArduCAM Mini 5MP rev B high res JPEG issue (#5)
Hi,
Great job on making a nice camera for an Arduino.
I have an issue when reading the pictures that have a larger resolution than 640 x 480. Namely the picture has an appropriate file size, however, when displayed half of the picture is missing. 
I am using Arduino MEGA 2560 and this sketch to test the ArduCAM mini 5MP:
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_OV5642_Video_Streaming
I am using the host application console for the test
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_host_v1.0.exe
Here is the screenshot of the program. It can be clearly seen that lower part of the picture is missing
I get a similar result when I read the data over UART as well.
Is there an issue with the image interpretation or is there something else wrong
Thank you in advance
Dusan

closed 2015-09-20 by ArduCAM

DVuckovc, 2015-09-21:
Hi,

Thank you for the replay I have just a small update. 

I haven't posted any responses as I am waiting for a custom PCB to test a hypothesis that the issue was't coming from the computer not reading the data fast enough, but the fact that the baudrate on an 8MHz Arduino cannot generate an error free UART communication. 
I didnt have any issues when writing the data to an SD card even for 5MP picutres. 

Best Regards
Dusan