ArduCAM/Arduino #6: ArduCAM mini with 4 Cameras Adapter Board SD card error

xDaya, 2015-11-25:
Hi,

I have been working with a normal ArduCAM shield before, and first of all I am very happy that it exists!
I have switched to an ArduCAM mini OV2640 now, because I want to make a smaller device that is more flexible. As I wanted to do some experiments with multiple cameras as well, I bought the 4 Cameras Adapter Board.

I have been able to get the combination running (1 ArduCAM mini with the adapter board) through the software you provided. However, my next step is to save pictures on the SD card. For that, I am trying to run the example code ArduCAM mini 2MP TimeLapse 2SD, bot on an Arduino UNO and an Arduino Mega 2560. On both boards it gives me the message 'SD card error' though.

Do you know what might be causing this problem and how I might solve it? At the moment I attached the board directly to the Arduino without extra wires.

Thank you!

ArduCAM, 2015-11-29:
Dear Sir,
Sorry for my late reply.
Would you please send us your configuration of the whole system.
I will try my best to help you resolve the problem.
 
Any question else,please feel free to let us know.
Best Regards,
Lee
 
-----------------------------------------------------
 
From: xDaya
Date: 2015-11-25 19:12
To: ArduCAM/Arduino
Subject: [Arduino] ArduCAM mini with 4 Cameras Adapter Board SD card error (#6)
Hi,
I have been working with a normal ArduCAM shield before, and first of all I am very happy that it exists!
I have switched to an ArduCAM mini OV2640 now, because I want to make a smaller device that is more flexible. As I wanted to do some experiments with multiple cameras as well, I bought the 4 Cameras Adapter Board.
I have been able to get the combination running (1 ArduCAM mini with the adapter board) through the software you provided. However, my next step is to save pictures on the SD card. For that, I am trying to run the example code ArduCAM mini 2MP TimeLapse 2SD, bot on an Arduino UNO and an Arduino Mega 2560. On both boards it gives me the message 'SD card error' though.
Do you know what might be causing this problem and how I might solve it? At the moment I attached the board directly to the Arduino without extra wires.
Thank you!

BrahMOs94, 2015-12-11:
Hello ,
I am Working with Arducam ov5642 module!
I want to trigger four modules at the same time and want to see the images via serial monitor or any .exe host application.
I am not able to find an host application for Arducam 0v5642 4Cams burst.ino!
Can anyone please help me??

ArduCAM, 2015-12-12:
Hi,
You can find the host application for 4 cameras from: www.arducam.com/downloads/app/ArduCAM_4CAM.exe 
Let me know if you need any more help.
Regards

admin@arducam.com
 
From: BrahMOs94
Date: 2015-12-11 14:24
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCAM mini with 4 Cameras Adapter Board SD card error (#6)
Hello ,
I am Working with Arducam ov5642 module!
I want to trigger four modules at the same time and want to see the images via serial monitor or any .exe host application.
I am not able to find an host application for Arducam 0v5642 4Cams burst.ino!
Can anyone please help me??

BrahMOs94, 2015-12-12:
Thanks alot for your quick response
With regards,
Adithya R

htr99, 2015-12-14:
Hi 

I also has same issue of xDaya. xDaya if you ave solved your issue can you post the solution. I appreciate it.

Thank you.

BrahMOs94, 2015-12-14:
hello lee,
Actually I have only two modules I tried triggering those modules using
ARDUCAM_5mp_4Cams_burst program and I also used the host application which
you sent me !
But there is a problem when I press the capture button the camera modules
are not capturing the photos,whereas when I change the resolution in the
dropdown list to the maximum these two camera modules are capturing photos!
I am not sure wheather i have explained my query effectively!
And I also want to know wheather I can trigger the camera modules at the
same time I mean in this program the camera modules are triggered
sequentially , I want to trigger these modules Simultaneously!!
Please reply as soon as possible,
Regards

xDaya, 2015-12-14:
I'm sorry I didn't reply anymore.
Thank you for your answer, I solved the problem by attaching the ArduCam mini without the shield; it seems as if the shield is not fully compatible with the SPI pins of the Arduino Mega 2560.

htr99, 2015-12-14:
Hi xDaya,

Thank for your response. Did you change the code? I men otherthan changing SPI CS pin definitions?

xDaya, 2015-12-14:
I didn't just change the SPI CS pin definitions, I rewired the ArduCam completely and didn't use the shield anymore. Besides that I believe I didn't change the code, and it makes sense, since the SPI pins on Arduino Uno are 11, 12 and 13, whereas on the Mega they are 50, 51 and 52. I think that is why the shield doesn't work properly with the Mega.

htr99, 2015-12-14:
Hi Xdaya,

In your first post you have mentioned on UNO board also you got same SD card error message. So then problem may exist with ardu cam adapter somehow. Did you successful with ARDU CAM ADAPTER board + UNO set up?

xDaya, 2015-12-14:
I'm afraid I haven't, because I was quite sure I needed the Mega, and wiring it without the adapter board was a better option for what I was trying to make anyway. If you want to try that I can show you how I made it work.

ArduCAM, 2015-12-15:
Hi All,
The arducam mini adapter board use ICSP connector for SPI signals, all arduino boards has the same pin definition on ICSP connector. So the adapter can work with any arduino boards and much easier to do the wring by hand.

admin@arducam.com
 
From: xDaya
Date: 2015-12-15 06:15
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCAM mini with 4 Cameras Adapter Board SD card error (#6)
I'm afraid I haven't, because I was quite sure I needed the Mega, and wiring it without the adapter board was a better option for what I was trying to make anyway. If you want to try that I can show you how I made it work.

ArduCAM, 2015-12-15:
Hi, from your description it seems that you didn't initialize the camera properly.
So please initialize the camera and set to correct resolution then start capture.
And as you can see there is no external hardware trigger pin on the ArduCAM mini board, so we have to use SPI command to trigger the capture.
The capture command time interval between each camera should be micro seconds, it is nearly the same time.
Let me know if you need more help.
Regards,
Lee

admin@arducam.com
 
From: BrahMOs94
Date: 2015-12-14 16:16
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] ArduCAM mini with 4 Cameras Adapter Board SD card error (#6)
hello lee,
Actually I have only two modules I tried triggering those modules using
ARDUCAM_5mp_4Cams_burst program and I also used the host application which
you sent me !
But there is a problem when I press the capture button the camera modules
are not capturing the photos,whereas when I change the resolution in the
dropdown list to the maximum these two camera modules are capturing photos!
I am not sure wheather i have explained my query effectively!
And I also want to know wheather I can trigger the camera modules at the
same time I mean in this program the camera modules are triggered
sequentially , I want to trigger these modules Simultaneously!!
Please reply as soon as possible,
Regards

htr99, 2015-12-15:
Hi Lee,
Any solution for my problem, I use Arduino uno, 4 Cam adapter board + your example code.

Thank you.

BrahMOs94, 2015-12-15:
Hello Lee,
I have checked my camera initialization there is nothing wrong in that! I
just wanna trigger two camera modules using ArduCAM 4Cams burst and that
ArduCAM exe file which you sent me.
I have checked the program so many times but I am not able to get
continuous shots using two camera modules. If I press the capture button in
the .exe application it is not responding. Even I tried commenting camera 3
and camera 4 in my ArduCAM program but nothing I'd happening. If I change
the resolution in my .exe file then camera's are working that too not in a
loop.. My question is can I trigger two camera modules using that host
application and the ArduCAM program????
(If you check the program only camera 1 has been initialised I don't know
the reason for that also, please let me know).
Thank you

htr99, 2015-12-15:
Hi xDaya,

Yes can you please send me any details and documents and the working code for my email address please. It will be a great help. My email : hthisurar@gmail.com

Thank you.

xDaya, 2015-12-16:
htr99,

Is it okay if I send you the code and documentation next week? I'm a bit busy at the moment.

htr99, 2015-12-16:
Hi xDaya,

Did you try it on UNO board?

BrahMOs94, 2016-01-02:
hello Mr Lee,
I am having the problem which i stated before till now!
Can you please help me by responding quickly!
With Regards

ArduCAM, 2016-01-04:
Sorry for the late reply.
I have update the example sketch and host app for 4 5MP camera demo.
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_OV5642_4Cams_Burst 
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_4CAM.exe 
Let me know if you need more help.

admin@arducam.com

BrahMOs94, 2016-01-04:
Thanks for the program and host application Mr.Lee,
It works perfectly fine.
But the thing is even the lowest resolution 320 x 240 is buffering when I
gave autocapture with the interval of 1000 milliseconds!
Can't i improve the performance of ArduCAM
And this host application is only supporting 32bit pc.
It would be helpful if I get the same application for 64 bit pc.
Thanks and waiting for your reply!
With Regards,

ArduCAM, 2016-01-04:
Hi again,
Yes, the bottle neck of the performance is the serial baud rate, it is limited to 921600bps on both Arduino and PC app.
In this case you have to delay enough time between sending two capture commands, or use fast MCU and use other interface like USB or Ethernet.
I'm using Win10 64bit with our app, there is no problem, what do you mean the app can't support 64bit pc?
Let me know if you need more help.
Regards,
Lee

admin@arducam.com

BrahMOs94, 2016-01-05:
Hello Lee
Sorry the problem is with my PC and the code is working fine. I am not able
to understand the complete sense of the program. Can you send me detail
description or explanation of the program. So that I can understand the
flow of the program.
Thank you.
Hi again,
Yes, the bottle neck of the performance is the serial baud rate, it is
limited to 921600bps on both Arduino and PC app.
In this case you have to delay enough time between sending two capture
commands, or use fast MCU and use other interface like USB or Ethernet.
I'm using Win10 64bit with our app, there is no problem, what do you mean
the app can't support 64bit pc?
Let me know if you need more help.
Regards,
Lee

admin@arducam.com

ArduCAM, 2016-01-05:
Hi BrahMOs94
Thanks for your questions. Would you please provide your email address.
And I would like to write a post about the working follow of the demo code and host application. I think it will help other fans as well.
Lee

admin@arducam.com

BrahMOs94, 2016-01-05:
Yes
adithya.ramgopal @gmail.com
And thank you Mr.Lee

BrahMOs94, 2016-02-03:
Hi Lee,
I asked for a detailed description of the ArduCAM program.
Can you provide me soon??
Thanks.

ArduCAM, 2016-02-03:
Hi BrahMOs94
Sorry for my late update. We will be on Chinese New Year (Monkey Year) holiday from Feb. 6th~Feb.15th.
Did you read this document yet?
http://www.arducam.com/downloads/shields/ArduCAM_Camera_Shield_Software_Application_Note.pdf 
http://www.arducam.com/software/

BrahMOs94, 2016-02-03:
No I didn't ! I'll have a look at this
Thanks

BrahMOs94, 2016-02-12:
hello Lee,
Can you provide me  information about the  ARDUCAM  schematics and the ISP
used ?
Thanks!

On Wed, Feb 3, 2016 at 2:58 PM, Adithya Ramgopal <adithya.ramgopal@gmail.com

ArduCAM, 2016-02-14:
As you can see the we use the build in ISP in the image sensor.
You can find more information from the OV5642 datasheet:  www.arducam.com/downloads/modules/OV5642/OV5642_V1.11.pdf