ArduCAM/Arduino #8: The ESP8266 wiring on the blog example are wrong

sumotoy, 2015-11-29:
I've try to replicate your example on Blog
http://www.arducam.com/arducam-supports-esp8266-arduino-board-wifi-websocket-camera-demo/
The wiring for SDA and SCL are actually wrong, follow you example cannot work.
SDA should be D2 and SCL should be D1 and will works.
To use with the hardware connection showed in the example (without fixing wires), you must modify Wire.begin(3,4);

ArduCAM, 2015-11-30:
ESP8266 has different firmware version, I use the generic one, not the Adafuit or some others. 
So the pin configuration may be different, you have to check the dedicate pins for I2C and SPI according to your fimware.
 
-----------------------------------------------------
 
From: max mc costa
Date: 2015-11-29 23:53
To: ArduCAM/Arduino
Subject: [Arduino] The ESP8266 wiring on the blog example are wrong (#8)
I've try to replicate your example on Blog
http://www.arducam.com/arducam-supports-esp8266-arduino-board-wifi-websocket-camera-demo/
The wiring for SDA and SCL are actually wrong, follow you example cannot work.
SDA should be D2 and SCL should be D1 and will works.
To use with the hardware connection showed in the example (without fixing wires), you must modify Wire.begin(3,4);

sumotoy, 2015-11-30:
In the example you are using Arduino ESP8266.
Take a look in variants/generic/pins_arduino.h
static const uint8_t SDA = 4;
static const uint8_t SCL = 5;
This for all devices!

Bitbanger1900, 2015-12-21:
Hi,

I also try to replicate this example with an Arduino ESP8266 and a OV5642. How do you manage to solve the String.h not found error @compile time?

Arduino/libraries/ESP8266-Websocket/WebSocketServer.h:48:20: fatal error: String.h: No such file or directory
 #include "String.h"
                    ^
compilation terminated.
Error compiling.

Bitbanger1900, 2015-12-22:
Hi,

I am able to replicate this example with an OV2640. When i replace the OV2640 with an OV5642 and modify the camera detect it does find the Camera but i always get an SPI1 interface error and i
also have no image when i capture.

What causes the SPI1 interface error with the OV5642 what it does not do with the OV2640. It is exactly the same wired.

ArduCAM, 2015-12-23:
I think it should the same as OV2640 camera. Can you run the OV5642 camera with Arduino UNO without any problem?
You have to make sure there is no problem with hardware first. I will try to interface OV5642 camera with ESP8266 tomorrow.

Bitbanger1900, 2015-12-23:
Hi,

I did not try to run the OV5642 whith an Arduino UNO as i run Linux on my systems and there is only PC console software for Microsoft. 
You are right. I did not think about the possibility the CAM  can be broken as it is detecting it fine.
I'll try as soon as possible.

Bitbanger1900, 2016-01-08:
Hi,

I can confirm the hardware is working. I tried the OV5642 with an Arduino UNO and the only thing i needed to do was set the speed to 115200.  I could emulate your .exe with Wine under linux without problems.

No SPI  interface errors seen so it seems to be the combination OV5642 and NodeMcu ESP8266. note that i am running a NodeMcu V3.

hieudini, 2016-01-28:
Hi,

I was able to get the OV5642 working with an ESP8266 v1.0 NodeMCU (ESP-12e) module.  There is a slight discrepancy on the website example for the I2C connections.  Make sure your connections are consistent with the definitions in the pins_arduino.h file.  

The SPI issues appear to be due to a 1-bit left rotation in the data returned from the SPI.transfer() function.  After compensating for this, I was able to capture and transfer images from the OV5642.

Not sure where the rotation is happening.  Could it be in the ArduCAM mini camera?  SPI version read back gives 0x43.

Bitbanger1900, 2016-01-29:
Hi Hieudini,

Would you be so kind to share some code how you did that 1-bit left rotation?

hieudini, 2016-01-29:
You will have to rotate to the right by 1 bit.  Modify the uint8_t SPIClass::transfer(uint8_t data) function in the .arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/SPI/SPI.cpp as follows:

`uint8_t SPIClass::transfer(uint8_t data) {
    while(SPI1CMD & SPIBUSY) {}
    // reset to 8Bit mode
    setDataBits(8);
    SPI1W0 = data;
    SPI1CMD |= SPIBUSY;
    while(SPI1CMD & SPIBUSY) {}
    uint8_t val = SPI1W0 & 0xff;
    val = (byte)(val >> 1) | (val << 7); // correction for bit rotation from readback
    return val;
    //return (uint8_t) (SPI1W0 & 0xff);
}`

ArduCAM, 2016-02-01:
Thank you hieudini for the workaround.
I found the SPI timing is different between AVR based microprocessor and ARM based microporcessor.
I think I have to lower the SPI speed and modify the firmware in the 5MP camera to make both microprocessor compatible.

admin@arducam.com
 
From: hieudini
Date: 2016-01-30 05:33
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] The ESP8266 wiring on the blog example are wrong (#8)
You will have to rotate to the right by 1 bit. Modify the uint8_t SPIClass::transfer(uint8_t data) function in the .arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/SPI/SPI.cpp as follows:
uint8_t SPIClass::transfer(uint8_t data) {
while(SPI1CMD & SPIBUSY) {}
// reset to 8Bit mode
setDataBits(8);
SPI1W0 = data;
SPI1CMD |= SPIBUSY;
while(SPI1CMD & SPIBUSY) {}
uint8_t val = SPI1W0 & 0xff;
val = (byte)(val >> 1) | (val << 7); // correction for bit rotation from readback
return val;
//return (uint8_t) (SPI1W0 & 0xff);
}

steelgoose, 2016-04-05:
I recently purchased an OV5642 to duplicate the "ArduCAM now Supports ESP8266 Arduino Board with WIFI Websocket Camera Demo" on a "Adafruit Feather HUZZAH ESP8266". I modified the "ArduCAM_Mini_OV2640_websocket_server" sketch for the OV5642. 

When trying to run the sketch I receive the following error message...
Header mismatch
Disconnecting client
Terminating socket

Has anyone got a sketch to run this demo using the OV5642 on the Adafruit ESP8266?

ArduCAM, 2016-04-06:
Yes, you can modify the following example for OV5642 camera, it is much better. We will also post a new example for OV5642 camera soon.
Please use example https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture 
and modify the ESP8266 core library with following changes from folder ..\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0 which normally located on C driver.
//--------------------------------------------------
[WiFiClient.h]
#define WIFICLIENT_MAX_PACKET_SIZE 2048
 
[ClientContext.h]
add the line 
_pcb->snd_buf = TCP_SND_BUF;
before
size_t room = tcp_sndbuf(_pcb);
 
[lwipopts.h]
#define TCP_MSS                         2048

//--------------------------------------------------
Then use the html page to control the camera:
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture/html/video.html 

admin@arducam.com

Bitbanger1900, 2016-04-06:
Hi,

I tried the new example and modified the Camera Check so it can be used with an OV5642 and my NodeMCU.
I also modified the files mentioned above.

Problem what i have is that when i'll fetch the image with wget and try to open it i get the message: Error interpreting JPEG image file (Not a JPEG file: starts with 0xff 0xb1).

when i go to the http://IP/capture with firefox i got the message the image cannot be displayed because it contains errors.

I used the modified SPI.cpp with the 1 bit right rotation to get rid of the SPI interface error.

Please help.

steelgoose, 2016-04-07:
I followed your instructions. At first I had a problem when running the camera off of the 3.3 board power supply, it keep shutting the board down. After switching it to a dedicated power supply everything seem to work except I had no pictures. Tried the 1 bit rotation and got the SPI interface error.

Transcript:
Connecting to westell5413
....WiFi connected
10.0.0.12
Server started
CAM Capturing
capture total_time used (in miliseconds):39
CAM Capture Done!
Size is 0.
send total_time used (in miliseconds):0
CAM send Done!

ArduCAM, 2016-04-07:
@Steve, Size is 0 means there is some problem with the sensor, I'm afraid that the sensor is not well initialized. Please use the predefined settings before normal operation.

steelgoose, 2016-04-07:
Which predefined settings are you talking about? I pasted all code in the example for the OV5642, I am not sure what I am missing or what setting need to be changed.

Bitbanger1900, 2016-04-07:
Hi Steelgoose,

I see in some init functions:
myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
for the
myCAM.clear_fifo_flag();

steelgoose, 2016-04-07:
HUZZAH... you got it... Thank you Bitbanger1900. As I mention before, the 3.3v on the board does not have enough power, the OV5642 drawing power off the board causing the board lock up. I am using a "Adafruit Feather HUZZAH ESP8266" which has a 3.7v lipo connector. I plug the lipo into the board and connected the OV5642 to the Bat pin. The OV5642 in running off of lipo voltage, 3.7v to 4.2v.

I disconnect the usb and it still work correctly off the 3.7v lipo over wifi. HUZZAH...

This is my Int. setup:

 //Check if the camera module type is OV5642
  //myCAM.wrSensorReg16_8(0xff, 0x01);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
  if((vid != 0x56) || (pid != 0x42))
    Serial.println("Can't find OV5642 module!");
  else
    Serial.println("OV5642 detected.");

  //Change to JPEG capture mode and initialize the OV5642 module
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  //myCAM.OV5642_set_JPEG_size(OV5642_320x240);
  //myCAM.OV5642_set_JPEG_size(OV5642_640x480);
  myCAM.OV5642_set_JPEG_size(OV5642_1280x720);
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
  //myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);

Bitbanger1900, 2016-04-08:
Servus Steelgoose,

Glad that i could help. Nice picture by the way :).

Just curious. did you use the https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture ? As this example produces in my setup
a corrupt jpg image. 
I hope the specialists can help me here as i am running out of ideas what this could be.

steelgoose, 2016-04-08:
Yes, the ...OV2640_Capture, is the one I used with updates shown above. Had a lot of problems with the amperage draw until I figured that one out, did a lot of crazy things. The board has only about 250mA usable. The OV5642 draws  more current than the OV2640. I had also had to remove duplicate header files from the documents/arduino/library folder that I had installed. 

I bought a couple more boards (NodeMCU), I will experiment with them, but they should all be the same firmware so I don't expect any problems... 

I can just plug the lipo into the HUZZAH ESP8266 and use my android phone to capture the picture.

The puppy is my best buddy, she is like a security system with super sensors and night vision...

Bitbanger1900, 2016-04-08:
Thanks! good to know that it is working with the OV5642. Then i must have a weird thing with the
firmware on mine. When i use my OV2640 it is working but when i simply replace the init for the OV5642 i get a corrupt image and i always have to do that one bit rotation thing.

Should i be possibe to flash the firmware? :)

steelgoose, 2016-04-08:
Try this, add your ssid.
https://www.dropbox.com/sh/cnikj9q550a8t2t/AACi-BHsnTkIo773Fu5qjcGxa?dl=0

Bitbanger1900, 2016-04-08:
Hi Steelgoose,

Thanks for sharing.

Got the same error. my jpg is always starting with:
00000000   FF B1 FF C1  00 20 94 8C  92 8C 00 02  02 02 00 00

steelgoose, 2016-04-08:
I don't know what to tell you, that is beyond my knowledge base... I will wire up my new board tonight with the OV5642 and see what happens.

steelgoose, 2016-04-09:
Well Bitbanger1900, I have bad news and I have good news... I connected the OV5642 to my new ESP8266 ESP-12E board. The bad news is that it does not work, I get a partial picture (like the first packet or something) and then the rest of it is gray-out (no picture). When the lipo was disconnected or low voltage (around 2.6v) I get the SPI error. Plug a charged lipo in and the SPI error went away.

The good news is, it is probably your board... if you call that good news. Good news as it narrows the problem down. I don't think the boards is bad, It may be that the board can't keep up with the speed of the camera processor. When I had the OV5642 on the HUZZAH ESP8266 taking a 1280x720 jpg, it displayed really fast (about a second). The new board  was considerably slower. It only displayed about half the picture taking a 320x240 jpd. I suppose they could work if the speed was slowed down... if that is the problem... might be something else.

These boards are normally used to turn on and off switches and retrieving sensor readings. Processing pictures takes considerable more processing power. I like the HUZZAH because it has a 2.7v lipo plug, a builtin charger for the lipo and you could connect a solar panel to charge it during the day, all for about $16.00... if you can find one. 

I have another board coming in tomorrow, but I think it the same as the one I just got. I also bought a OV2640 that will be delivered Sunday. The OV5642 appears to be a good test tool for testing out the ESP8266 boards. If I find another ESP8266 similar to the HUZZAH, I will buy it just to test it out.

Bitbanger1900, 2016-04-09:
Hi Steelgoose,

Thanks for testing! Glad that it is weekend :) much time to play around.

I bought three esp8266 12E boards with all the same result when i try with the 5642.
When i connect the 2640 it is working.
When i connect the 5642 and use the websocket demo it is working so i have 
more the feeling that this is software related and not hardware.

If i use the OV5642 i always have to do the one bit rotatition modification in the SPI.cpp.
The header is always starting with  FF B1 FF C1 00 20 94 8C 92 8C 00 02 02 02 00 00 
(using the capture demo). the header should start with D8 instead of B1. When i convert this back to binary it looks like first bit is missing what could be caused by that one bit right rotation in the SPI.CPP.  (D8 = 1101 1000 and B1 = 1011 0001).

I would say that when i add a trailing 1 to the buffer the image is correct again.
The only bad thing is that i am a not a software engineer and have no knowledge of 
C programming how to do this :(

Do you also have to do the one bit right rotation to get rid of the SPI interface error? As 
if there are OV5642's with different firmware on it.

steelgoose, 2016-04-09:
Interesting, I think this is a job for "Super Lee"... Every time I did the one bit rotation I got the SPI error and had to revert back. I wonder is there is some kind of firmware update for the esp8266 12E, or if that would even help.

When I first started with the OV5642 I was using the websocket demo but keep getting a Header mismatch error. What to I need to do to get the use the websocket demo?

Bitbanger1900, 2016-04-09:
yeah Super Lee! we can use your help here as this is getting more weird. Steelgoose and
i are using the same camera and have complete different results running the same example.

@steelgoose,
- I changed the init to use the OV5642 instead of the OV2640
- added  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
just after setting the resolution.
- modified to use the SPI.CPP to do that one bit right rotation 
(used the NodeMcu ESP8266-12E)

steelgoose, 2016-04-09:
Super Lee; with the OV5642, same example, on the Adafrute Feather ESP8266 ESP-12E board, I get a great picture, but on the NodeMCU ESP8266 ESP-12E board, It will only print the first packet of the image. What do you think it is?

steelgoose, 2016-04-10:
Bitbanger1900, What do I need to do to get the "ArduCAM_Mini_OV2640_websocket_server" example working. I am not familiar with setting up websockets or how this example works.

Bitbanger1900, 2016-04-18:
Hi Steelgoose.

Actually it is straitforward as the manual from Lee. Open the websocket example and modify it to use it with the OV5942. Then open the camera_demo.html in the corresponding html folder and enter the IP of the camera in the location part.
Then press the capture button and see the result :)

steelgoose, 2016-04-19:
Dahhhhh!... Thanks

Did you ever get your camera issue resolved?

Bitbanger1900, 2016-04-19:
Hi,

Nope. never got a feedback from Lee about that problem :(

There is another github project  what uses esp-open-rtos to run a 2640 directly on the esp8266
https://github.com/kanflo/esparducam

I'm going to give that a try and see if that runs directly on the esp8266 without the NodeMCU. I also
ordered a few of those boards but it will take some weeks to arrive.

steelgoose, 2016-04-19:
That's cool, I will have to try that when I have  more time. Who did you order your boards from?

I would like to connect one to GSM and send it to my cell phone...

ArduCAM, 2016-04-27:
@Bitbanger1900 We have update our library for ESP8266 board, please check the latest library and examples from http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/

Bitbanger1900, 2016-04-27:
Hi Lee,

Thanks! i ordered today that bord and can't wait to try it!

In the meantime i installed the latest Arducam libraries and try to compile ArduCAM_ESP8266_OV5642_Capture.ino example for the NodeMCU-1.0 but
it is failing to compile with the error:

ov5642-new.ino.ino: In function 'void setup()':
ov5642-new.ino:206: error: 'OV5642_CHIPID_HIGH' was not declared in this scope
ov5642-new.ino:207: error: 'OV5642_CHIPID_LOW' was not declared in this scope
'OV5642_CHIPID_HIGH' was not declared in this scope

Any idea where this causes?

ArduCAM, 2016-04-28:
@Bitbanger1900 Please read the document here for detail: http://www.arducam.com/downloads/ESP8266_UNO/ArduCAM_ESP8266_UNO_DS.pdf

steelgoose, 2016-04-28:
@Bitbanger1900 Check out this ESP8266 Wifi Range/Distance Tests. I will have to do some range testing with the ArduCAM... An inexpensive battery powered wildlife or security cam...
https://www.youtube.com/watch?v=7BYdZ_24yg0

Bitbanger1900, 2016-04-28:
Hi Lee,

I managed to compile the ArduCAM_ESP8266_OV5642_Capture.ino example for the NodeMCU.
The solution is to dowload the source for the ArduCAM_ESP8266_UNO and after that copy the ArduCAM library to the esp8266 library subdirectory... I can imagine that there is also another solution to get this working? At least the ArduCAM sources are not the same what is causing the compile error!

After uploading to the NodeMCU the image what i get is not okay.
[image]

ArduCAM, 2016-04-29:
@Bitbanger1900 , it is the know issue. Please use our new board library package for esp8266 from the instruction in post: http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/

MMIX2009, 2016-07-21:
Hi,
Is it possible to use the ESP8266 on the ArduCAM ESP8266 UNO as a software serial device? If so, what are the TX and RX pins?