ArduCAM/Arduino #11: Problem with ArduCam (OV5642) !!!!!!

shravanib, 2015-12-16:
Dear Sir,
I purchase arducam OV5642 last week.
Since the time I connected it, I keep on getting the "SPI Interface Error" evrytime I run the Host application for PC console.
I have not been able to see any image whatsoever.

I am new to arduino and thus wanted to know what does the error exactly mean? What is the actual problem? Is something wrong with the Arducam or the code or the arduino board itself (Uno R3).
I have tried all the connections and solutions you provided previously in the forum with no luck.

I would be highly obliged if you could please help me solve this problem.
Thank you.

ArduCAM, 2015-12-16:
"SPI Interface Error" means there is something wrong with the SPI interface between the camera and Arduino board.
The example sketch will detect if SPI interface is OK before normal operation.
The normal cause of this error, is incorrect SPI signals wiring especially the CS pin. It should be the same as defined in the example sketch.
So please check your CS pin wiring and CS pin definition in the example sketch.

admin@arducam.com
 
From: shravanib
Date: 2015-12-16 20:07
To: ArduCAM/Arduino
Subject: [Arduino] Problem with ArduCam (OV5642) !!!!!! (#11)
Dear Sir,
I purchase arducam OV5642 last week
Since the time I connected it, I keep on getting the "SPI Interface Error" evrytime I run the Host application for PC console
I have not been able to see any image whatsoever
I am new to arduino and thus wanted to know what does the error exactly mean? What is the actual problem? Is something wrong with the Arducam or the code or the arduino board itself (Uno R3)
I have tried all the connections and solutions you provided previously in the forum with no luck
I would be highly obliged if you could please help me solve this problem
Thank you

htr99, 2015-12-16:
Hi admin,
I get SD card error when try to open file. I use 4 cam adapter board + Arducam mini 2MP + SD card. When try to write SD crd without using camera code it works fine.

Thank you.

ArduCAM, 2015-12-16:
would you please let me know which the SD card CS pin is used?
You can not enable more than one CS form SD card or cameras at a time, they all share the SPI bus.

admin@arducam.com
 
From: htr99
Date: 2015-12-16 23:41
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] Problem with ArduCam (OV5642) !!!!!! (#11)
Hi admin,
I get SD card error when try to open file. I use 4 cam adapter board + Arducam mini 2MP + SD card. When try to write SD crd without using camera code it works fine.
Thank you.

htr99, 2015-12-16:
I am your example code named as "ArduCAM_Mini_2MP_4Cams_TimeElapse2SD.ino" and 4 camera shield.

htr99, 2015-12-16:
[image]

ArduCAM, 2015-12-16:
Please be patient. Did you use the example sketch ArduCAM_Mini_2MP_4Cams_TimeElapse2SD without modify anything?
If yes I can try it again here tomorrow and get back to you.

admin@arducam.com
 
From: htr99
Date: 2015-12-16 23:51
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] Problem with ArduCam (OV5642) !!!!!! (#11)

htr99, 2015-12-16:
yes with no modification. I'm using arduino UNO. I'm in a hurry because this should be presented within tomorrow. Pleae try it on UNO board. 

This is information ,

I connected 2MP cams on slot 1 and 2.
Images are shown on PC when streaming is done.
SD card write read is OK when it is tested individually.

After trying many things I have doubt that, is it  dynamic memory limitation issue. But I do not have mega board for testing.

Please try to test as soon as possible because I really nee this soon.
Thank you.

ArduCAM, 2015-12-17:
I have update the examples today, please try it out. Let me know if you still have problem with it.
ArduCAM_Mini_2MP_4Cams_TimeElapse2SD
ArduCAM_Mini_2MP_4Cams_TimeElapse2SD_LowPower 

admin@arducam.com
 
From: htr99
Date: 2015-12-17 00:36
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] Problem with ArduCam (OV5642) !!!!!! (#11)
yes with no modification. I'm using arduino UNO. I'm in a hurry because this should be presented within tomorrow. Pleae try it on UNO board. 
This is information ,
I connected 2MP cams on slot 1 and 2.
Images are shown on PC when streaming is done.
SD card write read is OK when it is tested individually.
After trying many things I have doubt that, is it dynamic memory limitation issue. But I do not have mega board for testing.
Please try to test as soon as possible because I really nee this soon.
Thank you.

htr99, 2015-12-17:
OK thanks, I will try and tell you the result.

shravanib, 2015-12-18:
Respected Sir,

I made the necessary changes and reviewed my connections. 
I have attached all the pins correctly.
But now I am facing another problem. 
After attaching the camera and starting the host application, I get a dialog box saying, "A device attached to the system is not functioning".
Also I checked and found out that SCK is not toggling. It is showing a voltage of '0' constantly with respect to the ground. 
Please do help with this problem.

Thanks and regards.

htr99, 2015-12-19:
Hi lee,

Any idea why images are not clear as shown in figure? Please tell me any tip to solve the issue.

Thank you

ArduCAM, 2015-12-19:
Hi htr99
From the photo, it seems that the JPEG data is corrupted.
Please connect one camera and use ArduCAM_Mini_OV2640_Video_Streaming example together with ArduCAM_host_v1.0.exe to check out if hardware has any problem.
The JPEG data from the camera should be well writen to the SD card, it can't be stopped when writing the SD card or else the data will be lost.

htr99, 2015-12-19:
Hi Lee, 
I actually could be able to solve the issue before just now. I did few modifications to the code.
I will post the code soon with mentioning changes. 
Thank you for your help.

ArduCAM, 2015-12-19:
Hi shravanib,
 "A device attached to the system is not functioning". means your Arduino board USB hang up.
It is the common case you hot plug the camera into the Arduino board. It is recommend to plug the camera to Arduino board first then power up the board.
Please remove the camera to check out if the Arduino board's SCK is toggling, make sure your hardware is no problem.
Lee

admin@arducam.com
 
From: shravanib
Date: 2015-12-18 19:54
To: ArduCAM/Arduino
CC: Lee
Subject: Re: [Arduino] Problem with ArduCam (OV5642) !!!!!! (#11)
Respected Sir,
I made the necessary changes and reviewed my connections. 
I have attached all the pins correctly.
But now I am facing another problem. 
After attaching the camera and starting the host application, I get a dialog box saying, "A device attached to the system is not functioning".
Also I checked and found out that SCK is not toggling. It is showing a voltage of '0' constantly with respect to the ground. 
Please do help with this problem.
Thanks and regards.

shravanib, 2015-12-21:
Respected Sir,
Thank you so much for all the help. The camera is working perfectly.

Regards.

JTurcott, 2016-12-09:
I know this is an old post(almost a year), however I see shravanib reported the issues resolved and I would like to get the working sketch along with correct devices (UNO, Mega etc.)  and pinouts they used.  Is this still available and/or did I overlook these?  Lastly, will this work with the 0V5642 if I include its libraries and define statement etc?  I've had the OV5642 for about 5 months and haven't found a complete article with working sketch(es) and I have the UNO, Mega, ESP8266, WeMos family etc ready to connect.

ArduCAM, 2016-12-20:
@JTurcott, I'm not sure what kind of OV5642 module you got. You need ArduCAM-Mini-5MP (OV5642) module to work directly with UNO, Mega, ESP8266. The OV5642 breakout module doesn't work with these board, it only work with ArduCAM shield.

rpierce141, 2017-03-30:
I have and Arducam Mini Module XOOOVGJ88J, OV5642 and am looking for the pin outs and sketches.  Are these available still?

AlexCivit, 2017-04-27:
Respected colleague @htr99 I'm now having the same problem that you had on the previous post, I'm getting corrupted pics from my ov5642. How did you solve that problem? It would be so kind from you if you post the code to solve it.

Thanks!

diegoto, 2017-05-22:
Good to all, I have a Arducam mini OV 5642, on the monitor I get camera connected, sd connected, capture done, ok.
I save the photos in the sd, but when I go to see them in my laptop I can not see them, every photo has a size of 45KB 48KB, someone has happened to this, I need help

diegoto, 2017-05-22:
solucionado

shakedma, 2017-05-24:
I purchase the arducam mini 5 Mp and try for a long time to run it with no success.
should it run from the IDE or should i download another software?
i was success with the wire configuration, but when i run it is show the next text:

ACK CMD ArduCAM Start!
ACK CMD SPI interface OK.
ACK CMD OV5642 detected.

and then when i hit the 0 to start shooting it start to write garbage

please advise