ArduCAM/Arduino #12: SPI Error with ArduCam-Mini and Arduino Due

alistairsmarter, 2015-12-17:
Hello,

I am receiving "SPI Interface Error!" when using the ArduCam-Mini with Arduino Due. Can you please confirm the correct pin configuration for this board and if there are any known software issues?

I am currently using:
SCL1
SCA1
GND
+5V
SPI Header SCK
SPI Header MISO
SPI Header MOSI
Pin 10

Many thanks

Oitzu, 2015-12-22:
@alistairsmarter which example are you using?
I'm asking because "Pin 10" can be configured in the sketch itself.

picrust37, 2016-03-02:
You should try using the Vcc and Ground SPI Header pins.

If that doesn't fix it, I would suggest checking the line that has Serial.begin. I had a software issue where the code had Serial.begin(921600).  I changed this to Serial.begin(115200) and then I was able to make it work.