ArduCAM/Arduino #13: Arducam mini Rev. B

mayhanturk, 2016-01-07:
Hii,
I have an arducam mini 2Mp Rev B and uno R3 board. I made pin connections from arduino 10>scl to arducam cs>scl pins (except vcc to 5V)
With Arducam_V1.exe application, I set the resolution 320x240 and baud rate 921600, and upload the ArduCAM_Mini_OV2640_Video_Streaming example in this repository.

But I couldn't take a valid image. When I capture a single shot, "OV2640 detected, Cam started single shot, Cam capture done", but take an error "A generic error occurred in GDI+".  I saved the image and analyze the hex format of the image. Jpeg markers seems ok, but it is not a valid image. Captured file is in the attachment.

[image]

Could u help me please?
Thanks,
myhn

mayhanturk, 2016-01-07:
Hii,
Finally found the solution!

The problem was baud rate mismatch. Baud rate higher than 115200 created problem on my windows application capture (I didn't try sd card saving).

Best regards.
myhn

closed 2016-01-07 by mayhanturk

ArduCAM, 2016-01-08:
Hi Myhn,
You are awesome , resolve this problem by yourself.
It is common issue, that higher baudrate will cause higher bit error rate.
It depends on your Arduino board USB-serial controller chip, our 16U2 on Uno board works fine at 921600bps. But it is more stable to use lower baudrate like 115200bps.
Let me know if you need more help.
Regards,
Lee