ArduCAM/Arduino #14: Take a snapshot and analyze rgb data

xseignard, 2016-01-10:
Hello,

I would like to collect RGB data from a snapshot. But I don't get on how to access this data, without storing it or whatever.

Any idea? Example on where to start?

Best regards,

Xavier

ArduCAM, 2016-01-10:
Hi Xavier,
You can capture QVGA RGB format image using example https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/REVC/ArduCAM_OV2640_Camera_Playback 
You don't need to save the image to file, because after capture the data is located in onboard frame buffer, you can analyze the rgb data by reading and processing from the frame buffer.
The RGB data is organized in RGB565 format, please read this post for detail: http://www.arducam.com/rgb565-format-issues/

Mausisagooddog, 2016-03-17:
I an trying this using a Mega board and the 5MP mini cam. 'class ArduCAM' has no member named 'set_mode'
Do I have the library lost on my computer or is it not included for the 5MP camera?

ArduCAM, 2016-03-17:
@Mausisagooddog Can you send the compile error message?

Mausisagooddog, 2016-03-18:
Arduino: 1.6.7 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Gudrun\Documents\Arduino\ArduCAMRGB565demo\ArduCAMRGB565demo.ino:53:1: warning: narrowing conversion of '240' from 'int' to 'const char' inside { } [-Wnarrowing]

 };

 ^

C:\Users\Gudrun\Documents\Arduino\ArduCAMRGB565demo\ArduCAMRGB565demo.ino:53:1: warning: narrowing conversion of '196' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Gudrun\Documents\Arduino\ArduCAMRGB565demo\ArduCAMRGB565demo.ino:53:1: warning: narrowing conversion of '196' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Gudrun\Documents\Arduino\ArduCAMRGB565demo\ArduCAMRGB565demo.ino:53:1: warning: narrowing conversion of '248' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Gudrun\Documents\Arduino\ArduCAMRGB565demo\ArduCAMRGB565demo.ino:53:1: warning: narrowing conversion of '224' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Gudrun\Documents\Arduino\ArduCAMRGB565demo\ArduCAMRGB565demo.ino: In function 'void setup()':

ArduCAMRGB565demo:88: error: 'class ArduCAM' has no member named 'set_mode'

   myCAM.set_mode(MCU2LCD_MODE);

         ^

C:\Users\Gudrun\Documents\Arduino\ArduCAMRGB565demo\ArduCAMRGB565demo.ino: In function 'void loop()':

ArduCAMRGB565demo:121: error: 'class ArduCAM' has no member named 'set_mode'

   myCAM.set_mode(CAM2LCD_MODE);     //Switch to CAM

         ^

ArduCAMRGB565demo:126: error: 'class ArduCAM' has no member named 'get_bit'

     if(!myCAM.get_bit(ARDUCHIP_TRIG,VSYNC_MASK))        //New Frame is coming

               ^

ArduCAMRGB565demo:128: error: 'class ArduCAM' has no member named 'set_mode'

        myCAM.set_mode(MCU2LCD_MODE);      //Switch to MCU

              ^

ArduCAMRGB565demo:130: error: 'class ArduCAM' has no member named 'set_mode'

        myCAM.set_mode(CAM2LCD_MODE);      //Switch to CAM

              ^

ArduCAMRGB565demo:131: error: 'class ArduCAM' has no member named 'get_bit'

        while(!myCAM.get_bit(ARDUCHIP_TRIG,VSYNC_MASK));   //Wait for VSYNC is gone

                     ^

ArduCAMRGB565demo:133: error: 'class ArduCAM' has no member named 'get_bit'

     else if(myCAM.get_bit(ARDUCHIP_TRIG,SHUTTER_MASK))

                   ^

ArduCAMRGB565demo:136: error: 'class ArduCAM' has no member named 'get_bit'

        while(myCAM.get_bit(ARDUCHIP_TRIG,SHUTTER_MASK))

                    ^

ArduCAMRGB565demo:148: error: 'class ArduCAM' has no member named 'set_mode'

          myCAM.set_mode(MCU2LCD_MODE);      //Switch to MCU, freeze the screen 

                ^

C:\Users\Gudrun\Documents\Arduino\ArduCAMRGB565demo\ArduCAMRGB565demo.ino: In function 'void GrabImage(char*)':

ArduCAMRGB565demo:173: error: 'class ArduCAM' has no member named 'set_bit'

   myCAM.set_bit(ARDUCHIP_TIM, MODE_MASK);

         ^

ArduCAMRGB565demo:181: error: 'class ArduCAM' has no member named 'get_bit'

   while(!myCAM.get_bit(ARDUCHIP_TRIG,CAP_DONE_MASK));

                ^

ArduCAMRGB565demo:217: error: 'class ArduCAM' has no member named 'clear_bit'

   myCAM.clear_bit(ARDUCHIP_TIM, MODE_MASK);

         ^

C:\Users\Gudrun\Documents\Arduino\ArduCAMRGB565demo\ArduCAMRGB565demo.ino: In function 'void Playback()':

ArduCAMRGB565demo:226: error: 'class ArduCAM' has no member named 'set_mode'

   myCAM.set_mode(MCU2LCD_MODE);         //Switch to MCU

         ^

exit status 1
'class ArduCAM' has no member named 'set_mode'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
I admit, it could be I have the installation wrong, but I did get the demo programs to work, so it is a bit confusing. the Arduino environment really doesn't tell you where it is looking for libraries. Attachment is the programming I am compiling. Granted, this is your demo for the OV2640, but I was hoping there was an updated file for the newer camera, and I would have thought it  should still verify/compile even without a board plugged in. 
 I have been looking for the method used in the working programs where you set the camera to jpeg with  myCAM.set_format(JPEG); but I don't see anything like "RAW" or 565 or similar command in any of the libraries. My  goal is to pull any one of the horizontal lines into an array.

The program highlights the line: myCAM.set_mode(MCU2LCD_MODE);  when it attempts to compile.
Yes, I am new at the Arduino, I do appreciate the help. 

 Doug

ArduCAM, 2016-03-18:
@Mausisagooddog
I'm afraid that you placed the arducam libraries to the wrong folder.
You should put all the ArduCAM libraries directly under the \arduino-1.x.x\libraries folder.

Mausisagooddog, 2016-03-18:
That's a Windows 10 feature.

Mausisagooddog, 2016-04-03:
I have this successfully working, after manually putting the libraries where they were needed and removing the extension cables. I am attempting to work with a stereo setup, and with a slight mod to the program I can download and save photos to the pc using the arducam.exe example program, so for my prototype purposes, I would like to continue with this hardware. 
**Is there somewhere a formal document summary listing of all the libraries and their calls**? Deciphering a working program goes a long way, but it would be nice to know if, for instance, if I can even use a BMP file with addressable RGB565 data with the 5mb mini cam, and what program library calls I should be using and what the return formats will be. Such documents should also note when a command is valid and to which library version.

ArduCAM, 2016-04-04:
@Mausisagooddog
Please find more information and document from: http://www.arducam.com/arducam-mini-released/  
The api is described here:
http://www.arducam.com/downloads/shields/ArduCAM_Camera_Shield_Software_Application_Note.pdf

Mausisagooddog, 2016-04-18:
Lee, I'm stuck. Using original .exe and the newer .exe with bmp selection.  I got the cameras working as a pair with jpeg, but I am having difficulty with the BMP files. I am trying to use your example program and eliminate jpeg references, 
This program will work on either myCAM1 or myCAM3 (I am working toward a stereo camera) but only returns a full white BMP file. I am only trying 1 camera at a time. I put some 'Serial.write("x"); in various places to try to track program flow.
The program returns:
Port Opened.
????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????ArduCAM Start!
A
SPI interface Error!
OV5642 detected.
+++B
selection made
CAM start single shoot.
CAM start single shoot.
[*]FMODE 3 CAM Capture Done!
and continues delivering full white photos...
PROGRAM FOLLOWS:

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

#define BMPIMAGEOFFSET 66

// set pin 10 as the slave select for the digital pot:
const int CS = 10;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;

const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};

ArduCAM myCAM1(OV5642, CS);
uint8_t read_fifo_burst(ArduCAM myCAM1);

void setup() {
  // put your setup code here, to run once:
  uint8_t vid, pid;
  uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
#else
  Wire.begin();
#endif
  Serial.begin(115200);
  Serial.println("ArduCAM Start!");

  // set the CS as an output:
  pinMode(CS, OUTPUT);

  // initialize SPI:
  SPI.begin();
  //Check if the ArduCAM SPI bus is OK
  myCAM1.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM1.read_reg(ARDUCHIP_TEST1);
  //Serial.println(temp);
  Serial.println("A"); //
  if (temp != 0x55)
  
  {
    Serial.println("SPI interface Error!");
    //while(1);
  }

[... 221 lines cut ...]
  myCAM1.set_fifo_burst();//Set fifo burst mode
  SPI.transfer(0x00);//First byte is 0xC0 ,not 0xff
  length--;
  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.write(temp_last);
      Serial.write(temp);
      Serial.println("H");
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
      break;
    delayMicroseconds(12);
  }
  myCAM1.CS_HIGH();
  is_header = false;
}

ArduCAM, 2016-04-19:
Dear Mausisagooddog,
can you show me the bmp photo you have got? I think you need to wait for few seconds after switching from jpeg to bmp format, then start a capture.

Mausisagooddog, 2016-04-20:
Nothing to see but full white.

ArduCAM, 2016-04-20:
@Mausisagooddog, would you please send the white photos to admin@arducam.com ?

paekitt, 2016-12-14:
Hello,

Arduino: 1.6.13 (Windows 10), Board: "nodeMCU 1.0(ESP-12E Modue)"

C:\Program Files (x86)\Arduino\libraries\ArduCAM-master\examples\ESP8266\ArduCAM_Mini_OV2640_websocket_server\ArduCAM_Mini_OV2640_websocket_server.ino: In function 'void setup()':

ArduCAM_Mini_OV2640_websocket_server:59: error: 'class ArduCAM' has no member named 'begin'

   myCAM.begin();

         ^

ArduCAM_Mini_OV2640_websocket_server:71: error: 'OV2640_CHIPID_HIGH' was not declared in this scope

   myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);

                        ^

ArduCAM_Mini_OV2640_websocket_server:72: error: 'OV2640_CHIPID_LOW' was not declared in this scope

   myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);

                        ^

C:\Program Files (x86)\Arduino\libraries\ArduCAM-master\examples\ESP8266\ArduCAM_Mini_OV2640_websocket_server\ArduCAM_Mini_OV2640_websocket_server.ino: In function 'void read_fifo_burst(ArduCAM)':

ArduCAM_Mini_OV2640_websocket_server:184: error: 'class ArduCAM' has no member named 'startSend'

   myCAM.startSend();

         ^

ArduCAM_Mini_OV2640_websocket_server:186: error: 'class ArduCAM' has no member named 'sendData'

   myCAM.sendData(0x00);

         ^

ArduCAM_Mini_OV2640_websocket_server:191: error: 'class ArduCAM' has no member named 'sendData'

     temp = myCAM.sendData(0x00);

                  ^

ArduCAM_Mini_OV2640_websocket_server:221: error: 'class ArduCAM' has no member named 'endSend'

   myCAM.endSend();

         ^

exit status 1
'class ArduCAM' has no member named 'begin'