ArduCAM/Arduino #15: Distorted image when saving to SD Card

macoy-marc, 2016-01-17:
Hi, sir. good day. I have a problem regarding the OV2640 Arducam Mini. When capturing a snapshot and saved it to SD, I got an output image like this. I need help very urgent. I hope you can give some advice on my concern. Regards!
[image]

and btw, here is the code(modified a bit according to use)

#include <Adafruit_CC3000.h>
#include <ccspi.h>
#include <SPI.h>
#include <string.h>
#include "utility/debug.h"
#include "RTClib.h"
#include <SD.h>
#include <Wire.h>
#include <Thread.h>
#include <ArduCAM.h>
#include "memorysaver.h"

#define ADAFRUIT_CC3000_IRQ   3 
#define ADAFRUIT_CC3000_VBAT  5
#define ADAFRUIT_CC3000_CS    10
Adafruit_CC3000 cc3000 = Adafruit_CC3000(ADAFRUIT_CC3000_CS, ADAFRUIT_CC3000_IRQ, ADAFRUIT_CC3000_VBAT,
                                         SPI_CLOCK_DIVIDER);

#define WLAN_SSID       "noparking"          
#define WLAN_PASS       "qwer123asdf"
#define WLAN_SECURITY   WLAN_SEC_WPA2

#define IDLE_TIMEOUT_MS  3000
#define WEBSITE      "192.168.10.100"
#define WEBPAGE      "/DesignProject/pages/add.php"

#define SD_CS 4
#define BMPIMAGEOFFSET 66
#define SPI_CS 53
const int pirPin = 48;
uint32_t ip = 3232238180;

ArduCAM myCAM(OV2640,SPI_CS);
RTC_DS1307 rtc;

void setup() {
  uint8_t temp;
  uint8_t vid,pid;
  #if defined (__AVR__)
    Wire.begin(); 
  #endif
  #if defined(__arm__)
    Wire1.begin(); 
  #endif
  Serial.begin(115200);
  Serial.println("Program Start!\n");
  
  //Initialization of SD Card
  Serial.println(F("Initializing SD Card!"));
  if (!SD.begin(SD_CS)) {
    Serial.println(F("Initialization failed!\n"));
    while(1);
  }
  /*
  Serial.println(F("Initialization done!\n"));
  //Initializing RTC
  Serial.println(F("Initializating RTC!"));
  if (!rtc.begin()){
    Serial.println(F("Failed to Initialize RTC!\n"));
  }
  //rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
  Serial.println(F("Initialization of RTC sucessful!\n"));
  */
  pinMode(SPI_CS, OUTPUT);
  /*
  pinMode(ADAFRUIT_CC3000_CS, OUTPUT);
  digitalWrite(SPI_CS, HIGH);
  digitalWrite(ADAFRUIT_CC3000_CS, LOW); 
  */
  SPI.begin();
  //ARDUCAM
  Serial.println("ArduCAM Start!");
  // set the SPI_CS as an output:
  pinMode(SPI_CS, OUTPUT);
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
  {
    Serial.println("SPI interface Error!\n");
    while(1);
  }
  
  myCAM.write_reg(ARDUCHIP_MODE, 0x00);

  //Check if the camera module type is OV2640
  myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
  if((vid != 0x26) || (pid != 0x42))
    Serial.println("Can't find OV2640 module!\n");
  else
    Serial.println("OV2640 detected\n");

  //Change to BMP capture mode and initialize the OV2640 module  
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  myCAM.OV2640_set_JPEG_size(OV2640_1600x1200);
/*
  //Initializing CC3000(Wifi Shield)
  Serial.println(F("Initializating Wifi Shield!"));
  if (!cc3000.begin())
  {
    Serial.println(F("Couldn't begin()! Check your wiring?"));
    while(1);
  }
  
  //connect to SSID; if err, display failed
  Serial.println(F("Connecting..."));
  if (!cc3000.connectToAP(WLAN_SSID, WLAN_PASS, WLAN_SECURITY)) {
    Serial.println(F("Failed to Connect!"));
    while(1);
  }
  //else display connected
  Serial.println(F("Connected!"));
  
  //request for DHCP to find IP address
  //will be deleted after configuring router
  //Don't leave until IP address, DNS, and gateway is assigned
  Serial.println(F("Request DHCP"));
  while (!cc3000.checkDHCP())
  {
    delay(100);
  } 
  */
  
}

void loop() {
  char str[8];
  File outFile;
  byte buf[256];
  unsigned long previous_time = 0;
  static int i = 0;
  static int n = 0;
  static int k = 0;
  uint8_t temp, temp_last;
  int total_time = 0;
  //myCAM.write_reg(ARDUCHIP_MODE, 0x01);      //Switch to CAM
  //temp = myCAM.read_reg(ARDUCHIP_TRIG);
  if(digitalRead(pirPin)==HIGH)
  {
    //Flush the FIFO
    myCAM.flush_fifo();
    //Clear the capture done flag
    myCAM.clear_fifo_flag();
    //Start capture
    myCAM.start_capture();
    Serial.println("Start Capture");

    while(!(myCAM.read_reg(ARDUCHIP_TRIG) & CAP_DONE_MASK));

    Serial.println("Capture Done!");
    //Construct a file name
    k = k + 1;
    itoa(k, str, 10); 
    strcat(str,".jpg");        //Generate file name
    //myCAM.write_reg(ARDUCHIP_MODE, 0x00);      //Switch to MCU, freeze the screen 
    //GrabImage(str);

    //Open the new file
    outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
    if (! outFile)
    {
      Serial.println("open file failed");
      return;
    }
    total_time = millis();
    i = 0;
    myCAM.CS_LOW();
    myCAM.set_fifo_burst();
    temp = SPI.transfer(0x00);
    //
    //Read JPEG data from FIFO
    while ( (temp != 0xD9) | (temp_last != 0xFF))
    {
      temp_last = temp;
      temp = SPI.transfer(0x00);
  
      //Write image data to buffer if not full
      if (i < 256)
        buf[i++] = temp;
      else
      {
        //Write 256 bytes image data to file
        myCAM.CS_HIGH();
        outFile.write(buf, 256);
        i = 0;
        buf[i++] = temp;
        myCAM.CS_LOW();
        myCAM.set_fifo_burst();
      }
    }
    //Write the remain bytes in the buffer
    if (i > 0)
    {
      myCAM.CS_HIGH();
      outFile.write(buf, i);
    }
    //Close the file
    outFile.close();
    total_time = millis() - total_time;
    Serial.print("Total time used:");
    Serial.print(total_time, DEC);
    Serial.println(" millisecond");
    //Clear the capture done flag
    myCAM.clear_fifo_flag();
    }
}

ArduCAM, 2016-01-18:
Hi Sir,
I'm sorry that I can't see you attached photo, please send to my email admin@arducam.com 
I suggest you to test the camera on Arduino UNO R3 board before moving to any other platform, you have to make sure if the camera is working,
Read the fifo is a little bit tricky, there are two reading mode, one is single read and the other is burst read. I think you can try single read before trying the burst read mode.
Please read the hardware application note for detail : http://www.arducam.com/downloads/shields/ArduCAM_Mini_2MP_Camera_Shield_Hardware_Application_Note.pdf 
Regards,
Lee

macoy-marc, 2016-01-18:
Okay Sir, thanks. Mailed you the details.

rflamino, 2016-04-25:
Hi macoy-marc, did you solved your problem? I am facing the same issue. Can you post how to solve this? Thanks in advance!

macoy-marc, 2016-05-02:
Yes sir, my guess is just on the power side. Because we have many modules
attached to the arduino in just one supply including the ArduCam.So I
separated the Arduino and ArduCam onna dedicated power supply and the rest
of the modules on another power supply.