ArduCAM/Arduino #16: Cannot Compile "ArduCAM_Mini_OV2640_Video_Streaming"

mstoer, 2016-01-20:
I recently purchased an ArduCAM mini 2P OV2640 (Rev. B) and am using it on an Arduino UNO with v1.6.5 IDE.    I downloaded the libraries from here and installed them and tested compiled the "ArduCAM_Mini_5MP_TimeElapse2SD.ino" example.  It worked, but the images stored on the SD card were out of focus.  I tried to use the video streaming example to help adjust the focus but it will not compile. I get this error:

ArduCAM_Mini_OV2640_Video_Streaming.ino: In function 'void loop()':
ArduCAM_Mini_OV2640_Video_Streaming:178: error: 'read_fifo_burst' was not declared in this scope
'read_fifo_burst' was not declared in this scope

I can compile all other examples including the video streaming example for the OV5642.  Could you tell me how to get this to compile or which example would be best to use to help set up the camera using the host app?

Thanks,
Marcell

ArduCAM, 2016-01-20:
Hi, You are right. It is missing declaration of the fuction read_fifo_burst.
You can add a line void read_fifo_burst(ArduCAM myCAM); on top of the sketch to declare it.
It is not a problem in previous Arduino IDE, but now it doesn't allow to miss the declaration.
Thanks, I will update example soon.

admin@arducam.com

mstoer, 2016-01-20:
That works now.

Thank you,
Marcell