ArduCAM/Arduino #18: arducam mini pid

tzapu, 2016-01-26:
HI,

 I just bought an Arducam and myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid); seems to return 41 instead of 42 as in all the examples.

Any idea why that is?
Cheers

ArduCAM, 2016-01-27:
Hi Sir,
As you can see from the OV2640 datasheet the OV2640_CHIPID_LOW register (0x0B) is 0x41 not 0x42.
Let me know if you have more questions.

admin@arducam.com

tzapu, 2016-01-28:
hi, thanks for the fast answer, it works fine i think with 0x41
all your samples show 0x42 though fro OV2640_CHIPID_LOW
are they for another model?

ArduCAM, 2016-01-28:
I think it is a mistake in the code, I will update the code soon, no other OV2640 model. I am not aware of it, because the camera works even if it reports error.

mozgy, 2016-02-17:
Hello, I have two ArduCAMs that have 0x42 there and work fine, with 0x41 it's "Can't find OV2640 module!" Did I get fake module ?

ArduCAM, 2016-02-18:
Hi Mario, I am aware that the OV2640 have different versions, so both 0x41 and 0x42 are valid.
Please try to take a photo to confirm it works well.

mozgy, 2016-02-18:
Hello, it works with 0x42 in my sketch code based on yours. I could not get html part from your examples to work.

mozgy, 2016-02-18:
this worked for me ->
```
  if ( ( vid == 0x26 ) && ( ( pid == 0x41 ) || ( pid == 0x42 ) ) ) {
    Serial.println("OV2640 detected.");
  }else{
    Serial.println("Can't find OV2640 module!");
    while(1);
  }
```

ArduCAM, 2016-02-19:
Yes Mario, you did the right thing. I think I will update the code as yours.

PegX, 2017-11-21:
@ArduCAM Hi, I meet a wield output with the following command:
sudo ./ov2640_capture -c test.jpg 320x240
and the result likes:
SPI interface OK!
The value of vid is 0xff
The value of pid is 0xff
Can't find OV2640 module!
Additional information is #define OV2640_CHIPID_LOW   0x0B and I changed the if condition likes this if((vid != 0x26) || (pid != 0x41)) . 
So i did not get why the vid and pid both are 0xff. It seems initialisation is wrong?

UCTRONICS, 2017-11-21:
@PegX
Hi,
Please try to 
sudo nano /boot/config.txt
uncomment dtparam=i2c_arm=on
uncomment dtparam=spi=on 
Then reboot again .

PegX, 2017-11-21:
@ArduCAM Thanks for your reply. However, I do not think this is the true reason since I did the similar step with  sudo raspi-config and i can find the following lines in config.txt file.
dtparam=spi=on
dtparam=i2c_arm=on
I suppose that would be the same thing with uncomment.

UCTRONICS, 2017-11-21:
@PegX
Hi,
Have you install those below tools?
sudo apt-get install wiringpi 
sudo apt-get install i2c-tools 
sudo apt-get install libi2c-dev 
sudo apt-get install python-smbus 
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/RaspberryPi/README.md
If your camera still can not work,you can test it on Arduino UNO board firstly to ensure your camera is OK.

PegX, 2017-11-21:
@ArduCAM Hi, I also tried with Arduino to make sure my ArduCAM is not broken. Actually, I understand where is my problem (but I am not sure, I will try this everning. )
With this link https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/RaspberryPi/README.md. 
"memorysaver.h must be configured to enable RASPBERRY_PI definition before make the library.
cd /home/pi/ArduCAM/example/RaspberryPi "
I utilize the default memorysaver.h which indiates as the Arduino? I suppose I did not enable RASPBERRY_PI. 
 
Thanks anyway. I will update this everning again.

PegX, 2017-11-21:
@ArduCAM By the way i successfully maked these source files. That means I shoud install wiringpi, i2c-tools, libi2c-dev as well as python-smbus previously. Thank you very much.

UCTRONICS, 2017-11-21:
@PegX
Hi,
You are welcome.
Let us know if you need more help.
Regards.

PegX, 2017-11-21:
@ArduCAM Hi Lee, Actually, I successfully to capture an image with the ArduCAM. 
I tried the following two directions and both of them are working normally. 
1. git the repo from this link https://github.com/ArduCAM/RaspberryPi. 
   And i inserted #define RASPBERRY_PI in memorysaver.h file
and then make and get the binaries. 
2 The other approach is that getting repo from https://github.com/ArduCAM/Arduino
  and I uncommented the #define RASPBERRY_PI in Arduino/ArduCAM/memorysaver.h file and then ran the make command in the directory of Arduino/ArduCAM/examples/RaspberryPi/. The output and also the result of capturing same. However, I just confused that why putting the raspberryPi example under the Arduino folder?

Anyway, thanks and I got my expect result.