ArduCAM/Arduino #20: ArduCAM Library not Compiling for Arduino DUE

picrust37, 2016-02-09:
Hi Lee,

I am working with ArduCAM-Mini 2MP and Arduino DUE, and I am running the lastest version of Arduino, 1.6.7.  I am trying to run some of the example code in the ArduCAM library, but I have not been able to get any of the downloaded library code to compile when Board is set to DUE.  All of the code in the library will compile when Board is set to UNO, but as soon as I switch Board to DUE(Programming Port) I get a long list of error warnings.  For example, compiling ARDUCAM_MINI_OV2640_Video_Streaming gives warnings such as the following (The complete list of warnings is attached as a text file):
```
In file included from C:\Users\myUserName\Documents\Arduino\libraries\ArduCAM\examples\mini\ArduCAM_Mini_OV2640_Video_Streaming\ArduCAM_Mini_OV2640_Video_Streaming.ino:23:0:

C:\Users\myUserName\Documents\Arduino\libraries\ArduCAM/ArduCAM.h:131:0: warning: "pgm_read_byte" redefined [enabled by default]

     #define pgm_read_byte(x)        (*((char *)x))
^ 

In file included from C:\Users\myUserName\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\cores\arduino/Arduino.h:31:0,

                 from sketch\ArduCAM_Mini_OV2640_Video_Streaming.ino.cpp:1:

C:\Users\myUserName\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\cores\arduino/avr/pgmspace.h:56:0: note: this is the location of the previous definition

 #define pgm_read_byte(addr) (*(const unsigned char *)(addr))

 ^
```
I was able to successfully run some basic code (blinking an LED light) with the DUE, so the way that I am connecting to the DUE does not seem to be the issue.  I have been reading a lot on the forums, and I have not found anything that seems to fully address my problem.  I have found some things that suggest this may be a problem due to PROGMEM only being compatible with AVR-based boards, while the DUE is ARM-based, but I am unsure how to proceed if that is the case.

Please let me know if you know anything regarding this error and if there are any modifications to the code that I should be making in order to run the library on DUE.  I would appreciate any help in resolving this issue.

Thank you for your time,
Jessica

For a complete error warning, please see:
[ArduCAM_DUE_Errors.txt](https://github.com/ArduCAM/Arduino/files/123644/ArduCAM_DUE_Errors.txt)

ArduCAM, 2016-02-11:
Hi Jessica, thanks for your question.
It is strange that I do not have this compilation problem here with my Arduino IDE 1.6.7.
Board Arduino Due (Programming Port) with version 1.6.6 add on package installed.
The definition of PROGMEM is defined in the ArduCAM.h file, so you have to look into this file for detail.
Let me know if you make any progress.

picrust37, 2016-03-02:
Hi Lee,

I have not managed to make the compilation warnings go away, but it appears they are just warnings and do not effect the functionality of the code.  I figured out that my wiring had been incorrect originally.  I also had to change the baud rate from Serial.begin(921600) to Serial.begin(115200).  Now I can get single images in the ArduCAM host application.

Thanks!

Patrickyp, 2016-07-21:
Hello,

I'm having issues as well getting the ArduCam to run with the DUE.  Could you tell me how you wired up the camera to the Arduino to get it to work eventually?

jmcwil, 2016-07-21:
@Patrickyp Hi, I would be happy to tell you how I wired it.

Basically follow the information [here](https://www.arduino.cc/en/Reference/SPI) for how to attach to the SPI header pins.  Make sure you put VCC on the SPI header.

The CS pin on ArduCAM can be wired to pin 10 (there are a couple others that will work, but this one is in the example code).  The SDA and SCL go to SDA1 and SCL1, respectively.

I have been running the OV2640_FIFO_UART example with the ArduCAM host application.  If you happen to find a way to use the ArduCAM without the host application please let me know!!

Patrickyp, 2016-07-22:
@jmcwil Thank you for your reply.  Just to confirm did you use the SPI header and not the ICSP header on the DUE?

ArduCAM, 2016-07-22:
@Patrickyp, you have to use the signals on ICSP, it is the default SPI signals for all Arduino boards.
Send the photo of your wiring, it may help diagnose your problem.