ArduCAM/Arduino #21: distorted image in latest http server demo

tzapu, 2016-02-18:
hello

When using `ArduCAM_Mini_OV2640_websocket_server` everything works fine

When i attempt to use `ArduCAM_Mini_OV2640_Capture` i get garbled frames in both video and static capture

sample 
[image]

could you help? i tried to look for code differences but byte buffers throw me off... init code seems the same

tzapu, 2016-02-18:
the issue seems to be in the read/transmit part of the sketch, i have tried the read code from the working sketch in the new one that distorts and capture seems to work fine
it also seems slower, si i am sure the code is not what it should be..
```
void camCapture(ArduCAM myCAM){
  WiFiClient client = server.client();
  
  size_t len = myCAM.read_fifo_length();
  if (len >= 393216){
    Serial.println("Over size.");
    return;
  }else if (len == 0 ){
    Serial.println("Size is 0.");
    return;
  }

  uint8_t temp, temp_last;
  static int i = 0;
  static uint8_t first_packet = 1;
  byte buf[2048];
  uint32_t length = 0;

  myCAM.CS_LOW();
  myCAM.set_fifo_burst();//Set fifo burst mode
  SPI.transfer(0x00);
  //Read JPEG data from FIFO
  while ( (temp != 0xD9) | (temp_last != 0xFF))
  {
    temp_last = temp;
    temp = SPI.transfer(0x00);

    //Write image data to buffer if not full
    if (i < 2048)
    {
      buf[i++] = temp;
      //yield();
    }
    else
    {
      if (first_packet == 1)
      {
        //sendClientData((char*)buf, 2048, 0x02);
        client.write((char*)buf, 2048);
        first_packet = 0;
      }
      else
      {
        client.write((char*)buf, 2048);
      }
      i = 0;
      buf[i++] = temp;

    }
    //yield();
  }
  //Write the remain bytes in the buffer
  if (i > 0)
  {
    //Serial.println("write last 256 bytes");
    client.write((char*)buf, 2048);
    //delay(10);
    //Serial.println("finish write last 256 bytes");
    i = 0;
    first_packet = 1;
  }
  //yield();
  myCAM.CS_HIGH();
  
}
```

tzapu, 2016-02-18:
it seems modifying buffer size from 4096 to 2048 makes it work, but also seems a lot slower in updating (reads half as much)...
is there anything that could be done?

ArduCAM, 2016-02-18:
Please modify the source code in the ESP8266 library like the following:

[WiFiClient.h]
#define WIFICLIENT_MAX_PACKET_SIZE 4096
 
[ClientContext.h]
add the line 
_pcb->snd_buf = TCP_SND_BUF;
before
size_t room = tcp_sndbuf(_pcb);
 
[lwipopts.h]
#define TCP_MSS                         2048

admin@arducam.com

tzapu, 2016-02-18:
thanks for the info.
i won t be able to test for a while as it s more involved and probably will be overwritten on any core update
but with 2048 as buffer size it seems to work fine for me in the mean time
thanks

mozgy, 2016-02-18:
Changing
```
static const size_t bufferSize = 2048; // 4096
```
from 4096 to 2048 works for me too -

[image]

that PCB on keyboard is my ESP8266 to ArduCAM breakout ..

ArduCAM, 2016-02-19:
Congratulations Mario. Breakout board looks nice.

tzapu, 2016-02-19:
nice, love it when you use a product to demo said product is working :D
which leads me to the question, how many arducams do you have ? :P
lovely breakout, looking foerward to see what you come up with

ArduCAM, 2016-02-19:
We now have two models of ArduCAM mini, 2mp and 5mp. We will design another 3MP soon.

tzapu, 2016-02-19:
hi, i meant @mozgy :P he took a pic of an arducam with another arducam :camera: 
but it s nice to know another 3mp is coming, thanks

mozgy, 2016-02-19:
As oshpark produces trices I bought two 2MP cams, here's PCB https://oshpark.com/shared_projects/pINKH5In.

janisalnis, 2016-03-08:
Hello Lee, thanks for sending Arducam through Ebay. 

Started to try it out. Arducam mini board draws nearly 100 mA. For a battery powered 
version would be probably nice to add a FET switch on power line. 

Downloaded the library from github on 05. March 2016.
Library should be placed in directory arduino/libraries/ArduCAM.
If the whole arduino-master directory was placed under the libraries folder, the compiler 
could not find *.h files. 

First tried an example with Arduino streaming to Windows PC:
"ArduCAM_Mini_OV2640_Video_Streaming".

Serial.begin(921600); did not work with Arduino Serial Monitor window.
changed to 115200
Arducam.exe downloaded from the Arducam webpage:
http://www.arducam.com/downloads/demo/arducam_host/ArduCAM_host.zip
did not support speed of 900kbps. Appeared that the program  was not the latest version and should use the program included with the github library example.

Next tried the example with ESP8266: "ArduCAM_Mini_OV2640_Capture"
Serial debugging gave message
Can't find OV2640 module!

if ((vid != 0x26) || (pid != 0x41))
looked in websocket example that needed to change camera id number to 0x42.

while(1); caused reset after a few seconds, as probably, the ESP8266 had no time to do 
other internal things.

In the program, I guess, it should also be nr 16 in the comment line.
// set GPIO15 as the slave select :
const int CS = 16;

I also experienced the problem described in this issue with the broken and mixed parts of the image.
Changing of buffer size to 2048 helped. 
static const size_t bufferSize = 2048;

There is an easy way to capture a picture without the demonstation html:
http://192.168.1.140/capture

Resolution can be changed with numbers between 0 and 8 in such a way:
http://192.168.1.140/?ql=3

I would like to make a PIR activated alarm automatically uploading a picture to a server. 
I am a big fan of MobileWebcam:
http://forum.xda-developers.com/showthread.php?t=950933
On a server is a mobilewebcam.php script that receives the picture file and puts it into a 
directory with a time tag. My webcam page is at http://asi.lv/alnis

Form Linux the upload line looks like that:
curl -F userfile=@/tmp/current.jpg barbara320.gotdns.com/webcamwr703nRiga/mobilewebcam.php

Php upload is more convenient than FTP upload as the file name can be automatically prepared on the server side.
I would be very happy if such picture file upload could be implemented on ESP8266.

Greetings from Latvia, Janis.

dmolner, 2016-03-27:
Great!

I with these is working automatic uploading pictures to my server.
I only had to change name of server and folder.

ArduCAM, 2016-03-29:
@Janis from Latvia, thanks for your detail information, it definitely
guides other users much easier. May I know your email address?

tzapu, 2016-03-29:
@janisalnis thank you dude very much for the http upload sample. i could not find that anywhere for the life of me.
if anyone is interested i ve got my work in progress project here:
https://github.com/tzapu/ESP8266ArducamTimelapse
timelapse machine with sd support, ntp sync, deep sleeping, powering off cam, powering off wifi, and a lot more.. but really in progress, did not have enough time to sort everything i wanted yet. 

http upload code is one thing that i have been missing dearly. thank you very much

Shdwdrgn, 2016-11-07:
Has there been any further progress on the bufferSize issue?  Like others, setting the value to 2048 gives me proper video output, but it is noticeably slower.  I'm only trying to stream a 320x240 video, but even that small size has some delay and a lot of skipped frames.

One other thing I noticed, and perhaps someone can explain this to me...  I added the timer code from the capture section into the streaming section, so now my serial console shows how long it is taking to output each frame, but the results are confusing.  When I first start running my ESP, the first frame will take about 4000ms, but then the next frame takes about 40,000ms -- and it continues alternating between a fast frame and a slow frame that takes 10x as long.  Even worse, it slows down quickly over time.  After letting it run for about 30 minutes, the frames are alternating between 800,000ms and 8,000,000ms!

My theory on this is that serverStream() is being called multiple times, resulting in overlapping frames and a slow memory leak.  I want to test this by adding a static flag at the beginning of the function that exits if the flag is already set, thus only allowing a single instance to run -- unfortunately I have not had a chance to write this and test it out.  Also has anyone checked how this program work when called from multiple devices at the same time?

Shdwdrgn, 2016-11-08:
Sorry, I found a mistake in the timing code I added.  It looks like it's actually taking about 140ms (consistently) to create and transmit each frame, but that number is still creeping up over time and hits about 170ms in 30 minutes.

So... ignore most of the last two paragraphs in my last comment, but I'd still like to know if there's a way to fix the bufferSize issue to use a larger value?  Images of 320x240 appear to contain 6148 or 6149 bytes, and it would be great to capture that in a single pass.

VIVEKSUTAR0709, 2017-11-14:
Hello Sir/Madam
have a problem in Websocket Camera application
It shows me Websocket Connected while clicking on Connect (file attached)

but when I press Send Button it won't show anything..

please help mi with this.
[image]

supprot, 2017-11-14:
It means the CC3200 is running without problem. The MT9D111 is not initialized very well, so please check the connection and the sensor register initialization.

VIVEKSUTAR0709, 2017-11-14:
I am just putting that MT9D111 in the socket of AurduCam. how to initialize it...can you tell mi that..?

Thank you.

VIVEKSUTAR0709, 2017-11-14:
Do i need to change any thing in the code provide by CC3200SDk?

ArduCAM, 2017-11-14:
You can follow the instruction below to restore the factory firmware (please use the files from http://www.arducam.com/downloads/CC3200_UNO/factory_default/websock_camera.rar  )
http://www.arducam.com/downloads/wifi_camera/factory_default/restore_factory_default.pdf

VIVEKSUTAR0709, 2017-11-14:
i already did that...but it won't show any output while click on Send button

xelfer, 2017-12-19:
Are there any other suggestions in resolving this issue? I have a Feather huzzah ESP8266 with the OV5642_MINI_5MP_PLUS (The only option that worked.. how do I tell if it's a "PLUS" or not? I purchased this: https://www.iot-store.com.au/products/arducam-mini-module-camera-shield-with-ov5642-5-megapixels-lens). 

I've made the changes to WiFiClient.h, ClientContext.h and lwipopts.h

I've tried `static const size_t bufferSize = 2048;`

and I'm still getting results like this when I try to go any bigger than 320x240. Sometimes 640x480 works. Earlier when I was trying I managed to get 1600x1200 consistently without a problem but now I can't seem to get more than 1/4th of the photo. 

320x240 (working at the moment):
[image]

640x480 (working at the moment):
[image]

1024x768:
[image]

1280x960:
[image]

Full code I'm using is here: https://gist.github.com/xelfer/7707464220aff93390255fe97b033351

Thanks