ArduCAM/Arduino #22: send photo from arducam shield rev c to server

khalidzuhd, 2016-02-20:
hello 
im using arducam shield  rev c  and ov 2640 without screen 
 wont to send captured photo from sd card  to server using wifi mod esp2866
can you help me with good code ?

ArduCAM, 2016-02-22:
Khalidzuhd, you can use the code from: https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/ESP8266/ArduCAM_Mini_OV2640_Capture

amarsumberel, 2016-02-22:
Hello, I am using Chipkit dp32 for arducam OV5642. Do you have ported library or how to use this camera with chipkit32?
thank you.

ArduCAM, 2016-02-24:
Not yet, but we can help you when you have any problem porting our Arduino library to chipkit32.

amarsumberel, 2016-02-24:
@ArduCAM Do you have any tutorial documents to port library from arduino to PIC32 or ChipKit? I am using PIC32MX250F128B MCU, I want to take photo then save image in the sdcard using OV5642 arducam mini. 
Thank you.

ArduCAM, 2016-02-25:
@W1ND. This is the chipkit library and example for our ArduCAM-Mini-2MP, hope you can help us to improve it and port to 5MP.
www.arducam.com/downloads/porting/Chipkit/

amarsumberel, 2016-03-03:
@ArduCAM Lee, Thank you very much. It's very useful camera development library. Is it possible to take good resolution image (Not using high compression method) by Arducam mini?

ArduCAM, 2016-03-08:
@W1ND, sorry for my late reply. As you can see there is limited frame buffer on the camera.  Uncompressed image need a lot of memory, so you have to trade off between the resolution and compression.

khalidzuhd, 2016-04-04:
im sory i dont know how !!
can you helo me with the code 
ardocam rev c (ov2640)
with esp8266 
please please im desperate

khalidzuhd, 2016-04-04:
i need to use ftp to send the image