ArduCAM/Arduino #23: Connecting an Arducam Mini Module Camera Shield with OV2640 to an Uno

eladyt, 2016-02-24:
Hi,

Reading around, I think the correct connections of the pins are:
CS -> D10
MOSI -> D11
MISC -> D12
SCK -> D13
GND -> GND
VCC -> +5V

But where should SDA and SCL connect to? 

Also, how are these set in the code?

Thanks,

Elad

eladyt, 2016-02-24:
I just realized: It's an Arduino Uno (v1), not V3. Any way to connect it?

ArduCAM, 2016-02-24:
As you can see more information about Arduino Duemilanove board from the page: https://www.arduino.cc/en/Main/arduinoBoardDuemilanove 
I2C: analog input pins A4 (SDA) and A5 (SCL).
So you have to connect these two pins to ArduCAM correspondingly.

eladyt, 2016-02-24:
Thank you, Lee.

I connected the camera as you suggested. I'm running the low power mode demo now, and getting this on the serial output:

CAM Capture Done!
524287
Over size.

Incidentally, I had to change the serial rate to 9600, otherwise I only saw rubbish on the serial monitor.

Does the "over size" warning mean that the images are corrupt? If not, where are the captured images saved? 

Thanks!

ArduCAM, 2016-02-25:
@Elad, it seems that the camera is not initialized well, did you try our example without any modification? The baudrate speed depends on your board's USB to serial interface chip.

eladyt, 2016-02-26:
@ArduCAM Thank you for your help. I think I'm making progress.

I'm running ArduCam_Mini_2MP_TimeElapse2SD (See below).

The camera now seems to work (I get "OV2640 detected" on the serial output). However, since the camera is connected to D10-13, I can't connect the SD card there. I tried to connect the camera to D0-D3, but it didn't work.

The SD card is connected to D4-7, and the SD_CS is specified as 4. 

Can you suggest how to wire both camera and SD correctly?

Thanks!

---------
// ArduCAM demo (C)2015 Lee
// web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules, and can run on any Arduino platform.
//
// This demo was made for Omnivision OV2640 2MP sensor.
// It will run the ArduCAM Mini 2MP as a real 2MP digital camera, provide both JPEG capture.
// The demo sketch will do the following tasks:
// 1. Set the sensor to JPEG mode.
// 2. Capture and buffer the image to FIFO every 5 seconds 
// 3. Store the image to Micro SD/TF card with JPEG format in sequential.
// 4. Resolution can be changed by myCAM.set_JPEG_size() function.
// This program requires the ArduCAM V3.4.0 (or later) library and ArduCAM Mini 2MP shield
// and use Arduino IDE 1.5.2 compiler or above

#include <UTFT_SPI.h>
#include <SD.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

#if defined(__arm__)
#include <itoa.h>
#endif

#define SD_CS 4
// set pin 4 as the slave select for SPI:
const int SPI_CS = 10;

ArduCAM myCAM(OV2640, SPI_CS);
UTFT myGLCD(SPI_CS);

void setup()
{
  uint8_t vid, pid;
  uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
#else
  Wire.begin();
#endif
  Serial.begin(9600);
  Serial.println("ArduCAM Start!");
  // set the SPI_CS as an output:
  pinMode(SPI_CS, OUTPUT);

  // initialize SPI:
  SPI.begin();
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55)
  {
    Serial.println("SPI interface Error!");
    while (1);
  }

  //Check if the camera module type is OV2640
  myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
  if ((vid != 0x26) || (pid != 0x42))
    Serial.println("Can't find OV2640 module!");
  else
    Serial.println("OV2640 detected");

  //Change to BMP capture mode and initialize the OV2640 module
  myCAM.set_format(JPEG);

  myCAM.InitCAM();
  //myCAM.OV2640_set_JPEG_size(OV2640_640x480);
  //myCAM.OV2640_set_JPEG_size(OV2640_1600x1200);

  //Initialize SD Card
  if (!SD.begin(SD_CS))
  {
    //while (1);		//If failed, stop here
    Serial.println("SD Card Error");
  }
  else
    Serial.println("SD Card detected!");
}

void loop()
{
  char str[8];
  File outFile;
  byte buf[256];
  static int i = 0;
  static int k = 0;
  static int n = 0;
  uint8_t temp, temp_last;
  uint8_t start_capture = 0;
  int total_time = 0;
  ////////////////////////////////

  start_capture = 1;
  delay(5000);
  
  if (start_capture)
  {
    //Flush the FIFO
    myCAM.flush_fifo();
    //Clear the capture done flag
    myCAM.clear_fifo_flag();
    //Start capture
    myCAM.start_capture();
    Serial.println("Start Capture");
  }

  while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));

  Serial.println("Capture Done!");

  //Construct a file name
  k = k + 1;
  itoa(k, str, 10);
  strcat(str, ".jpg");
  //Open the new file
  outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
  if (! outFile)
  {
    Serial.println("open file failed");
    return;
  }
  total_time = millis();
  i = 0;
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();
  temp = SPI.transfer(0x00);
  //
  //Read JPEG data from FIFO
  while ( (temp != 0xD9) | (temp_last != 0xFF))
  {
    temp_last = temp;
    temp = SPI.transfer(0x00);

    //Write image data to buffer if not full
    if (i < 256)
      buf[i++] = temp;
    else
    {
      //Write 256 bytes image data to file
      myCAM.CS_HIGH();
      outFile.write(buf, 256);
      i = 0;
      buf[i++] = temp;
      myCAM.CS_LOW();
      myCAM.set_fifo_burst();
    }
  }
  //Write the remain bytes in the buffer
  if (i > 0)
  {
    myCAM.CS_HIGH();
    outFile.write(buf, i);
  }
  //Close the file
  outFile.close();
  total_time = millis() - total_time;
  Serial.print("Total time used:");
  Serial.print(total_time, DEC);
  Serial.println(" millisecond");
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Clear the start capture flag
  start_capture = 0;
}

eladyt, 2016-02-26:
My bad. I didn't realize MOSI, MISO and clock were shared by the SD and the camera. It now works perfectly!

Thanks for your support!

CyprianLucheli, 2017-06-01:
what are the pin connections for UNVSIM800 SIM800 to arduino please? esp 3V9, DCIN, VDD and 
[image]
R

Vazrul, 2017-07-07:
please , how to connect omnivision camera ov2640 to arduino , and the pin omvision camera ov2640 have 20 pin ???