ArduCAM/Arduino #24: Issue getting capture done (camera write FIFO done flag)

efritz09, 2016-03-08:
Hello,

For a class project, we're trying to interface with the 2MP Arducam Mini through a Raspberry Pi using bare metal programming. We have working I2C and SPI drivers and are attempting to get any kind of image data from the camera. To guide us, we've heavily examined your code for the Arduino, as we assumed it'd be closer to the bare metal programming we're doing than using your code for the Raspberry Pi with Linux. (ArduCAM_SPI_OV2640_FIFO_UART.ino) Using an Arduino, we modified it to print BMP. It confirms that the capture is done and successfully dumps the values to the serial monitor.

Our problem is that we can't seem to get a capture done from the camera. Here's what we can get:
1) We can successfully check the SPI communications to the camera by querying the test register 0x00 with any value, and we can read the same value back
2) We can read the ArduChip version through the 0x40 register and get the version (0x40) back.
3) We set the mode to MCU to LCD mode by writing 0x00 to 0x02. It's listed as reserved in the manual, but we noticed this was done in your code so we tried it anyway.
4) We can read the sensor address through i2c successfully
5) We can successfully write the long list of commands in ov2640_regs.h, and we're writing the correct commands for Bitmap. 
6) We then flush the fifo, clear the fifo, and start a capture. We've confirmed that we're writing the same values as your code (0x84,0x01    0x84,0x01    0x84,0x02)
7) We then write 0x41, 0x00, and read back the value in this register. In your code, it takes approximately 20-30 reads before capture is complete. Our code never reaches that point. Your code returns numbers like 0x01 until it hits 0x08 or 0x09 which has the done bit set (bit 3), our code returns 0x40, sometimes 0x00 or 0x044

Do you have any insight as to why we can't get a capture done? What do the other bits correspond to in 0x41? We've spent 20+ hours debugging this with a logic analyzer to confirm that our SPI and I2C work, and we're writing the same addresses and values as your code does. We've stepped through your code and made sure to send identical commands. The project is due in a few days so a speedy reply would be appreciated.

As an aside, we notices that flush_fifo() and clear_fifio_flag() are identical functions, is there a reason to send this command twice?

efritz09, 2016-03-09:
We figured out the issue!

The datasheet [here](http://www.arducam.com/downloads/shields/ArduCAM_Mini_2MP_Camera_Shield_Hardware_Application_Note.pdf) says that the polarity is 0 and the phase is 1. We found experimentally that the phase should be 0, and after this we see that the timing diagram shows that the phase should be 0. Please update this!

closed 2016-03-09 by efritz09

ArduCAM, 2016-03-10:
@Eric, yes you are right, it is typing mistake. we will update the document soon.

eric, 2016-03-10:
Wrong eric.

ArduCAM, 2016-03-10:
Sorry , it should be Eric Fritz.