ArduCAM/Arduino #25: ArduCam Library not compiling

diasrobson, 2016-03-08:
this is the error message i get its the same error on all the of the camera examples can you plz help   

Arduino: 1.7.8 (Mac OS X), Board: "Arduino M0"

ArduCAM_Mini_2MP_TimeElapse2SD.ino:16:22: fatal error: UTFT_SPI.h: No such file or directory
compilation terminated.
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

dprophet, 2016-11-21:
This is not an issue. He simply didnt add the library UTFT4ArduCAM_SPI to his IDE

VIVEKSUTAR0709, 2017-11-07:
hii...
I am trying to compile the .INO file in Energia foe ArduCAM_CC3200_UNO borad and getting

C:\Users\DriveZero\Desktop\ArduCAM_MT9D111_Camera_Playback\ArduCAM_MT9D111_Camera_Playback.ino:29:21: fatal error: ArduCAM.h: No such file or directory

 #include <ArduCAM.h>

                     ^

compilation terminated.

exit status 1
Error compiling for board ArduCAM UNO w/ CC3200 (80MHz).

please help mi with this.