ArduCAM/Arduino #28: Board not working with 4 OV5642. Only with 2 or 3.

Displav, 2016-04-03:
Hello,

I have purchased the Arducam Mini shield and 4 OV5642 cams and I am currently trying to implement some of the provided examples but have come across some technical difficulties.

Specifically, although I can successfully upload the “ArduCAM_Mini_5MP_4Cams_TimeElapse2SD_LowPower” to the Arduino, the code runs consistently only in the cases when 2 camera modules are connected.

Plugging 3 cameras results in the desired behavior in some cases, while in others, the Arduino "freezes" (L led is ON) and is not recognized when connected to the PC.

Plugging 4 camera modules always leads to the aforementioned "freezing" behavior.

I have tried powering the Arduino from a 12V transformer without noticing any improvement.

Adjusting the shield's jumper between 3.3 and 5V doesn't seem to make any difference either.

It seems I've reached a dead end, so some insight that would help me put the whole thing together would be greatly appreciated.

ArduCAM, 2016-04-04:
@Display, Please first make sure if each camera works alone with the Arduino before any further testing. The OV5642 camera comsumes a lot of current,  so you need to add additional power supply to the mini USB port on the muti-camera adapter board and try again. Let me know if you have any news.

Displav, 2016-05-21:
The issue was the current.

I used both USB ports to give power to both arduino and the adapter board.
Also using a 2A power adapter from and giving power to the arduinos USB port also works.

Thanks for your help!