ArduCAM/Arduino #29: Save captured image to computer

picrust37, 2016-04-06:
Hi,

I am currently searching for a way to capture an image with the ArduCAM and transmit it over Radio Frequency back to a computer.  Is there a way to save the captured images directly to a file on the computer without using the host application?  I am using this as an intermediate step in solving my problem, but I have not found a way to get the file on the computer without using an SD card.  Does this require the use of an SD card?  I am trying to find the best way to approach this problem, and I would appreciate any help and advice.  

Thank you!

ArduCAM, 2016-04-06:
@picrust37
Sorry, I don't understand your question. If you want to save the image on computer, you have to transfer the image data to computer first via Radio or UART, then you need to write a simple program to save the received image data to file. Let me know if you need more help.

picrust37, 2016-04-06:
Thanks for the quick response!  I am sorry my question is vague, but your answer is beginning to give me clarity.

I am confused if the simple program to receive the data is something that can be coded in Arduino, or if it is better to use another language.  I am also having trouble obtaining the raw data from the ArduCAM.  I have currently been using the ArduCAM with the ArduCAM host application and the example, ArduCAM_SPI_OV2640_FIFO_UART, but I have been unable to save any images using it.  My goal is to save the files directly to my computer, not to an SD card.  If you can provide any additional information to help clarify how one may receive the transmitted data, I would be greatly appreciative.

Thank You!

ArduCAM, 2016-04-06:
@picrust37, I think you still don't understand my last reply. As you can see Arduino is connected to PC via usb or bluetooth, you have to send the image data over these interface like my example does.
The PC host application just received the image data, display it on the screen and can also save the image to file if you check the save options in the GUI.

atiqamuhammad, 2017-12-19:
Hi,
My project is similar to this and im getting the same problem. I need to save picture directly to computer from arducam ov5640. But my arducam is connected with an ultrasonic sensor and it will take picture only when the sensor is triggered. But im unable to save picture in my computer. Can u please tell a solution for this?
Thankyou.

atiqamuhammad, 2017-12-19:
One more thing,
I do not want to use any arducam app to capture picture. I need to take picture automatically and save it to the computer.

ArduCAM, 2017-12-21:
You have to tell us how you connect the camera to your computer.
If you use a Arduino board such as UNO R3 to PC using USB port, you have to read the camera data and send the image data over the USB port. You need to write your own software to receive the image data if you don't want to use our host app on PC.

dider7, 2018-08-05:
Hi, I’m using ArduCam Mini 5MP (OV5642) with the Arduino Uni R3. I’m able to capture a single jpeg image and send it to a PC over serial port. But it takes a significant amount of time to send the image data if it’s captured at the highest resolution. Is there a faster way to transfer the data?

(I followed the video streaming example and only used the bit used to capture a single image.)

UCTRONICS, 2018-08-06:
@dider7 
The baud rate of the serial port determines the transmission speed.

dider7, 2018-08-17:
Thanks @UCTRONICS. Changed the baud rate to 921600 and it's a lot faster.

emmanuel-nwogu, 2020-10-25:
> Hi, I’m using ArduCam Mini 5MP (OV5642) with the Arduino Uno R3. I’m able to capture a single jpeg image and send it to a PC over serial port. But it takes a significant amount of time to send the image data if it’s captured at the highest resolution. Is there a faster way to transfer the data?

Hi @dider7, do you still have the code for capturing and converting to JPEG as stated here? Would really appreciate it.

Dickoabc123, 2023-07-17:
hey @dider7 @emmanuel-nwogu if you got this working and wouldn't mind sharing the code, that would be great.

emmanuel-nwogu, 2023-07-18:
I wrote [this](https://github.com/ArduCAM/Arduino/issues/572) a while ago, hope it helps :)

Dickoabc123, 2023-07-18:
That is very kind of you, thank you.
I managed to get a little bit working late last night and will carry on with it today if I get time.
If I get stuck I’ll make sure to look over the link you E provided. Thanks again :)x

Best Regards
Carl Dickinson