ArduCAM/Arduino #36: Arducam mini 5mp + ESP8266

labajo, 2016-04-19:
I'm trying to use an Arducam mini 5MP with the websocket example and it doesn't run for me.

I upload to my esp8266 the following program but It does not run well.
https://gist.github.com/labajo/99a2be6098db086430bf927b4275f6a8

I cannot see a special trace to find out the problem.

```
SPI1 interface Error!
OV5642 detected.

Connecting to iMac

WiFi connected
Server started
192.168.2.3
Client connected
Analyzing request headers
Got Line: GET / HTTP/1.1
Got Line: Host: 192.168.2.3
Got Line: Connection: Upgrade
Got Line: Pragma: no-cache
Got Line: Cache-Control: no-cache
Got Line: Upgrade: websocket
Got Line: Origin: null
Got Line: Sec-WebSocket-Version: 13
Got Line: User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_11_2) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/44.0.2403.155 Safari/537.36
Got Line: Accept-Encoding: gzip, deflate, sdch
Got Line: Accept-Language: es-ES,es;q=0.8
Got Line: Sec-WebSocket-Key: KNtP+FRULpiEevBrx1imSA==
Got Line: Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
Got Line: 
Websocket established
capture
```
Could you help me?

ArduCAM, 2016-04-20:
@Juan
There is SPI timing error with 5MP when used in ESP8266, please try our special library released for ESP8266 from https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/tree/master/libraries/ArduCAM

dmolner, 2016-04-20:
Hello,

I test but in SPI I have 170 not 55

labajo, 2016-04-24:
Hi,

with the special library I get the following error:

```
0
chksum 0x0f
csum 0x0f
ArduCAM Start!
SPI1 interface Error!

Soft WDT reset

ctx: cont 
sp: 3fff2940 end: 3fff2b50 offset: 01b0

>>>stack>>>
3fff2af0:  3fff1afc 3fff1078 3fff1afc 40202574  
3fff2b00:  feefeffe feefeffe feefeffe feefeffe  
3fff2b10:  feefeffe feefeffe feefeffe feefeffe  
3fff2b20:  feefeffe feefeffe feefeffe 3fff1b28  
3fff2b30:  3fffdad0 00000000 3fff1b20 40207434  
3fff2b40:  feefeffe feefeffe 3fff1b30 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

I had to add the following definitions in the ArduCAM_Mini_OV5642_Capture example:
```
#define OV5642_CHIPID_HIGH 0x300a
#define OV5642_CHIPID_LOW 0x300b
```

But they are defined in keywords and ov5642_regs.h

ArduCAM, 2016-04-26:
@Juan
Sorry for my late reply. It seems that the SPI error and go to the while(1) loop which causes the esp8266 reboot.
Please check the following link for the new library.
http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/ 
http://www.arducam.com/downloads/ESP8266_UNO/ArduCAM_ESP8266_UNO_DS.pdf 
Let me know if you need more help.

ArduCAM, 2016-04-27:
@dmolner We have update our library for ESP8266 board, please check the latest library and examples from http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/

dmolner, 2016-04-27:
I test these in my ESP8266 and OV5642 but I have a message "SPI1 interface Error!"
I include a print and I have 170 not 55
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  ->Serial.println (temp);
```

// Arducam mini controlled by ESP8266.  Arduino IDE 1.6.5. Photo upload to a php script on a webserver using html POST.
// Libraries from Arducam github placed under arduino/libraries/arduCAM. Power consumption ca 200 mA from 3.3V.
// PIR module controls FET that switches on 3.3 power Lipo+buck regulator used.  ESP connects to WiFi and starts taking and uploading photos.
// http://www.arducam.com/tag/arducam-esp8266/
// ESP12-E module: CS=GPIO16   MOSI=GPIO13   MISO=GPIO12   SCK=GPIO14  GND&GPIO15  VCC&EN=3.3V SDA=GPIO4  SCL=GPIO5  GPIO0=button  GPIO2=BlueLED

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

const char* ssid     = "myssid"; // Your ssid
const char* password = "mypw"; // Your Password

WiFiClient client;

#define DEBUGGING  // Enabe debug tracing to Serial port.
#define MAX_FRAME_LENGTH 64   // Here we define a maximum framelength to 64 bytes. Default is 256.
#define CALLBACK_FUNCTIONS 1  // Define how many callback functions you have. Default is 1.
const int CS = 16; // chip select of camera
ArduCAM myCAM(OV5642, CS);

int led=2; //blue led pin on GPIO2 of ESP-12E  switching it on during picture taking.

void Camera(ArduCAM myCAM){      //reads out pixels from the Arducam mini module
//  pinMode(led, OUTPUT); digitalWrite(led, LOW); // Illuminate with blue LED on GPIO2.
  myCAM.clear_fifo_flag();
  myCAM.start_capture();
  while (!myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK));
  Serial.print("Picture captured. ");

  
  size_t len = myCAM.read_fifo_length();
  if (len >= 393216){ Serial.println("Over size."); return;}
  else if (len == 0){ Serial.println("Size is 0."); return;}
  Serial.print("Length in bytes: "); Serial.println(len); Serial.println();
  myCAM.CS_LOW(); myCAM.set_fifo_burst(); SPI.transfer(0xFF);

  
  if (client.connect("myHost", 80)) { 
  while(client.available()) {String line = client.readStringUntil('\r');}  // Empty wifi receive bufffer

    
  String start_request = ""; String end_request = "";
  start_request = start_request + 
  "\n--AaB03x\n" + 
  "Content-Disposition: form-data; name=\"userfile\"; filename=\"arducam.jpg\"\n" + 
  "Content-Transfer-Encoding: binary\n\n";
  
  end_request = end_request + "\n--AaB03x--\n";  // in file upload POST method need to specify arbitrary boundary code

  uint16_t full_length;
  full_length = start_request.length() + len + end_request.length();

    Serial.println("POST /php/arducam/arducam5.php HTTP/1.1");
    Serial.println("Host: myhost");
    Serial.println("Content-Type: multipart/form-data; boundary=AaB03x");
    Serial.print("Content-Length: "); Serial.println(full_length);
    Serial.print(start_request); Serial.println("Here are sent picture data"); Serial.println(end_request); 

    client.println("POST /php/arducam/arducam5.php HTTP/1.1");
    client.println("Host: myhost");
    client.println("Content-Type: multipart/form-data; boundary=AaB03x");
    client.print("Content-Length: ");
    client.println(full_length);
    client.print(start_request);

   
  // Read image data from Arducam mini and send away to internet
  static const size_t bufferSize = 1024; // original value 4096 caused split pictures
  static uint8_t buffer[bufferSize] = {0xFF};
  while (len) {
      size_t will_copy = (len < bufferSize) ? len : bufferSize;
      SPI.transferBytes(&buffer[0], &buffer[0], will_copy);
      if (!client.connected()) break;
      client.write(&buffer[0], will_copy);
      len -= will_copy;
      }
  
     client.println(end_request);
     myCAM.CS_HIGH(); digitalWrite(led, HIGH);
  }

  // Read  the reply from server
  //delay(5000); 
  while(client.available()){ String line = client.readStringUntil('\r'); Serial.print(line);}
  client.stop();  
  while(client.available()){ String line = client.readStringUntil('\r'); Serial.print(line);}
}

void setup() {
  uint8_t vid, pid, temp;
  Wire.begin();
  Serial.begin(115200);
  Serial.println("ArduCAM Mini ESP8266 uploading photo to server");
  
  pinMode(CS, OUTPUT); // set the CS as an output:
  SPI.begin(); // initialize SPI
  SPI.setFrequency(4000000); //4MHz

  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  Serial.println (temp);
  if (temp != 0x55){
  //if (temp != 0x170){
    Serial.println("SPI1 interface Error!");
    while(1);
    }
  

  //Check if the camera module type is OV5642
  myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
  if ((vid != 0x56) || (pid != 0x42))
    Serial.println("Can't find OV5642 module!");
  else
    Serial.println("OV5642 detected.");

 
 
  //Change to JPEG capture mode and initialize the OV5642 module
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  myCAM.OV5642_set_JPEG_size(OV5642_2592x1944);
  myCAM.set_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
  myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00); 
  

  // Connect to a router
  WiFi.mode(WIFI_STA);
  Serial.println("Connecting to AP specified during programming");
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    }
  Serial.print("\r\nWiFi connected IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  }
  

void loop() {

  Camera(myCAM);
  
  Serial.println("Sleeping ...");
  delay(60000);
}
`
```

ArduCAM, 2016-04-28:
@dmolner Please use the new library for 5MP when used with ESP8266, there is a SPI workaround. http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/

labajo, 2016-04-29:
I get the same error.

The last attempt was made using this library.
https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/tree/master/libraries/ArduCAM

I installed the board and I get the same error with the example:

ArduCAM_Mini_OV5642_Capture
```
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
ArduCAM Start!
SPI1 interface Error!

Soft WDT reset

ctx: cont 
sp: 3fff2940 end: 3fff2b50 offset: 01b0

>>>stack>>>
3fff2af0:  3fff1afc 3fff1078 3fff1afc 40202574  
3fff2b00:  feefeffe feefeffe feefeffe feefeffe  
3fff2b10:  feefeffe feefeffe feefeffe feefeffe  
3fff2b20:  feefeffe feefeffe feefeffe 3fff1b28  
3fff2b30:  3fffdad0 00000000 3fff1b20 40207434  
3fff2b40:  feefeffe feefeffe 3fff1b30 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
ArduCAM Start!
SPI1 interface Error!

Soft WDT reset

ctx: cont 
sp: 3fff2940 end: 3fff2b50 offset: 01b0

>>>stack>>>
3fff2af0:  3fff1afc 3fff1078 3fff1afc 40202574  
3fff2b00:  feefeffe feefeffe feefeffe feefeffe  
3fff2b10:  feefeffe feefeffe feefeffe feefeffe  
3fff2b20:  feefeffe feefeffe feefeffe 3fff1b28  
3fff2b30:  3fffdad0 00000000 3fff1b20 40207434  
3fff2b40:  feefeffe feefeffe 3fff1b30 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

I'm using the following schematic:
http://www.arducam.com/arducam-supports-esp8266-arduino-board-wifi-websocket-camera-demo/esp8266_diagram-2/

mkeyno, 2016-05-02:
hi every one , is it possible to convert esp-arducam  to wifi IP camera can support RTSP or ONVIF protocol ?for streaming the video which protocol is fastest  (UDP, TCP, Websocket,.....)?
thanks for your reply

dmolner, 2016-05-02:
@mkeyno Yes is possible but the problem is because is a serial camera and only can see less 1fps.
I test with Ispy and work very good.

mkeyno, 2016-05-02:
@dmolner why serial , all OV76xx have SPI interface , I think all @ArduCAM library work with SPI or I2c, check this link
http://www.aliexpress.com/item/Hot-OV7670-VGA-Camera-Module-Lens-CMOS-640X480-SCCB-W-I2C-Interface-For-Arduino-Factory-Price/32659604527.html?ws_ab_test=searchweb201556_10,searchweb201602_2_10017_10034_10021_507_10022_10020_10009_10008_10018_10019,searchweb201603_9&btsid=076502c5-c8ba-4066-97ab-3f6677f094e4&smToken=e703a6bc3cab4bbe89391d605b7138b1&smSign=7JFCYIz0%2B5KedbbZR8M1Jg%3D%3D
I was hopping can made ip camera with this module and ESP or at least can send the image on motion detection but I dong know how fast this combination will be

dmolner, 2016-05-02:
@mkeyno correct!
I'm sorry!
I confused because use with an ESP8266 and this is a serial module.
for this I only have 1 fps.

Sorry!

labajo, 2016-05-30:
Have you seen the problem @ArduCAM? It is the schematic right?

ArduCAM, 2016-05-31:
@labato, sorry the new example sketches use the new schematic wiring, please check the wiring here; http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/esp8266-pinout/ 
The SPI error will enter while(1) loop and eventually cause the watch dog reset.

AhmadZMQ, 2016-10-06:
Hi @ArduCam,
1.Here I have got a success to shoot video and their saving in SD card by using "ArduCAM_ESP8266_V2_OV2640_Video2SD" Arduino code provided in the ArduCam library but problem arises is to send the video to the server.
2.Also i want to get control on the video timing as per requirment i.e.10s 20s,40s etc. by pressing external key.
How it can be,please suggest the way to solve this issue.Any suggestion from your will be appreciated.
Thanks

mkeyno, 2016-11-12:
dose anyone success  video streaming with OV_7670?

ArduCAM, 2016-11-13:
@mehrdad,
OV7670 deosn't have buildin JPEG engine, this bandwidth is too large to do video streaming. And OV7670 is quite old and low resolution. 
Recommend to use OV2640.

mkeyno, 2016-11-13:
@ArduCAM  I was hoping to build the lowest price IP camera with  OV7670 , so as you said I should switch to another module

ArduCAM, 2016-11-13:
@mehrdad, how do you build the lowest cost IP camera with OV7670?

mkeyno, 2016-11-13:
it was idea base on your repo. long ago I did some effort to directly embed video stream on my website without any third party software or plugin , the result not good as I expected until find what have you done with ESP and OV's modules   but as I didn't find any video streaming example with OV7670, I pasted my question here, but as you said I should pay more for my project

ArduCAM, 2016-11-14:
@mehrdad, I would like to know more about how do you make IP camera with OV7670 even if you said it is not very good.

mkeyno, 2016-11-15:
@ArduCAM  it was plan and  I didn't make anythings yet but I'm going to use more powerful module or just use  OV7670  for motion detection purpose and upload image to the server

ArduCAM, 2016-11-15:
@mehrdad, I think at least you have a basic idea how to stream OV7670 video over IP, then I can give you suggesion on that topic.
From my experience, you need a powerful hardware to do compression and network streaming.

edwardcoyne, 2016-12-20:
I am attempting to interface with a esp32 using the esp8266 codepaths and am encountering the same issue as above where I get 170 back from the SPI initialization test  instead of 85 (0x55). Did anyone ever figure out why? 

I am using the 0V2640 arducam mini.

ArduCAM, 2016-12-22:
@edwardcoyne, current arducam library doesn't support esp32 processor. You have to modify our library to communicate with  esp32 yourself.

edwardcoyne, 2016-12-22:
ya I had figured as much, If I get it working I will do a PR.

 I had just been curious what the cause of the 170 result listed above was to see if it was the same as what was causing my issue. For my case if I switched SPI.setDataMode() to MODE2 or MODE3 it will get the correct result back from that call, I am not certain which it should be though as both seem to work.

I don't have everything working yet but that check passes.