ArduCAM/Arduino #37: Display of the images of an ArduCam Mini 5MP v5642

loziDani, 2016-04-20:
Hi everyone!

I have tried the example provided by ArduCam to make it works and it do it perfect using the ArduCam_host.exe. So I was able to check that the camera works.

My question is the next. I am doing a project where I want to send the image of the camera to a webpage. I don't want to use nothing like ESP8266 modules, I want to send it while my Arduino Nano is connected to my PC via USB, using (in my case) COM3 port.

ArduCam_host.exe leave me to choose which port I've connected my Arduino, so selecting COM3 port, I can see what my camera is displaying.

There is some way to send the image of my Arducam to a webpage in the same way it does to the Arducam_host.exe application?

The main purpose of this is to be able to detect red objects with the camera throught the webpage, and send to my Arduino Nano a command like "if red detected, then turn on the motors" for example.

I have read that Arduino is not able to process images because is out of its posibilities, so I want to do the processing in the webpage.

Maybe someone knows how to send the image to a webpage instead of the Arducam_host.exe based on the ArduCam_Mini_OV5642_Video_Streaming example of Arduino IDE?

I hope to explain well my issue!

Thanks a lot! Any help will be so grateful!

ArduCAM, 2016-04-27:
loziDani  We have update our library for ESP8266 board, please check the latest library and examples from http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/

loziDani, 2016-04-29:
Thanks ArduCAM for answer! I am checking the libraries and I still don't understand how the image is sent to the ArduCAM_host.exe program. maybe sounds stupid, but there is some way to see how ArduCAM.exe takes the image from the port, i mean, can I see the source of the program?

Another question is that, making some experiments, I am able to see what the camera is catching using Serial Port COM3 and ArduCAM_host.exe without wiring all the pins of the camera. I have disconnected SDA and SCL pins and I only have connected CS,MISO,MOSI and SCK pins to my Arduino Nano. This makes me think that SDA and SCL pins are only for control, and MISO (which is Output) sends the image to the Arduino. 

With this conclusions, what I don't know at all is if the bytes of the image are also transmitted throught SDA pin or only throught MISO pin, because I want to make reads of the images of the camera using I2C, asking for the bytes placed in SDA.

Is this possible? If any explanation of what im trying is needed ask me =)

thanks for any help or suggestions

ArduCAM, 2016-04-29:
@loziDani the image data is read via SPI interface rather than I2C, then transfer via usb-serial interface. Please try this link for source code http://www.arducam.com/downloads/app/ArduCAM_Mini_C_sharp_20150413.zip

loziDani, 2016-04-29:
Thanks @ArduCAM !! You have not idea how much this is going to help me to undersand how it is sending the image to COM port. Let me tell you that I am doing my final project using your cam and I have some doubts that only here could be response. And all of you help a lot!

With this camera I am trying to take an image and recognize a color from it. Let me ask you about how the camera send the image please:

I am trying to understand watching the datasheet of the Arducam module and the datasheet of the ov5642 and i am not able to see how the camera send the image, let me to explain me:

Talking with the mentor of my project, he told me that the images have different type of formats (i knew that), and using RGB format, I will be able to receive and image with some bytes, which will correspond with the colors RED, GREEN and BLUE. If I work with this bytes, I will be able to choose what color I want to look for. Something like (the ranges i describe for the colors are unreal):
 
if the pixel is for example: [0x00:0x0F] -> is red
if the pixel is for example: [0x10:0x1F] -> is green
if the pixel is for example: [0x20:0xFF] -> is red

pseudocode:

x = take image(arducam);
if (x>=0x20 && x<=0xFF) then 
    the pixel is red;

Then, if I see some red pixels together, I can say that there is a red object in front of me. This treatment of the image I will do it in the Arduino, PIC or whatever I want to use (via SPI as you said).
My question is:

How the image is sent to the microcontroller? The camera will send 3 bytes (one for each color) with the info of the image and I will be able to treat it? It is possible to configure the communication with the camera to receive the image in RGB to treat it after receive it?

I am thinking in this because if I use the SPI interface, I will send spi_read() instructions, receiving 1 byte each call. This byte could correspond with a color as defined in RGB format? Like for example:

image[0] = spi_read(); //receive red byte
image[1] = spi_read(); //receive green byte
image[2] = spi_read(); //receive blue byte

thanks for any help you could give me. I am sorry for this long question =(. But nothing better that you could help me with this type of complex doubts. I am breaking my head a lot and I know that I will become an expert in your module =D

I hope don't to disturb you with my question =(..

Thanks!!

ShilRenz, 2018-02-03:
good afternoon arducam, I am currently working on a project right now and I have a problem, it is somehow related to this issue.
I have this arduino uno and arducam ov5642 but I am yet a newbe in this programming IDE and i don't know how to display the image from the arducam to my PC. I am hoping that you can help me with this.

thank you!!

supprot, 2018-02-04:
@ShilRenz 
Of course,
You can Download our code from https://github.com/ArduCAM/Arduino.git
For more details, please refer to our vedeo demo:
https://www.youtube.com/watch?v=hybQpjwJ4aA&feature=youtu.be

ShilRenz, 2018-02-05:
@supprot  
thank you so much for the answer, I think this will help a lot.

ShilRenz, 2018-02-05:
[image]

@supprot 
good evening
can I ask why I got this error?

ShilRenz, 2018-02-05:
[image]

@supprot  
good evening, i followed the video you sent me but the camera is not detected. what might be the problem with this?

supprot, 2018-02-06:
@ShilRenz
Hi, 
In the video demo, we use the OV2640 sensor, so we enable the OV2640 and disable others sensor in the memorysaver.h file.If you are using OV5642, you should enable #define OV5642_MINI_5MP_PLUS and disable others. Can you send us your configure in the memorysaver.h file ?
Wtat's more, How do you connect your camera?

ShilRenz, 2018-02-06:
[memorysaver.zip](https://github.com/ArduCAM/Arduino/files/1697426/memorysaver.zip)
@supprot 
this is my mamorysaver.h and I connect the camera to arduino just the same as the video.

[image]

supprot, 2018-02-06:
@ShilRenz

We find your i2C signal connection is error.
Please  swap the SDA and SCL signal .

ShilRenz, 2018-02-06:
OH I see. 
thank you so much, my fault. @supprot

supprot, 2018-02-06:
@ShilRenz 
You are welcome.