ArduCAM/Arduino #38: Help need conect arducam ov2640 with module Sd card

AfroDjeff, 2016-04-21:
Hellow. my problem is. I have an arducam ov2640, i wanna save the photos in a module SD card but when i conect ov2640 and module sd in a arduino mega, arducam doesnt works but whe disconect module Sd arducam works. maybe 1 pin in the code affect the other pls help me thx

ArduCAM, 2016-04-27:
@AfroDjeff Please make sure if your sd card shield is well designed for SPI bus sharing, there are a lot of bad design for the sd card shield on the market.

Buihcmvn, 2016-06-02:
I have the same but it'll be fixed when I use 2 power supply 4 two devices (one for SD and another for Arducam. GND have to connect together). but now comes the next problem. I load this program (ArduCAM_Mini_2MP_TimeElapse2SD) on my Arduino_ Uno. the program run well until this CODE comes (while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));) it stands still. what are ARDUCHIP_TRIG , CAP_DONE_MASK ?? can so help me thx

ArduCAM, 2016-06-05:
@Bulhcmvn
You have to use external power supply, because 4 camera consume a lot of power when running simultaneously.
CAP_DONE_MASK is not set because of the camera is not initialized properly. Please try the external power supply or make sure one camera works before adding more cameras.

Buihcmvn, 2016-06-29:
at first. Thanks for your answer, but my problem still there.
when i connect Arducam like the first picture (with out cable except power supply) it works very well
[image]

but when i connect like the under picture (with external power supply 5V-1A)
[image]

then become 
[image]

the message "Capture Done!    Open file failed" comes when i reset external power supply (OFF in 5s and ON again)

what is here problem ??? --> thank for your help

ArduCAM, 2016-07-06:
@Buihcmvn, from the error message it seems that you did the wrong connection.
Please check your wiring and make sure the wiring match with the pin  used in the example.