ArduCAM/Arduino #39: Windows source code and/or description for the ArduCAM_4CAM.exe

ad5jo, 2016-04-22:
Is there a description of the code used to pull the picture data from the ArduCAM-M-5MP?

I downloaded the code from GitHub. The source code didn't seem to be provided.

It seems that the resolution is 320x240 and not anything close to the resolution of the 5MP camera with the supplied executable: ArduCAM_4CAM.exe 413
Selecting different resolutions didn't seem to change the image quality.

ArduCAM, 2016-04-22:
@Dave, you can find the source code for the windows host application from: www.arducam.com/downloads/app/ArduCAM_Mini_C#_20150413.zip
The host app send change resolution command to the arduino, then arduino send the i2c sequence to change the image resolution.
The only possible reason, is the camera not recognized by the host app. Please try the example code without change anything  and try again. 
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_OV5642_Video_Streaming

ad5jo, 2016-04-23:
Hello Lee,
The link to the source code is not working:
This is somewhat embarrassing, isn’t it?
It seems we can’t find what you’re looking for. Perhaps searching, or one of the links below, can help
Can you double check the spelling or correct the URL?

Thank you,
Dave

ArduCAM, 2016-04-23:
@Dave, sorry for the trouble.
Please forgive me if I response late or made any mistake, I have massive work to do here, please understand.
Please try this link http://www.arducam.com/downloads/app/ArduCAM_Mini_C_sharp_20150413.zip

ad5jo, 2016-04-23:
Hi Lee,
That link worked. Thanks for the quick reply.
Dave

ad5jo, 2016-04-23:
Hi Lee,
The exe built from the source code does work properly. Thanks for supplying the source code.
I think the exe that gets downloaded in the zip file is broken.
For others interested, I used the Micro Soft Visual Studio 2013 Update 5 Community. It is free for non-professional use or business use if you revenue is under $1 M per year.. I think. 
Lee's project is written in C#.
The only thing I did to compile the release, is to create a test certificate. This task is located by clicking PROJECT, arduCAM properties..., Signing.
Some options:
1. uncheck the box 'Sign the ClickOnce manifest' or
2. Create Test Certificate... button or
3. If you already have a certificate you don't need instruction from me. Select you certificate.

Thank you,
Dave Durkee

closed 2016-04-23 by ad5jo