ArduCAM/Arduino #40: arduicam mini connection with gprs

meshboy, 2016-04-23:
pls, how can i connect the device with gprs for real time streaming instead on viewing locally

ArduCAM, 2016-04-25:
@meshboy, you can send the image stream to the GPRS device over uart, and you have to learn how to transfer data over gprs youself.

meshboy, 2016-04-25:
Thank you very much.

---------------------------------------------------------------------------- MESHILEYA, Emmanuel Seun
Software Developer | 08164663359 | linkedin | facebook

claymelugin, 2016-07-01:
meshy consider using the Particle Electron, it makes cellular connection far easier, and it supports the existing code that is Arduino compatible.  I'm not done yet but its all there for me to get it working.