ArduCAM/Arduino #42: Blurry image when used in motion, Exposure time adjustment?

gonzoveliki, 2016-05-13:
Hello, 
recently I've purchased few ArduCAM mini modules OV2640 and OV5642...
The goal is for evaluating the camera module build quality and possibility to be used in a mass production devices with both 2MP and 5MP variants.
So far I am amazed with the products overall quality, PCB board,  soldering, lenses, product packaging software support and documentation - everything is better than all other competitive products.
However, during the tests I've figured out, that camera performs perfect when stationary, and produces
a blurry image, even when moved slowly during footage. This effect is observed both with 2MP i 5MP devices, and with various resolutions. Is there any way for setting exposure time by software?
In the provided libraries all registers are configured via table based values, for each model and resolution.
How can we find detailed description of these values, and which are responsible for fine tune
of colour tint, white balance, scan modes and resolutions, and exposure times as well.
Thank you
Best Regards

ArduCAM, 2016-05-13:
@gonzoveliki Customized camera settings like resolution, frame rate, exposure and etc can be supported, please contact admin@arducam.com for detail.

gonzoveliki, 2016-05-15:
Hello, I've addressed this to admin@arducam.com and so far there is no reply.
Please advice where I can find detailed registers description of the image sensor (via i2c) and 
capture FPGA (via spi)...
Regards

Displav, 2016-05-21:
Hello Lee,

This is an issue that me as well would like to solve.

Could you please give some more information?
Is it possible to make the adjustments on the cameras we already have or there is not a way to do this?

Also I would like to know if the compression rate is possible to be adjusted.

ArduCAM, 2016-05-24:
@Display, Four 5MP camera consume a lot of power when working at the same time, that's why we design low power mode for that, but there will be peak current when start them together.
For adjust compression, please change the register 0x4407 value, 0x04 is good, 0x08 is fair, 0x0C is low quanlity. Please make sure the total size doesn't exceed the total frame buffer.

Displav, 2016-05-24:
What about the Exposure time @gonzoveliki  asked about?
Could you give some more informations about that as well?

Displav, 2016-06-04:
Hello again!

@gonzoveliki  Did you get an any info for the exposure from @ArduCAM  Lee?

Please provide us with any more info about this...  It would make a great difference in quality to have the exposure time a little bit faster!

Thank you very much!

wmelton, 2016-06-16:
@gonzoveliki - were setting information shared regarding exposure, etc?

MarkMMullin, 2016-08-08:
These are handy for figuring out the registers  -- http://www.uctronics.com/download/cam_module/OV2640DS.pdf and http://www.uctronics.com/download/cam_module/1Inch4_5_Megapixel_OV5642_CMOS_Camera_Module_DS_V1.1.pdf

aldobranti, 2016-08-31:
@Displav @gonzoveliki 
: I am adding my voice to your call to get this into the API
I can read this http://www.dragonwake.com/download/camera/ov5642/OV5642-DS.pdf until I go blue in the face but I'm sure @ArduCAM could do this in 20 minutes @dis

ArduCAM, 2016-09-01:
@aldobranti, It is not very easy to raise the frame rate even if the datasheet says you can. There is limited info about the sensor, we have to build the settings from scratch, and gain a little bit more experience than you though.
You need to pay a lot of time to trial and fail.

aldobranti, 2016-09-01:
Thankyou for explaining @Arducam. I am not the only user who would value this feature. .A. 

 adjustment? (#42)

https://github.com/ArduCAM/Arduino/issues/42#issuecomment-243943766

wdouglas1999, 2017-04-09:
Hi, I am too evaluating the OV5642 and have the same blurry images with movement. I have the version with 8Mbyte FIFO. My test program is as follows. 
I capture multiple images until the fifo is full.
Then I copy the images to SD card.
When I look at the images I see a few blurry images when there is movement.
From my observation since the images are stored directly to the FIFO it seems that this is an issue with the arducamchip and OV5642.
Has anyone managed to get this fixed some how?