ArduCAM/Arduino #44: ArduCAM Mini 2MP picture out of focus

gregeverett, 2016-05-23:
Hi, I have an ArduCAM Mini 2MP and I first set it up on an Arduino Uno, and using the ArduCAM_v1.0 host application I could take Single Shoot and Continuous Shoot no problem, and the picture was in focus and good. 
I tried to setup using the "ESP8266 ESP-12E WIFI Wireless Dev Board for Arduino IDE UNO WeMos D1" board as I want the WiFi and websocket functions.  Got this working as well but the pictured is blurred and out of focus.
I change the Mini 2MP back to the Uno board and this is now also out of focus.
Have tried different patch cables between the board and camera in case of a broken wire but still out of focus.
Any ideas what has happened?  Is seems to be an issue with the camera?
[image]

ArduCAM, 2016-05-23:
@gregeverett,
It is common issue that when you moving the camera and touched the lens accidently, then the lens is out of focus.
In order to refocus the lens, using the ArduCAM_v1.0 host application and switching to continuous shoot mode, rotate the lens back and forth to focus at the object.

gregeverett, 2016-05-23:
Hi Lee, this has sorted it out.  I did not realise there was a manual focus on the lense.  Thanks!
I have also ordered your ESP8266 ESP-12E UNO Board for ArduCAM Mini Camera compatible Arduino UNO R3 board for further testing.  thanks again, Greg

closed 2016-05-23 by gregeverett