ArduCAM/Arduino #45: No output on LCD screen after uploading.

WOL7, 2016-05-23:
Hello. I purchased the Arducam RevC+ shield and OV5642 5MP camera module recently. I connected it to the Arduino Uno R3 but after compiling the code in the 1.5.5r-2 IDE, I got nothing on the LCD screen. I also noticed that the UTFT_SPI and itoa were not highlighted in orange like the other library file names in the #include directives in the early lines of the code, yet these files exist in the libraries folder of the compiler. I'm not sure what the problem is, especially after the code can successfully be compiled and uploaded to the arduino. I'd be most grateful for any help with this.

WOL7, 2016-05-23:
Here is the example file I'm using for the digital camera function, which I got from your repository.
[ArduCAM_OV5642_Digital_Camera.txt](https://github.com/ArduCAM/Arduino/files/277031/ArduCAM_OV5642_Digital_Camera.txt)

ArduCAM, 2016-05-24:
@WOL7, you need to copy both ArduCAM and UTFT_SPI libraries to the Arduino library directory. 
And you have to modify the memory_saver.h file to uncomment the line to enable OV5642 camera module then compile the example sketch.
You can open the serial monitor to view the debug information printed from the UNO board to figure out where the problem is.

WOL7, 2016-05-28:
Thanks for the reply, @Lee. I had the first thing and then I uncommented the ov5642 line in memorysaver.h but my screen is still black. Is this because I don't have the lens mount? The serial monitor also displays "Arducam Start!" And then some other characters that don't make sense...

WOL7, 2016-05-28:
I realised I was using an OV2640 instead of a OV5642 and made he necessary changes. Thanks @Lee.