ArduCAM/Arduino #46: 5MP OV5642

blpanther, 2016-05-24:
Hi, I have bought the 5MP OV5642 Sensor Arducam Mini Camera Shield JPEG output for Arduino Mega2560 - hoping that I can actually use it with arduino.

After spending countless amount of time looking for a code to use it, way to connect I kind of gave up....
Why is it so hard to provide a simple instructions to follow? Even to find how it should actually be connected to arduino uno - probably most popular is quite a task.....
I downloaded all sample codes I could find to just be able to see the picture from the camera - none of them will actually verify in arduino ide.... 
Can I please ask for a link to file which works? I had countless amount of errors, last one I tried is:
ArduCAM_Mini_OV5642_Video_Streaming.ino
which comes with following errors:

/Users/xeon/Desktop/ArduCAM_Mini_OV5642_Video_Streaming/ArduCAM_Mini_OV5642_Video_Streaming.ino: In function 'void setup()':
ArduCAM_Mini_OV5642_Video_Streaming:76: error: 'OV5642_CHIPID_HIGH' was not declared in this scope
   myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
                         ^
ArduCAM_Mini_OV5642_Video_Streaming:77: error: 'OV5642_CHIPID_LOW' was not declared in this scope
   myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
                         ^
exit status 1
'OV5642_CHIPID_HIGH' was not declared in this scope

many thanks for any advice - I really would like nothing more than to see the picture from the camera....

rkhanna24, 2016-05-25:
The issue you specifically raised ( 'OV5642_CHIPID_HIGH' was not declared in this scope) may be resolved by modifying "memory_saver.h" to uncomment the preproccessor directive corresponding to your device. I'm assuming you are using the OV5642, so you should uncomment "#define OV5642_CAM" and  comment the other #define's. Hopefully this helps you to see a pic!

blpanther, 2016-05-25:
Many thanks...
I have made the change in memory_saver.h
Now I get the following error:

sketch/ArduCAM.cpp: In member function 'void ArduCAM::OV5642_set_JPEG_size(uint8_t)':
ArduCAM.cpp:619: error: 'ov5642_dvp_fmt_global_init' was not declared in this scope
  wrSensorRegs16_8(ov5642_dvp_fmt_global_init); 
                   ^
ArduCAM.cpp:624: error: 'ov5642_dvp_fmt_jpeg_qvga' was not declared in this scope
    wrSensorRegs16_8(ov5642_dvp_fmt_jpeg_qvga);
                     ^
ArduCAM.cpp:631: error: 'ov5642_dvp_fmt_jpeg_vga' was not declared in this scope
    wrSensorRegs16_8(ov5642_dvp_fmt_jpeg_vga);
                     ^
ArduCAM.cpp:638: error: 'ov5642_res_720P' was not declared in this scope
    wrSensorRegs16_8(ov5642_res_720P);
                     ^
ArduCAM.cpp:645: error: 'ov5642_res_1080P' was not declared in this scope
    wrSensorRegs16_8(ov5642_res_1080P);
                     ^
ArduCAM.cpp:651: error: 'ov5642_dvp_fmt_jpeg_qxga' was not declared in this scope
    wrSensorRegs16_8(ov5642_dvp_fmt_jpeg_qxga);
                     ^
ArduCAM.cpp:657: error: 'ov5642_dvp_fmt_jpeg_5M' was not declared in this scope
    wrSensorRegs16_8(ov5642_dvp_fmt_jpeg_5M);
                     ^
sketch/ArduCAM.cpp: In member function 'void ArduCAM::InitCAM()':
ArduCAM.cpp:813: error: 'ov5642_dvp_fmt_global_init' was not declared in this scope
     wrSensorRegs16_8(ov5642_dvp_fmt_global_init); 
                      ^
ArduCAM.cpp:815: error: 'ov5642_dvp_fmt_jpeg_qvga' was not declared in this scope
     wrSensorRegs16_8(ov5642_dvp_fmt_jpeg_qvga); 
                      ^
exit status 1
'ov5642_dvp_fmt_global_init' was not declared in this scope

ArduCAM, 2016-05-26:
@blpanther, did you put the libraries in the right folder? They should be right under /arduino-1.6.7/libraries directly like this

blpanther, 2016-05-26:
It is arduino 1.6.9,  the screenshot of libraries is attached.
Would you mind confirming which arduino sketch is the easiest to get to work?
Many thanks
Raf
[image]

losttime, 2016-05-26:
@blpanther How did you download the ArduCAM library? You don't seem to have all the files and the file names don't match 100%.

You can download from the [root of the project](https://github.com/ArduCAM/Arduino) by clicking the Download button in the top right (possibly "Clone or download"). That should get you a zip file. Unzip that file and inside you should have 3 directories, `ArduCAM`, `OV7670FIFO`, and `UTFT4ArduCAM_SPI` (along with some other stray files). That `ArduCAM` directory is what you want, and you want to put it in your `libraries` directory (don't put just the contents of the `ArduCAM` directory in your `libraries` directory - copy the full directory).

You'll end up with something like this:
```
libraries/ArduCAM/ArduCAM.cpp
libraries/ArduCAM/ArduCAM.h
libraries/ArduCAM/memorysaver.h
libraries/ArduCAM/ov5642.h
...
and all the other files in the ArduCAM library
```

If you don't want to redownload, you can probably build this back up from what you have now by turning what you have now into what's listed above.

1. Create and ArduCAM directory in your libraries directory, and put the arducam files in that new directory.
2. Rename `OV5642_regs.h` to `ov5642_regs.h` (lowercase "o" and "v") so it matches the `#include` found in `memorysaver.h`

I don't know if these will do the trick, but they might and will help keep the library in line with what others have and make further debugging easier.

ArduCAM, 2016-05-27:
@Blpanther and David, thank your David, you give more detail description on the directory structure for the libraries.
I think it is the library convention for the Arduino like compiler.

bebetin, 2017-04-21:
@blpanther 

hi, did you ever get your camera to work? 
if so, how were you able to see the pictures and what was your main source code?
im going through the same problems you did.

bebetin, 2017-04-24:
@ArduCAM 

Hi, Im pretty sure i have set up all the libraries properly. only thing i dont know is how to view the pictures. since i dont have a memory card, how can i view the pictures?

supprot, 2017-04-24:
@blpanther 
Hi,
If you don't have SD card,we suggest you use our Host_V2 and our videostreaming demo.
You can see the picture from the host_V2
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows
Let us know if you need more help.
ArduCAM support team.

bebetin, 2017-04-29:
@supprot 
hi,

i ran the host v2 demo and it recognizes that theres an arduine plugged in; but theres no image coming up on it. is there a certain way the camera needs to be hooked up to the arduino,or is there a souce code that needs to be u[loaded to it? theres no instructions on how to use it and i tried to get it to work but simply couldnt. any pointers?

adrier, 2017-10-22:
Hi @losttime and @ArduCAM 

I'm in the same situation as @blpanther.
I have downloaded and sorted the libraries as required and nether the SD or any of the apps seems to be working, even if Arduino (Uno) seems to be connecting OK with the cam (OV5642).
'memorysaver.h' is set up as you advised.
I currently have the error "'VSYNC_LEVEL_MASK' was not declared in this scope" with the file ArduCAM_OV5642_Digital_Camera.ino

Could you help me to set the camera up please?

Thank you

supprot, 2017-10-22:
@adrier
Hi,
You are using our old demo,We have update our demos.Please download them from 
https://github.com/ArduCAM/Arduino.git
Let us know if you need more help.
Regards.