ArduCAM/Arduino #47: Poor picture with 5MP mini hooked up to Sparkfun Thing

losttime, 2016-05-26:
I've got a 5MP ArduCAM mini hooked up to a [Sparkfun Thing](https://www.sparkfun.com/products/13231). I've been able to get it to take a picture and upload it to my webserver, but the picture is streaked and has odd artifacts.

[image]

Any thoughts on what I can do to get a better picture?

losttime, 2016-05-26:
In order to get the 5MP mini working on the Sparkfun Thing I started from the [ArduCAM_Mini_OV5642_LowPowerMode example](https://github.com/ArduCAM/Arduino/blob/23d6acb20b21e5a501b12abff5715f8797a0096d/ArduCAM/examples/mini/ArduCAM_Mini_OV5642_LowPowerMode/ArduCAM_Mini_OV5642_LowPowerMode.ino), added in parts of the [ArduCAM_ESP8266_OV5642_Capture example](https://github.com/ArduCAM/Arduino/blob/37516993bb6d97ef83cbfe7943dce192b88754dc/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_OV5642_Capture/ArduCAM_ESP8266_OV5642_Capture.ino), changed it up so it's a client instead of a server, and had to make a few specific modifications.

1. Comment out the bit rotation "correction" parts of the library - helps in stopping "SPI1 interface Error!" and "Can't find OV5642 module!" errors
  1. [ArduCAM.cpp - line 185](https://github.com/ArduCAM/Arduino/blob/37516993bb6d97ef83cbfe7943dce192b88754dc/ArduCAM/ArduCAM.cpp#L185)
  2. [ArduCAM.cpp - line 719](https://github.com/ArduCAM/Arduino/blob/37516993bb6d97ef83cbfe7943dce192b88754dc/ArduCAM/ArduCAM.cpp#L719)
  3. I guess these aren't needed for the Thing (?)
2. Set buffer size to 2048 - stops the picture from "tearing"

There are likely other things that didn't carry over as part of combining the two examples.

MarkMMullin, 2016-08-08:
I am working with 2640's, however I have seen all of them first show a strong green tint to the first captured image, even after a 1 second wait post arducam initialization and sensor register program load - the lab I work in is darker, so the image has an overall green cast, however this disappears on subsequent images -  I believe it's due to white balance, but I haven't worked it out in detail.  But I'd wonder if that behavior is consistent or stabilizes (somewhat) after the first few images. If you want I can provide access to some images that I've tried to share with the Arducam folks, however those images also have a LOT more defects (noise, problems with the rolling shutter, and really sad all black images with noise)