ArduCAM/Arduino #48: SPI1 Interface Error with ESP8266

losttime, 2016-05-27:
I'm using the ArduCAM library with a [Sparkfun Thing](https://www.sparkfun.com/products/13231). I believe the bit rotation fix for ESP8266 is causing a persistent "SPI Interface Error", at least for this model of ESP8266 board.

Here's some test code:

```c
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

const int CS = 16;

ArduCAM myCAM(OV5642, CS);

bool spiWorks() {
  int testValue = 0x55;
  myCAM.write_reg(ARDUCHIP_TEST1, testValue);
  uint8_t temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp == testValue){
    return true;
  }
  Serial.println("SPI1 interface Error!");
  Serial.print("Expected: ");
  Serial.println(testValue, 2);
  Serial.print("Got: ");
  Serial.println(temp, 2);
  return false;
}

void setup() {
  pinMode(CS, OUTPUT);
  Wire.begin(4, 5);
  
  Serial.begin(115200);
  Serial.println("ArduCAM Start!");

  SPI.begin();
  SPI.setFrequency(4000000); //4MHz

  if (spiWorks()) {
    Serial.println("Can do stuff?");
  } else {
    Serial.println("Cannot proceed");
  }
}

void loop() {
}
```

As you set different values for `testValue` in the `spiWorks()` function you can see that the result read back from the register (`temp`) is the same value that was sent it, but rotated right by 1 bit.

```
Expected:1010101
Got: 10101010

Expected: 1
Got: 10000000
```

That rotation is happening in `ArduCAM.cpp` to get around some ESP8266 issues. Maybe those issues don't exist for this board?

I tried rotating the value being set and that gets around this problem, but causes other problems down the road. In fact, this board works well without any bit rotation "fixes" at all.

I wonder if there's a better was to solve the ESP8266 bit rotation issue so there won't need to be specific fixes for specific boards.

ArduCAM, 2016-05-27:
@David,
We use our own ESP8266 board for the OV5642 camera http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/ 
But I think it is the same thing as the Sparkfun board. We also try to update our firmware on the camera board to avoid the bit rotation, for this moment we have to stick with the software bit rotation.
Do you want me send you a free sample ESP8266 UNO board for testing. You can send me your shipping address to admin@arducam.com

losttime, 2016-05-27:
So the bit rotation is probably due to the camera module firmware, not the ESP8266 board?

Are there different versions of the 5MP mini firmware out in the wild? Maybe older ones require bit rotation but newer ones do not? Or should all 5MP mini modules require the bit rotation? My module appears to not need it (but it is having image quality issues, so I could be wrong #47)

aelkman, 2016-05-28:
Hello I am having trouble with a CC3200 Launchpad with the same error. I tried your test script and temp keeps returning '0'. I am wondering if my camera is broken. I'm using a OV2640 2MP ArduCAM mini. I have tried powering with both 3.3V and 5V sources. The SPI seems to be sending out signals. Here is a picture of my MOSI signal with the clock, does this seem right?

[Signal](http://i.imgur.com/o0sRiuN.png)

Sorry to jump into your thread, but I've got a final project due for class and I'm not sure what is wrong...

losttime, 2016-05-30:
@aelkman It doesn't look like a bit rotation issue exactly, but I don't have enough experience with the hardware side of things to have much of a clue what's going on.

jahearnco, 2017-06-08:
Not sure if this helps, but I just got the same error(s) before I realized that I had improperly grounded the camera. In another thread the issue was solved by grounding SPI pins via resistors. Possible those resistors set ground reference that wasn't set already for the camera (i.e. it's potential might be floating?).

uhwsin, 2021-03-03:
Try set the bit direction use MSBFIRST or LSBFIRST