ArduCAM/Arduino #49: SPI Interface error with CC3200 and ArduMini 2MP (OV2640)

aelkman, 2016-05-28:
Please follow the link below for all the details.

[SPI.txt](https://github.com/ArduCAM/Arduino/files/287689/SPI.txt)
(save as .cpp for nice formatting)

Had some problems with the pinmux setup, so I used my own MAP commands from other projects instead... still no success. 

I'll attach my SPI.cpp file for reference here.

[SPI1 interface Error!](http://forum.arduino.cc/index.php?topic=403505.0)

aelkman, 2016-05-28:
[](url)

aelkman, 2016-05-30:
Hey I'm having further issues, nowhere online does there seem to be replicated problems of what I'm experiencing, maybe you can help me debug:

Write (top) then read (bottom): http://imgur.com/mUkv8Po
Signal for write looks 100% correct. 

SPI Settings: http://i.imgur.com/YFd67BW.png
Note I am using Mode 1 since the device says to use POL=0, PHA=1
@4MHz (ArduCAM has a max clock freq of 8MHz should not be an issue)

Test sketch to try and echo SPI in/out signal: http://i.imgur.com/KeZ7IiT.png
Shows the 0x67 DATA write command at ADDR 0x00

Results on terminal from the sketch: http://i.imgur.com/B0Mm2E2.png
The results are varying, sometimes 0 sometimes 111111 and sometimes just 1111

I've talked to 3 other people about this, no one has any clue what could be wrong.

ArduCAM, 2016-05-30:
@aelkman, I think you should SPI mode 0 which is POL = 0, and PHA = 0, there is a mistake on the document.

aelkman, 2016-05-30:
Ok I am using Mode 0 here. The MISO looks 100% correct now to me unless that is a problem that it is sending data during the CMD phase? It says it should be blank in the technical document.

Here is the write: http://i.imgur.com/UiUWeLS.png
Here is the read: http://i.imgur.com/bGMV3Y7.png

aelkman, 2016-05-30:
Ok I changed it to Mode 0, the signal looks even better now

connorhart, 2016-07-15:
I used to work with old generic floppies (5 1/4) on the 1541 hd for commodore64. I was in 7th grade and the projects were my own. I just wanted to write an array, read it back, make changes to the data, write it, and when read back have consistency as would be expected.  I returned to that little project many times over the months and never no matter how much I simplified the elements of what was needed could I get consistent results. Time passed..years..but I just morphed with collected data that most maybe every single routine I wrote worked perfectly. One spot on the very cheap floppies and it failed..the 1541 drive was probably not the best at hitting the target which was poorly marked too..it wasn't me it was the faulty equipment even maybe multiple faults...so, look for faults in equipment not your design.