ArduCAM/Arduino #50: SPI Issues with ArduCAM Mini

donfelipex, 2016-06-10:
I would like clarification on how to properly interface with the ArduCAM mini after reading the documentation. I am attempting to integrate it into an embedded system with an LCD, SD card, and an IR camera. I am using a Teensy MCU. I have encountered major issues attempting to capture an image directly to an SD card with this system. I can see by other issue posts that I am not alone in this. 

What are the SPI settings to properly communicate with the device? In the Arduino example code the SPI bus is never configured. Looking at the documentation for the hardware application note I find this: "The ArduCAM SPI slave interface is fixed SPI mode 0 with POL = 0 and PHA = 1." That information is contradictory since POL=0 and PHA=1 are defined as SPI mode 1(https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus#Mode_numbers). Is it mode 1 or 0? Is it MSB of LSB first? What clock speeds will it work with? I would like to use transactional SPI but given the ambiguity in the documentation it has been difficult to implement. 

Operating the device at 5V adds significant noise to my system. Any device on it's MOSI line has its communication distorted greatly. Looking at the SPI signals(CS, MISO, MOSI) on an oscilloscope, I see that the device's signal low is at 2V(the device is sourcing current) and not 0V. This problem is not present when operating the device at 3.3V or lower voltages. How is its voltage translation implemented? Is it using a clamp diode? This noise is a source of distress because it makes it impossible to use another device on the same bus as the Arducam. 

In the ArduCAM library, are the functions CS_LOW() and CS_HIGH() just enabling and disabling the chosen chip select for the device? My system has a decoder so I have to modify the library myself, but I am unclear what is being done with the "cbi(P_CS, B_CS)" and "sbi(P_CS, B_CS)" commands.

ArduCAM, 2016-06-10:
@donfelipex, thanks for your comment.
There is mistyping on the document, it should be SPI mode 0 POL=0 and PHA=0, not mode 1.
You can use 3.3V or 5V for the power supply depends on your backend processor's voltage, for example Arduino UNO/MEGA need 5V while Arduino DUE need 3.3V. We use MOSFET as a level shifter. 
CS_LOW() and CS_HIGH() functions are used to toggle the CS pins, this is software CS pin, and we may use multiple CS pins for multiple cameras on the same SPI bus. So for each SPI transaction, you have to pull the CS low and pull it back to high when completion.
You can also send me email to admin@arducam.com if you have more questions.

donfelipex, 2016-06-11:
Thanks for the response. What clock speed should the SPI run at? 
I understand that it can be run at different voltages. My issue is it that running it at 5V I get a serious amount of noise that is not present when running the system at 3.3V. In 5V operation, the ArduCAM is sourcing current, pulling the SPI lines to 2V when at logic level low. Is this expected? I have isolated CS and MISO lines, but shared MOSI and CLK lines. Adding the ArduCAM to the system distorts these signals to an unusable level.

MarkMMullin, 2016-08-08:
I've gone through 16 Arducam 2640's at this point and the one comment I can add is that no Arducam has been capable of reliable communications at 8Mhz -  I use the Beaglebone which can only do clock rates at a power of 2, and 4 Mhz seems to work fine, at least for comms -  I think a number of other systems also have SPI clocking that only does powers of two - would love to see pics of the the scope traces you discuss so I can compare them with what I'm seeing