ArduCAM/Arduino #53: What is the minimum number of bytes that have to be read when using set_fifo_burst()?

DasHammer, 2016-06-14:
What is the minimum number of bytes that have to be read when using set_fifo_burst()? Is it possible to read a single image with multiple burst transactions? Do we need to discard the first byte of only the first transaction? or do we need to discard the first byte of every transaction?

I am trying to read a single image with multiple burst transactions:

```
   uint32_t imageSizeInBytes = myCAM.read_fifo_length();
   bool first = true;
   while (imageSizeInBytes > 0)
   {
      uint32_t numBytesToFetchFromCamera = (imageSizeInBytes < bufferSize) ? imageSizeInBytes : bufferSize;
      myCAM.CS_LOW();
      myCAM.set_fifo_burst();
      if(first) SPI.transfer(0x00); //discard first byte read.
      first = false;
      SPI.transfer(&buffer[0], numBytesToFetchFromCamera);
      myCAM.CS_HIGH();
      imageSizeInBytes -= numBytesToFetchFromCamera;
      //Send image chunk via radio here...
   }

```

ArduCAM, 2016-06-14:
@DasHammer,
As you can see from the user guide 
You can break the burst read into small amount multiple burst transactions, see below picture.
The very first burst transaction should ignore the first dummy byte, and after that all rest of burst transaction should use the first byte as normal image data.
You can see the demonstration example https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_2MP_TimeElapse2SD/ArduCAM_Mini_2MP_TimeElapse2SD.ino 
line 130 to line 161, which burst read 256 bytes chunk data at a time and write to sd card, and read another 256 bytes and so on. It is exactly what you need.

DasHammer, 2016-06-15:
Thanks! I have another question. I noticed that the JPEG images returned by
the camera don't always start with 0xFF 0xD8. Is this expected?

closed 2016-06-15 by DasHammer

ArduCAM, 2016-06-15:
@DasHammer, it should always start with FF D8 and end with FF D9, let me know what did you see.

DasHammer, 2016-06-15:
Here's the raw data from a 160x120 JPEG:

255,248,255,224,0,16,74,70,73,70,0,1,1,1,0,0,0,0,0,0,255,251,0,67,0,12,8,9,11,9,8,12,11,10,11,14,13,12,14,18,30,20,18,17,17,18,37,26,28,22,31,44,38,46,45,43,38,42,41,48,54,69,59,48,51,65,52,41,42,60,82,63,65,71,74,77,78,77,47,62,85,91,84,75,90,69,76,77,74,255,251,0,67,1,13,14,14,18,16,18,35,20,20,35,74,50,42,50,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,74,255,196,0,31,0,0,1,5,1,1,1,1,1,1,0,0,0,0,0,0,0,0,1,2,3,4,5,6,7,8,9,10,11,255,196,0,181,16,0,2,1,3,3,2,4,3,5,5,4,4,0,0,1,127,1,2,3,0,4,17,5,18,33,49,65,6,19,81,97,7,34,113,20,50,129,145,161,8,35,66,177,193,21,82,209,248,36,51,98,114,130,9,10,22,23,24,25,26,55,55,39,40,57,46,52,61,54,55,60,61,58,67,68,69,70,71,72,73,74,83,84,85,86,87,88,89,90,99,100,101,102,103,104,105,106,115,116,117,118,119,124,121,126,131,132,133,134,135,136,137,138,146,147,148,149,150,151,152,153,154,162,163,164,165,166,167,168,169,170,178,179,180,181,182,183,188,189,190,226,195,228,229,198,199,232,201,202,242,211,212,245,214,215,248,249,250,241,226,243,244,229,246,247,232,249,250,241,250,251,244,253,254,247,248,253,254,255,228,0,31,1,0,3,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,1,2,3,4,5,6,7,8,9,10,11,255,228,0,181,17,0,2,1,2,4,4,3,4,7,5,4,4,0,1,2,119,0,1,2,3,17,4,5,33,49,6,18,65,81,7,97,113,19,34,50,129,8,20,66,145,161,177,193,9,35,51,82,248,21,98,114,209,10,22,36,52,241,37,241,23,24,25,26,38,39,40,41,42,53,54,55,56,61,58,67,68,69,70,71,72,73,74,83,84,85,86,87,88,89,90,99,100,101,102,103,104,105,106,115,116,125,119,127,124,125,122,194,195,132,197,198,135,200,201,138,146,147,212,149,214,151,152,153,154,162,163,164,165,166,167,232,169,170,178,179,180,181,182,183,184,189,190,226,195,228,229,198,199,232,201,202,242,211,244,245,246,215,216,249,250,242,227,244,245,230,247,248,233,250,250,243,252,253,246,255,252,253,254,255,224,0,17,8,0,124,0,160,3,1,33,0,2,17,1,3,17,1,255,250,0,12,3,1,0,2,17,3,17,0,63,0,251,209,45,72,175,94,87,50,28,48,104,17,41,231,52,89,23,97,219,83,24,164,17,99,149,126,171,118,42,193,44,94,96,225,21,154,240,58,203,179,29,122,85,199,82,100,172,118,122,69,159,249,109,21,15,94,244,217,87,255,34,74,246,148,125,32,144,172,96,107,195,152,127,134,178,43,205,175,254,66,75,254,23,252,133,173,255,0,255,173,255,104,127,196,238,111,173,122,24,79,241,19,45,139,178,255,0,232,16,41,35,63,45,100,87,160,101,49,41,149,44,206,196,102,163,53,207,50,74,6,149,26,191,108,252,36,221,78,222,106,75,184,155,237,56,61,51,199,9,15,75,21,231,215,13,128,72,235,132,158,63,241,155,118,126,252,127,113,42,255,34,165,105,18,105,159,243,61,83,94,191,44,74,171,250,54,140,15,16,143,253,31,141,99,87,37,127,226,24,179,71,227,255,0,250,24,182,255,0,126,190,61,109,63,226,103,55,214,191,44,18,191,33,63,132,167,233,251,73,138,252,54,48,99,77,70,106,25,4,102,163,106,229,153,37,70,76,244,124,175,156,59,71,45,72,49,72,104,54,208,69,34,195,20,157,40,1,67,84,240,220,52,103,131,77,54,182,26,99,175,179,127,137,131,251,37,100,188,109,27,109,97,131,93,60,254,254,147,53,212,208,248,247,254,134,237,127,255,174,167,93,76,106,179,255,107,219,203,255,0,130,102,255,19,47,105,163,128,13,247,25,161,88,72,191,151,145,93,127,210,47,126,157,76,68,53,27,80,254,37,145,53,68,107,154,95,200,43,154,97,90,249,150,119,12,229,57,104,98,68,240,211,230,40,54,66,252,205,52,197,72,171,12,49,26,102,242,40,51,176,244,108,84,173,178,125,219,32,254,106,147,243,31,217,38,195,110,240,251,254,153,249,255,131,93,79,187,30,253,102,85,245,106,254,76,191,252,38,79,68,89,255,0,132,126,198,77,190,126,153,72,245,53,29,254,137,107,5,163,190,25,77,157,171,208,116,97,249,245,37,193,28,251,84,70,176,145,238,68,106,9,14,5,114,214,209,18,202,240,246,158,43,246,142,212,5,105,152,164,88,180,248,244,13,19,169,169,120,170,76,217,14,194,209,245,41,171,73,21,98,55,180,7,255,212,6,54,67,237,67,68,56,22,172,110,90,218,101,144,117,83,145,91,114,106,11,168,223,127,163,27,73,97,145,95,198,93,136,183,239,153,149,69,161,208,172,233,199,53,6,163,58,255,0,103,205,239,95,252,159,186,102,114,141,80,181,97,35,149,144,57,170,114,182,107,203,196,72,146,188,56,169,212,246,188,70,117,196,144,115,78,41,82,107,97,133,105,184,160,9,5,72,40,52,67,212,244,170,245,113,52,68,161,169,255,0,35,142,107,68,89,27,91,115,197,61,35,199,53,86,229,212,159,83,82,199,82,248,68,114,14,61,107,66,253,135,248,223,155,254,30,53,127,107,79,95,239,90,185,255,199,62,245,3,246,242,56,89,86,74,171,37,124,252,129,21,255,157,27,87,153,208,248,45,43,84,168,213,153,188,69,97,77,229,35,81,2,211,169,15,160,254,243,148,208,90,31,190,159,26,180,191,165,146,6,199,126,82,245,173,192,143,127,104,197,122,31,193,162,118,245,77,117,241,43,127,57,106,97,85,95,66,153,53,93,251,217,231,247,141,209,247,205,89,149,228,170,178,87,151,92,236,133,170,58,251,78,130,120,155,53,101,43,41,26,193,147,41,165,219,69,180,55,19,111,52,109,160,176,197,20,134,3,165,59,52,198,59,125,165,13,78,243,188,28,14,181,27,77,67,151,66,36,67,254,130,58,26,95,181,131,255,171,167,15,138,149,55,103,177,203,81,115,9,36,139,142,181,85,253,125,107,170,165,88,203,99,143,149,145,253,166,26,243,237,192,113,86,162,146,148,139,142,245,160,106,113,75,115,169,6,41,54,210,52,66,109,165,251,74,218,140,110,218,80,172,127,80,162,3,254,160,62,241,166,180,170,188,47,20,255,17,191,11,210,111,219,196,191,122,79,57,72,200,154,102,107,67,17,251,218,62,172,159,94,142,127,80,25,156,26,41,136,74,85,56,52,224,178,146,102,173,35,214,91,29,52,228,78,42,96,128,214,136,249,136,191,87,185,166,241,105,59,38,88,111,141,127,84,79,63,21,46,93,132,218, [... cut ...] ,241,32,139,147,205,43,154,127,68,87,164,174,146,135,169,165,197,65,34,82,131,64,11,74,167,21,34,37,237,21,153,67,104,166,2,209,64,229,162,164,2,141,189,52,132,56,252,181,11,85,34,79,255,217,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0

ArduCAM, 2016-06-15:
@DasHammer, I think you've got the SPI timing issue. Would please let me know what the hardware platform and which ArduCAM you are using?
Would you please use the UNO R3 to test it out?

DasHammer, 2016-06-15:
I am using the arduino nano (16mhz)