ArduCAM/Arduino #54: SPI and I2C clock speed issues

rflmota, 2016-06-15:
Hello,

I've been having some troubles with putting ArduCAM Mini 5 MP (rev. B) to work with an STM32F4 microcontroller. The problem seems to be with the I2C and/or SPI clock configuration. I've tried with a combination of I2C clock of 10 KHz (since I didn't manage to put it to work with 100 KHz clock) and SPI clock of 5.625 MHz (< 8 MHz like the datasheet suggests), and I'm trying to acquire a raw BMP picture, but sometimes I can only read 0's from the FIFO.

Can you tell me if there is any exact specification for both the I2C and SPI clock speeds?

Thank you for your attention.

claymelugin, 2016-07-21:
I'm also having the same issue and can't find the clock speed setting for I2C but I do see in the data sheets that SPI is at 8MHz.

Is CS pulled low in your attempts to read/write as I don't see this addressed in the code except for SPI.

I'm wondering if CS also impacts I2C bus operation as thats where I'm struggling .

claymelugin, 2016-07-26:
In the ArduCAM Mini example code it shows   Wire.begin (921600); as the setting for the I2C clock speed but I'm still not able to get any communications that make sense, and the I2C locks up at that speed setting just reading the Device ID registers.

Lee can you help ?

ArduCAM, 2016-07-27:
@Clay, we didn't set any I2C speed in our code. You see the wrong code, it should be Serial.begin(921600) instead of Wire.begin(921600).

claymelugin, 2016-07-27:
Hi Lee,

I tried the following example from GitHub ArduCAM here is the link.  

https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_OV2640_LowPowerMode/ArduCAM_Mini_OV2640_LowPowerMode.ino 

It was far easier to understand this code as it was simpler and more specific to the Mini_OV2640.

I still cannot get a I2C link working, I’ve tried entering no speed, and other speed, I see on the analyzer the commands that go out but there is some fault and the camera does not pick up the clock line to make the reply to the master. The new analyzer equipment helps me see that the camera is not responding on I2C, so I thought you may have a clue as to why it would not respond to the Device ID read command.

My default speed on the I2C is 100 KHZ clock, and i’ve copied the commands from the example link above, and from the main line of code (Rev C)

My system is from Particle.io, not Arduino.  I’m modifying pinouts to work with my processor (STM32F05 32 bit device 3.3V device). I’m trying settings that match and I have checked I2C and SPI wiring multiple times.   But I2C interface is not that difficult so I’m not sure what to try next.  

Here is a copy of my code.

Question:  What is the range of allowable I2C speeds that can be used with the ArduCAM mini 2MP ?

If you have a though how to debug this please tell me, other wise I may have to change systems soon to make progress.

Best regards
Clay

> 
> @Clay, we didn't set any I2C speed in our code. You see the wrong code, it should be Serial.begin(921600) instead of Wire.begin(921600). 

  Wireless Strategy & Business Development for the Connected World
  3525 Del Mar Heights Road #1000
  San Diego, CA 92130
  858.945.5610  |  clay@RMACtechnologypartners.com <mailto:clay@RMACtechnologypartners.com>  |  www.RMACtechnologypartners.com <http://www.rmactechnologypartners.com/>

ArduCAM, 2016-07-27:
@Clay, You need to check on Arduino UNO R3 board first to make sure the hardware is working then move to your platform.
One of our customer has tested the 2MP camera with Photon platform using Particle.io without any problem.
I suggest you to power the camera with 3.3V instead of 5V when you connect to your processor ((STM32F05).

exp0nge, 2018-08-23:
Hi, @ArduCAM 

This question was never answered. What is the recommended clock speed for i2c?

exp0nge, 2018-08-23:
oops, 100 Khz seems to work fine :)