ArduCAM/Arduino #55: struck with connections

mmlmahesh, 2016-06-26:
i'm unable to connect my ov7670 camera module to arduino please tell me the connections details of cam and storing it in sd card

ArduCAM, 2016-06-27:
@mmlmahesh,
You can find the pin connection from https://github.com/ArduCAM/Arduino/blob/master/OV7670FIFO/OV7670FIFO.ino 
// Camera input/output pin connection to Arduino
#define WRST  25      // Output Write Pointer Reset
#define RRST  23      // Output Read Pointer Reset
#define WEN   24      // Output Write Enable
#define VSYNC 22      // Input Vertical Sync marking frame capture
#define RCLK  26      // Output FIFO buffer output clock
// set OE to low gnd

// FIFO Ram input pins
#define DO7   28     
#define DO6   29   
#define DO5   30   
#define DO4   31   
#define DO3   32   
#define DO2   33   
#define DO1   34
#define DO0   35 

// SDCARD
// MISO, MOSI, and SCK are also available in a consistent physical location on the ICSP header; 
// this is useful, for example, in designing a shield that works on the Uno and the Mega. 
// On the Arduino Mega, this is 
// 50 (MISO) 
// 51 (MOSI) 
// 52 (SCK) 
// 53 (SS)

Shelva, 2016-09-01:
Hi ArduCAM. These wiring details doesn't make sense.
What pins from the OV7670 connect to what pins on Arduino UNO? There are so many wiring diagrams on the internet, but none of them seem to work.
Would you please post a Wiring Diagram that works with this code?
Thanks
Shelva