ArduCAM/Arduino #56: taking two images and image oversize

aph3xtwin, 2016-06-30:
Hi,

I've actually two issues regarding a similar topic.
First, I'd like to take two images within a very short time frame. The camera buffer can't take 512kB so no place for two full res images. Could it be possible to shutter, transfer the buffer to the esp8266 via SPI and shutter again?

My second issue regards the camera fifo-size itself. When I take a very contrast-rich image, jpeg compression doesn't really work amd the image get's larger. The buffer overflows and no image has been taken. Is there any workaround for this, too?

thank you very much!

ArduCAM, 2016-07-03:
@goetz
Thanks for your questions. Good news to let you know that next week we
will release a new ArduCAM-5MP-Plus which has 8MByte FIFO size and you can
take continuous pictures until the 8MP is full. And can also record more
than 30seconds short video clips. I think it will cover all your
questions.

aph3xtwin, 2016-07-03:
Mr. Lee, you just made my day!!!

Thank you a lot!

aph3xtwin, 2016-07-04:
Hi Mr. Lee,
I just contacted my reseller in china to get me some of your new modules.

But still I have issues getting two images even with low resolution.
It is important for my application that I can take two images and pull up a gpio in between, like this:

```
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
#define CALLBACK_FUNCTIONS 1
#define MAX_FRAME_LENGTH 64
const int CS = 16;
ArduCAM myCAM(OV5642, CS);

void setup() {
  Serial.begin(115200);
  Serial.println("ArduCAM Start!");
  pinMode(CS, OUTPUT);
  SPI.begin();
  SPI.setFrequency(4000000); //4MHz
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
  myCAM.OV5642_set_JPEG_size(OV5642_640x480);
  myCAM.set_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);  //enable low power
  Serial.println("ArduCAM Setup END!");
}

void loop() {
  myCAM.clear_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);//disable low power
  delay(2000);
  Serial.println("Start Aufnahme");
  myCAM.clear_fifo_flag();
  myCAM.start_capture();
  Serial.println(myCAM.read_fifo_length());
  
  myCAM.start_capture();
  Serial.println(myCAM.read_fifo_length());
  delay(2000);
}
```
When I try this, the fifo-length stays zero.
Can you help me with that?

aph3xtwin, 2016-07-11:
no help with that?