ArduCAM/Arduino #58: ArduCAM ESP8266 UNO - sketch upload

Pelledabimbo, 2016-07-05:
I am working with the ArduCAM ESP8266 UNO board for Arduino Camera but I am having trouble uploading sketches. Compiling the ESP8266 example sketches (https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/blob/master/libraries/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_V1_OV5642_Capture2SD/ArduCAM_ESP8266_V1_OV5642_Capture2SD.ino) works fine but when I try to upload a sketch I get:

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

I followed the introductions to install the ArduCAM_ESP8266_UNO add-on package and it worked with no problems. I am using the micro-usb connection via com-port to upload the sketches. I here is a screenshot showing my tools settings (ESP8266_UNO). I have played around with the different settings but I still get the same error.

Do I need more power to the board or what could I be doing wrong? Anyone that has faced similar issues?

Thanks!

Per
[image]

ArduCAM, 2016-07-06:
@Pelledabimbo, from your error message, it seems that your computer didn't recognize the board, have you install the USB-Serial driver yet (CH340g) ? Can you see any COM port device on Windows Device Manager?

Pelledabimbo, 2016-07-10:
Hi,
thanks for the quick answer. As far as I can see I have the USB drivers for CH340 (I cant see a g though). Here you have my port settings:
[image]

I also seem to have issues with compiling the example sketches. On your webpage, for the ArduCAM ESP8266 UNO board, it says there are example scripts (e. g. ArduCAM_Mini_ OV5642_Capture2SD) for the "ArduCAM ESP8266 UNO board for Arduino Camera" but I cant find them in the example list. Are these the named ArduCAM_ESP8266_V2_OV5642_Capture2SD? 
When I try and upload this script I get the following:
[image]

Should I not have "ArduCAM_ESP8266_UNO" selected as board?

Best regards

Per

Pelledabimbo, 2016-07-11:
Hi again,

I seemed to have found a way to upload my sketches. I need to press the reset button on the board just before I press the upload button in the Arduino IDE. Do you know why this is? 

I am sorry for these trivial questions.

Best regards

Per

closed 2016-09-12 by Pelledabimbo