ArduCAM/Arduino #60: Corrupt Image returned from ArduCAM_ESP8266_OV5642_Capture

MH6, 2016-07-10:
I am getting corrupt images returned from the ArduCAM_ESP8266_OV5642_Capture example.

I'm also getting stack errors occasionally.

I am using an Adafruit Feather HUZZAH ESP8266 module for testing currently. 

Here is an example of the image being returned: https://i.imgur.com/ASSzKhE.jpg

And here is a snippet of the errors returned after trying to capture that image:

```
QL change to: 2
CAM Capturing
capture total_time used (in miliseconds):49
CAM Capture Done!
Size is 0.
send total_time used (in miliseconds):0
CAM send Done!
CAM Capturing

Soft WDT reset

ctx: cont 
sp: 3fff2c10 end: 3fff2f10 offset: 01b0

>>>stack>>>
3fff2dc0:  00000001 0000000d 3fff2f5c 4020b4a1  
3fff2dd0:  00000041 3fff13d8 3fff1e0c 4020a230  
3fff2de0:  0000967f 3fff2f5c 00000008 4020a280  
3fff2df0:  4020174a 000007d0 3fff2f5c 4020a310  
3fff2e00:  00000001 3fff2f5c 3fff13d8 40207312  
3fff2e10:  3fff42f0 0000001f 3fff2e50 00000001  
3fff2e20:  00000001 00000001 3fff41e8 4020c236  
3fff2e30:  0000000a 3fffdc20 3fff41e8 402089d2  
3fff2e40:  402089dc 3fff41e8 3fff2e70 40208a0e  
3fff2e50:  3fff4628 0000000f 00000008 4020d1d4  
3fff2e60:  402089dc 3fff41e8 3fff13e4 40208bc9  
3fff2e70:  3fff42f0 0000000f 00000008 40207ca3  
3fff2e80:  3fff1420 401011f3 3fff1ef0 00000001  
3fff2e90:  00000000 0000000a 4020ac38 3fff1ef0  
3fff2ea0:  4020174a 00000001 00000001 3fff1ee0  
3fff2eb0:  3fffdc20 000003e8 3fff13e4 40208cc4  
3fff2ec0:  3ffea700 00000000 000003e8 00008eae  
3fff2ed0:  3fff1404 3fff46d8 00000016 40101941  
3fff2ee0:  4020ac59 00000000 3fff1ed8 40207128  
3fff2ef0:  3fffdc20 00000000 3fff1ed8 4020ac81  
3fff2f00:  00000000 00000000 3fff1ef0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
ArduCAM Start!
Can't find OV5642 module!

Soft WDT reset

ctx: cont 
sp: 3fff2d00 end: 3fff2f10 offset: 01b0

>>>stack>>>
3fff2eb0:  3fff2f5c 3fff13d8 3fff2f5c 40206fb1  
3fff2ec0:  00000000 00000000 00000000 00000000  
3fff2ed0:  00000000 00000000 00000000 00000000  
3fff2ee0:  00000000 00000000 00000000 3fff1ee0  
3fff2ef0:  3fffdc20 00000000 3fff1ed8 4020ac76  
3fff2f00:  00000000 00000000 3fff1ef0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
ArduCAM Start!
Can't find OV5642 module!

Soft WDT reset

ctx: cont 
sp: 3fff2d00 end: 3fff2f10 offset: 01b0

>>>stack>>>
3fff2eb0:  3fff2f5c 3fff13d8 3fff2f5c 40206fb1  
3fff2ec0:  00000000 00000000 00000000 00000000  
3fff2ed0:  00000000 00000000 00000000 00000000  
3fff2ee0:  00000000 00000000 00000000 3fff1ee0  
3fff2ef0:  3fffdc20 00000000 3fff1ed8 4020ac76  
3fff2f00:  00000000 00000000 3fff1ef0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
ArduCAM Start!
Can't find OV5642 module!

Soft WDT reset

ctx: cont 
sp: 3fff2d00 end: 3fff2f10 offset: 01b0

>>>stack>>>
3fff2eb0:  3fff2f5c 3fff13d8 3fff2f5c 40206fb1  
3fff2ec0:  00000000 00000000 00000000 00000000  
3fff2ed0:  00000000 00000000 00000000 00000000  
3fff2ee0:  00000000 00000000 00000000 3fff1ee0  
3fff2ef0:  3fffdc20 00000000 3fff1ed8 4020ac76  
3fff2f00:  00000000 00000000 3fff1ef0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
ArduCAM Start!
Can't find OV5642 module!

```

Any help is appreciated!

ArduCAM, 2016-07-11:
@MH6, From your error message, it seems that the watch dog timer start working.
Programming  the ESP8266 is tricky, you have to pay attention when you write your code.
You can not use large loop in the code, which will cause WDT reset. You have to yield() function call to give time frame for the underlying tiny OS.

MH6, 2016-07-11:
Ah interesting, I didn't realize that's what WDT was. I am using the provided example code with no changes, I will try writing my own code from scratch and see if I have better luck then.

Thanks!