ArduCAM/Arduino #61: Bitrotation issue with the Adafruit Feather HUZZAH ESP8266 and ArduCam OV5642 Module

TU-KL-Crazyflie, 2016-07-11:
Hi,
I'm I using the Adafruit Feather HUZZAH ESP8266 together with the ArduCam OV5642 Module, just like in this post #60

Following the latest instructions from ArduCam 

> We have update our library for ESP8266 board, please check the latest library and examples from http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/ 

and setting the  #define OV5642_CAM_BIT_ROTATION_FIXED parameter in the memorysaver.h, I still have trouble getting the ArduCAM_ESP8266_OV5642_Capture example to work at all.

Without the bitrotation fix, I get the inevitable SPI Interface Error. Since instead of 0x55 the ESP reads back 0xAA from the camera.

However if I'm using the newest library with bitrotation, checking if the camera module type is OV5642 fails with the message "Can't find OV5642 module!"
Instead of  vid = 0x56 and pid = 0x42, the ESP reads back vid = 0xEF and pid = 0xFE. Strangely this register is read correctly, if I disable the bitrotation.

Using the bitrotation fix and manually setting the if statement to: if((vid != 0xEF) || (pid != 0xFE))  helps getting the ESP to the point, where it connects successfully to my wifi network.
Unfortunately a soft WDT Reset occurs if I the try to take a snapshot.

`OV5642 detected.

Connecting to TP-LINK_547026
.WiFi connected

192.168.178.110
Server started
QL change to: 1
CAM Capturing

Soft WDT reset

ctx: cont 
sp: 3fff29c0 end: 3fff2cc0 offset: 01b0

>>>stack>>>
3fff2b70:  00000001 0000000d 3fff1c6c 40206a4d  
3fff2b80:  00000041 3fff11e8 3fff1b84 40205ed4  
3fff2b90:  0000dd36 3fff1c6c 00000008 40205f18  
3fff2ba0:  4020153a 000007d0 3fff1c6c 40205fa8  
3fff2bb0:  00000001 3fff1c6c 3fff11e8 40202936  
3fff2bc0:  3fff1c4c 000000ff 000000ff 40204470  
3fff2bd0:  00000001 00000001 3fff33fc 4020765e  
3fff2be0:  00000000 00000102 3fff33fc 40204466  
3fff2bf0:  3fff33fc 3fff1234 3fff33fc 402044a2  
3fff2c00:  00000000 00000000 00000000 40206f44  
3fff2c10:  3fff33fc 3fff1234 3fff11f4 40204531  
3fff2c20:  3fff338c 0000000f 00000008 40203548  
3fff2c30:  3fff1234 3fff11f4 3fff11f4 00000001  
3fff2c40:  00000001 40203ad8 0000000f 402034d2  
3fff2c50:  00000000 00000000 3fff11f4 3fff1c98  
3fff2c60:  00000001 3fff1218 3fff11f4 40204797  
3fff2c70:  3ffea660 00000000 000003e8 40207654  
3fff2c80:  00000000 3fff3844 feefeffe feefeffe  
3fff2c90:  3fffdad0 00000000 3fff1c90 40202758  
3fff2ca0:  3fffdad0 00000000 3fff1c90 4020752c  
3fff2cb0:  feefeffe feefeffe 3fff1ca0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
ArduCAM Start!
temp: 55
OV5642_CHIPID_HIGH: EF
OV5642_CHIPID_LOW:FE
OV5642 detected.

Connecting to TP-LINK_547026
.WiFi connected

192.168.178.110
Server started`

I would really appreciate it if someone with an Adafruit Feather HUZZAH ESP8266 and the ArduCam OV5642 Module could post his working code.

Thx a lot! Any help is hugely appreciated.

ArduCAM, 2016-07-11:
@TU-KL-Crazyfile, from your error message it seems that the I2C is not connected correctly or the camera is dead. Please make sure if the OV5642 is identified before next step.
Or else, it will cause the WDT reset due to the blocking code when accessing the OV5642.

TU-KL-Crazyflie, 2016-07-11:
Well interestingly after commenting out or deleting the line //  myCAM.wrSensorReg16_8(0xff, 0x01); 
in the Setupprocedure, the OV5642_CHIPID_HIGH/LOW registers are successfully read, and I was able to capture the attached snapshots with 640x480 resolution.
[image]

[image]

Could you maybe explain, what thiis line of code is supposed to do?

And do you have any suggestions on how to fix the tearing problem? Setting the buffer to 2048 did not improve anything.

MH6, 2016-07-11:
So I took my setup to work to see if I had any luck there. Unfortunately I'm having more trouble than I was at home. 

I'm now having the same issue you were having with getting back 0xEF & 0xFE when reading **OV5642_CHIPID_HIGH** & **OV5642_CHIPID_LOW**. I am not using the **OV5642_CAM_BIT_ROTATION_FIXED** define, because if I do then the initial test fails and I get _SPI1 Interface Error_ messages.

Using the same exact code I was using at home results in _"Can't find OV5642 module!"_. The only differences I can think of between my two setups are possibly the Arduino verion or ESP8266 library version.

Very confusing. I think I will try testing out the ArduCam with a SparkFun Thing board. I've also ordered a ESP8266 Arduino board from ArduCam, along with a 2MP version of the ArduCam. Going to see if I have any better luck with those.

MH6, 2016-07-12:
So I received my Arducam ESP8266 Uno board w/ 2MP Mini OV2640 camera, but I'm still getting strange image tearing/corruption with the default examples provided. The only time I can get a valid image is when capturing at 160x240. Anything bigger results in image corruption.

Image @ 160x240: http://i.imgur.com/7qXFfbN.jpg
Image @ 640x480: http://i.imgur.com/1Dabj1V.jpg

Something fishy is going on in the **transferBytes** function I think, but I can't really wrap my head around it.

When I get my hands on an SD card I will try the SD card example.

ArduCAM, 2016-07-15:
@MH6, it is the now issue. The default ESP8266 library didn't allocate enough memory for the image buffer. You need to use our modified ESP8266 library for higher resolution.
https://github.com/ArduCAM/ArduCAM_ESP8266_UNO 
http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/