ArduCAM/Arduino #62: Compling for chipKit WF32 board

mstoer, 2016-07-14:
I'm trying to compile the video streaming app for my chipKit WF32 board.  It works on an Uno without problems.   With the chipKit set as the baord (under Arduino 1.6.9) I get the following errors:

```
Arduino: 1.6.9 (Windows 10), Board: "chipKIT WF32"
In file included from ... Projects\ArduCAM\Arduino Sketches\ArduCAM_Mini_OV2640_Video_Streaming\ArduCAM_Mini_OV2640_Video_Streaming.ino:23:0:

...\Documents\Arduino\libraries\ArduCAM/ArduCAM.h:338:3: error: 'regtype' does not name a type
   regtype *P_CS;
   ^

...\Documents\Arduino\libraries\ArduCAM/ArduCAM.h:339:3: error: 'regsize' does not name a type
   regsize B_CS;
   ^

exit status 255
Error compiling for board chipKIT WF32.
```

Has anyone got these to compile on chipKit boards?

ArduCAM, 2016-07-15:
@mstoer, we have simple porting for chipkit, please download from www.arducam.com/downloads/porting/Chipkit/testArducamAPI.zip
It is appreciate if you can help us improve the code for chipkit.

mstoer, 2016-07-16:
Thank you, I'll give it a try.