ArduCAM/Arduino #63: 5MP OV5642 Sensor without Arduino Shield

dotacoach, 2016-07-15:
Hello! I am new to ArduCam and Arduino. I purchased an Arduino UNO and an OV5642 Sensor which did not include the Arducam Mini Camera Shield. I am (of course) now struggling to connect the two. 

I am wondering if there is any way to buy a "mini camera shield" without the camera sensor.  I don't want the LCD Rev.C+, just looking for a simple way to grab a JPEG. Thanks for any suggestions; or even just confirming my suspicion that this is _not possible_.

ArduCAM, 2016-07-15:
@Pajama, You can't connect OV5642 camera module to Arduino UNO directly, you need additional ArduCAM shield like this one http://www.uctronics.com/arducam-f-camera-module-shield-for-arduino-uno-mega2560-due.html?___SID=U  
It is just the one you needed without the LCD.

kennsippell, 2016-07-15:
Can I do it with something like this? The "mini" shield? http://www.uctronics.com/mini-arducam-module-camera-shield-w-5-mp-ov5642-for-arduino-uno-mega2560-board.html?___SID=U

ArduCAM, 2016-07-15:
@Pajama, yes you can use this ArduCAM Mini 5MP camera which has build in OV5642 sensor. So you can't use your OV5642 module that you already have.

kennsippell, 2016-07-15:
Okay I decided to just buy a new camera + mini shield due to some size constraints. Thanks for your help!

closed 2016-07-15 by kennsippell