ArduCAM/Arduino #65: OV5642 Rev. B Arduino DUE SPI error

Patrickyp, 2016-07-16:
Hello,

I'm trying to connect OV5642 Rev. B with Arduino DUE.  
https://www.amazon.com/Arducam-Module-Megapixels-Arduino-Mega2560/dp/B013JUKZ48/ref=sr_1_fkmr0_4?ie=UTF8&qid=1468623788&sr=8-4-fkmr0&keywords=ARDUCAM+REV+B
However I'm getting a SPI interface error when trying to run some of the examples.

My connections:
Arducam             DUE
CS         ---------> 10
MOSI     ---------> MOSI(SPI HEADER)
MISO     ---------> MISO(SPI HEADER)
SCK       ---------> CLK(SPI HEADER)
GND      ---------> GND
VCC       ---------> 5V
SDA       --------->SDA(20)
SCL       --------->SCL(21)

Can someone confirm that this board is working with the DUE and whether my connections are correct?  Thanks.

ArduCAM, 2016-07-18:
@Patrickyp, there is SPI timing issue with Arduino DUE on OV5642 mini camera. So you need to uncomment the line #define OV5642_CAM_BIT_ROTATION_FIXED in the memorysaver.h file as a workaround fix.
The newly upgrade firmware in ArduCAM-Mini-5MP doesn't have this timing issue anymore.

Patrickyp, 2016-07-19:
Thanks for the reply.  I tried commenting out the line but still getting an SPI error.  Should I just buy a Rev. C?

Patrickyp, 2016-07-20:
Okay I changed the top of the file to the following but I'm still getting a spi error.

//Uncomment the following definition when you use them
//#define OV7660_CAM
//#define OV7725_CAM
//#define OV7670_CAM
//#define OV7675_CAM
#define OV2640_CAM
//#define OV3640_CAM
//#define OV5642_CAM
#define OV5642_CAM_BIT_ROTATION_FIXED
//#define MT9D111_CAM
//#define MT9M112_CAM
//#define MT9V111_CAM	
//#define OV5640_CAM
//#define MT9M001_CAM	
//#define MT9T112_CAM
//#define MT9D112_CAM

Is there anything else I can try to fix this?
Also how do I upgrade the firmware if that is an alternative solution?

ArduCAM, 2016-07-20:
@Patrickyp, did you try the #define OV5642_CAM setting? The new Rev.B 5MP camera with new firmware doesn't need bit rotation marco.
If it still can't work, please try it on Arduino UNO R3 to verify if the hardware is OK, and let me know, thanks.

Patrickyp, 2016-07-20:
Yep, I tried with only OV5642_CAM_BIT_ROTATION_FIXED uncommented(above), only OV5642_CAM uncommented and both uncommented.  Also the hardware is working because I got the camera to work on a UNO before moving to the DUE.

lena94, 2016-08-01:
Hi Patrickyp,

did the camera woke on UNO? do you have any tutorial which can help me?

Patrickyp, 2016-08-02:
@lena94 You want to first connect the arducam to their respective pins on arduino (remember sck/clk cs/ss are same thing).  Then connect the arduino to your computer and run one of the example sketches on this github.  If you managed to get an SD card connected to your arduino with the arducam then there is a sketch that saves images to that SD card.  Otherwise you will need to either write a program or use the arducam host application to read the serial port and process the image.  Let me know if you need help.

lena94, 2016-08-03:
@Patrickyp  Thank you for your reply. 
I can't get the photo in the arducam host application. 

[image]

my connections in Arduino uno:
SCL - A5
SDA - A4 
VCC - 5V
GND - GND
SCK - 13 
MISO - 12 
MOSI - 11
CS - 10 

and this is the code I am trying: 
[new.txt](https://github.com/ArduCAM/Arduino/files/399088/new.txt)

Thank you,
Lena

lena94, 2016-08-03:
@Patrickyp  Hello , 
I need your help, which download section for ArduCAM project do you use?

[image]

Thank you,
Lena

Patrickyp, 2016-08-03:
@lena94 For my Uno I used the 2 pins above AREF (next to the reset button) for SDA/SCL not sure if it's the same for yours.  Also you want to use the code on this github as those code from the download section are old.  I never used the host application but it looks like there is also a baudrate setting I would confirm that it is the same number as in the code.

khaledsoultan63, 2016-08-10:
I am having the same issue here. I managed to get it working on the nano but not the DUE.