ArduCAM/Arduino #66: Error "Over Size" when running ArduCAM_Mini_OV5642_Video_Streaming with ArduCAM_host_v1.0.exe

kennsippell, 2016-07-25:
Hello! I have an Arducam-Mini-5MP Rev.B with Arduino UNO R3. I have uploaded the example [ArduCAM_Mini_OV5642_Video_Streaming](https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_OV5642_Video_Streaming/ArduCAM_Mini_OV5642_Video_Streaming.ino) to the arduino and am running ArduCAM_host_v1.0.exe on my PC with huge successes.

I can reliably take photos at low resolution (1280x720) in all tested settings.
However, when I go to max resolution (2592x1944), I  receive situational errors "over size" and photos are corrupt with exactly 524288 in size. 

The error consistently occurs when I point the camera outside my window, and consistently does not occur when taking photos inside my house.  At 1280x720, a typical photo inside my house is 80kb and a typical photo out the window is 190kb (more than double in size).  

Am I just running up against the memory constraints of the hardware?

kennsippell, 2016-07-25:
Testing this further, this "over size" limit is being hit fairly consistently for image sizes 2592x1944, 2048x1563, and 1920x1080. The best image quality I seem to be able to get reliably outdoors is 1280x720. Any assistance is much appreciated since I seem to be limited to <1MP current.

ArduCAM, 2016-07-26:
@Pajama, There is limited frame buffer total 512KB on ArduCAM-Mini-5MP(OV5642) camera, you have to ajust the compress register to reduce the total image size to fit for the frame buffer when you use higher resolution.
We will release another ArduCAM-Mini-5MP-Plus which has 8MByte frame buffer and can accept higher quanlity full size image and other advance features to cover this issue.

kennsippell, 2016-07-26:
Do you know when that product is scheduled to release? In the mean time are there any options for larger frame buffers?

ArduCAM, 2016-07-27:
It is ready but not release for public at the time being, you can order directly from us.
See the document below:
www.arducam.com/downloads/shields/ArduCAM_Mini_5MP_Plus_OV5640_Camera_Module_DS.pdf
www.arducam.com/downloads/shields/ArduCAM_Mini_5MP_Plus_Camera_Shield_Hardware_Application_Note.pdf

kennsippell, 2016-07-27:
What is the price? Can you ship to Canada?  I am interested to order. What are my next steps?

Do you have documentation on adjusting compression settings with the OV5642 that I have already? I'd like to continue prototyping while I wait for a camera. I think I can find the appropriate registers at http://www.dragonwake.com/download/camera/ov5642/OV5642-DS.pdf but if you had some sample code for this then I'd be interested.

ArduCAM, 2016-07-27:
@Pajama, Please contact sales@arducam.com for new camera purchase. 
If you want to adjust the compression quantization scale, please check the register 0x4407, the larger value the more compression is.

kennsippell, 2016-07-27:
Why does setting a resolution of OV5642 to 2592x1944 set register 0x4407 to 0x08?https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/ArduCAM.cpp#L638

According to OV5642 documentation, the default value for 0x4407 is 0x0C. 

This means that setting the resolution to 320x240 or 2592x1944 have side effects which reduce the compression ratio. I believe this register change persists even if you change the resolution again to any of the other values. 

Can you help me understand this? Is this a bug?

ArduCAM, 2016-07-27:
@Pajama,
It is not a bug, you have to write the compression register each time you change the resolution. If you have problem with the JPEG size, you can write 0x0C or larger values to this register.

kennsippell, 2016-07-27:
Ya sorry, I realized what was going on and deleted that comment. I do appreciate your guidance though thanks!

closed 2016-08-01 by kennsippell

cultivosMushrooms, 2018-08-09:
@ArduCAM Hello, i hope that everybody are fine. I have some doubts with OV5642 module camera. I'm using a module camera of OV5642 to obtain some images and then send it through ESP32 to smartphone. I'm using a ARM microcontroller to do this. Now, i don't know how to configure the register to change the image's resolutions. For begin i want to take a picture of QVGA(320x240) to store in the SRAM of microcontroller that is 512KB and later send to ESP32.
Could you help me about what registers i need change to modify the resolutions?
Thank you so much.

UCTRONICS, 2018-08-10:
@cultivosMushrooms 
You can refer to https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/ov5642_regs.h
There are multiple resolutions configure in the file.