ArduCAM/Arduino #67: can i use arducam library for arduino and ov7670?

mavaraiehasti, 2016-07-31:
i want take picture with ov7670 and use arduino uno and with esp8266 transfer to android mobile...
i dont have arducam.

ArduCAM, 2016-08-01:
@mavaraiehasti, you can't connect the OV7670 to ESP8266 directly, because the ESP8266 does not have enough pins and much slower than the OV7670 data stream.

mavaraiehasti, 2016-08-03:
yeah, i want to use arduino uno and esp8266 and ov7670 are connect to it.@arducam

ArduCAM, 2016-08-07:
@mavaraiehasti, Please use ArduCAM-Mini-2MP(OV2640) to work with ESP8266, we have ready to use examples.
Please read the user guide from: www.arducam.com/downloads/shields/ArduCAM_Mini_2MP_ESP8266_EvaluationKit_DS.pdf

mavaraiehasti, 2016-08-07:
In my country there is not this module and i cant use it. I live in iran