ArduCAM/Arduino #69: ArduCam_host_v1.0

lena94, 2016-08-01:
Hello,

I'm trying to connect OV5642 Rev. B with Arduino auno.  
https://www.amazon.com/Arducam-Module-Megapixels-Arduino-Mega2560/dp/B013JUKZ48/ref=sr_1_fkmr0_4?ie=UTF8&qid=1468623788&sr=8-4-fkmr0&keywords=ARDUCAM+REV+B

My connections : 
CS- 10
MOSI - 11
MISO - 12
SCK -13 
GND - GND 
VCC - 5V
SDA - A4 
SCL- A5 

the code I am trying is attached. 

[ArduCAM_SPI_OV5642_FIFO_UART.txt](https://github.com/ArduCAM/Arduino/files/394045/ArduCAM_SPI_OV5642_FIFO_UART.txt)

The problem is that AuduCam application does not display any picture.
[image]

Thank you

MarkMat320768, 2016-08-02:
> image

Forgive me for making no more contribution than this but I asked a tech 
because I was at a loss and before he was taken away from me by his boss 
with a fake emergency as it was becoming to proprietary to inform me he 
told me 'yes you can just use the facility in Linux where you can make 
any device a 'mass storage device' and instead of storing the camera's 
data to a local micro sd card you can have it read all the way back 
through and to your NAS or whatever and save the stream of data' you can 
save raw censor ship data and analyze it later if you want to.

The camera is thought of by the computer the same as it thinks of a HDD 
or any mass storage it's just cast on the device and read that way. I 
haven't done it but I sure remember it as it was what I was trying to 
get closer to. A wide angle usb camera (hard to find) with auto focus..I 
don't need IR vision because motion detection turns on lamps..if the 
thief is clever anyway they can see IR light with a cell phone just 
driving by at night casing the place. They would also if they 
investigate although they would have to approach to learn that lights 
come on because of either heat or motion detection. But most people 
aren't trying to catch that guy it's the casual hoodlum that that lives 
5 doors down that hates you for some reason that vandalizes your car or 
smashes in your doorbell lamp every time you replace it.

A usb won't go far enough in some cases you have to go to Ethernet/IP 
cam (and I'm not using wireless I already don't know enough without 
mastering that also) but you can gain PoE and get out under the eve with 
no electrician to run power..and while the IP cameras they say 
'generally' are of a higher quality (because of uses..a usb cam is for 
right there at the desk) have what you do really need-- the wider angle 
lense and upgrades like auto-focus and auto-iris control..darker the 
iris opens up, etc.

But in the right circumstance multiplied by a thousand I can deploy the 
lesser expensive USB camera and save a lot of money. That camera as mass 
storage switched to Ethernet from USB just before going under the eve of 
the corner of the house for instance (can be out there remote a PoE  
switch or more than one for dividing up multiple zones of multiple 
cameras or just the one for one set of multiple cams--but those PoE 
switch boxes are not cheap) then reverse back to host computer, switch 
back to USB (I love those boxes I have one that has on/off buttons and 
one without they say you can extend the right kind out 40 times plugging 
one into the next).....isn't there a converter/switch for USB to 
Ethernet because I can get the tool and wire it myself but I rather it 
be a plug/converter seems as if there should be one.

Just look at the prices of IP cams and USB cams...look at the results on 
your monitor..I get what I need from a decent USB cam that looks the 
same as an IP cam that costs $280 when the USB cam is $32...and less if 
it's a lone raspb/ardui cam.

I'm beginning to think that there's a plus in it although more expense 
to the system that it actually must bring in the support hardware (more 
raspbi)..I'm okay if it will run in hot and cold in the attic/dry...but 
that is not what I wanted.

I wanted that flat cable camera oh so sweetly inexpensive feeding back 
and total cost (not considering cat6 or switches/converters) ..maybe $28 
for a very nice camera and so tiny too.

So, this is a wonderful area to explore. My concern mainly has been 
durability and of course the fact that you can't, as far as I know, have 
one without the other..that, if camera rasb/ardui then you're going too 
have some of that hardware out there too..not just the camera itself as 
it does not drive itself it has support..that is more money depending on 
which box/what box I'm not read up on it but installation by 
installation if you're doing it as a service goes a lot

..but it was a nice moment when I imagined a $28 camera instead of those 
IPs --and the money whores with their surveillance kits and custom 
recorder boxes (you know, a NAS) lul and I'm trying to design for 
everyman (not every stupid man).

I would be glad to offer for less..1 year free service contract..just 
replace parts if needed..I rather have a little in peace than all of it 
in hell. Just people's homes and so on need protection without paying 
out the nose and that has been my target. And no a deer camera doesn't 
cut it.

Camera looks good.

I saw someone posing this in the form of--here is a list I want it to 
do..but I didn't remind at that time about the ability to just make a 
camera behave as a mass storage device and read it off..and not pay the 
safety data micro chip makers ever more and more for larger storage..as 
this doesn't fly it's installed.

MarkMat320768, 2016-08-02:
oh doh sorry lena I have so many threads..I thought this was about motion/zoneminder cam stuff ..pretty long mistake heh...hmm now how am I gonna interject to zoneminder... sorry again..ill get rid of it...um, just a sec