ArduCAM/Arduino #72: Arducam Rev C+ shield runs out of fifo

aldobranti, 2016-08-10:
I have a rev C shield attached to a mega 2560 and an ov5642 5mp. The setup executes the demonstration program Arducam_OV5642_Digital_Camera and captures a 5MP image to SD card _most of the time_. Occasionally it produces a truncated and corrupt JPEG.

My own code uses the same fifo reading loop as the demonstration (+print statements)
`i = 0;
  temp = myCAM.read_fifo();
  //Write first image data to buffer
  buf[i++] = temp;
  
  //Read JPEG data from FIFO
  while ( (temp != 0xD9) | (temp_last != 0xFF) )
  {
    temp_last = temp;
    temp = myCAM.read_fifo();
    //Write image data to buffer if not full
    if (i < 256)
      buf[i++] = temp;
    else {
      //Write 256 bytes image data to file
      Serial.print("*");
      outFile.write(buf, 256);
      i = 0;
      buf[i++] = temp;
    }
  }
  Serial.print("finished (?) temp=0x");Serial.print(temp,HEX);Serial.print(" temp_last=0x");Serial.println(temp_last,HEX);
`

I get a * for each 256b block ( between 1k and 11k bytes written to SD and the exit conditions show temp == 0xD9 & temp_last == 0xFF

Side question : what do 0xD9 and 0xFF encode

I was thinking that I needed to allow the fifo to settle down even after a check of
`myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK)`, delaying 1s does not change anything; regrettably read_fifo_length is not supported on the REV C, 

attached is a captured, but corrupt jpeg

[image]

can you suggest anything ?

MarkMMullin, 2016-08-10:
Reading the bytes blind and stopping on the first occurrence of FFD9 won't always work to find the end of the file (that's what FFD9 encodes as a segment marker)  -  [here's](http://stackoverflow.com/questions/4585527/detect-eof-for-jpg-images) a stack overflow post on it  

 this may not be your only issue, but if fifo read length is really gone (which seems a horrible thing), then you have to parse the segments coming back to detect the end marker.

aldobranti, 2016-08-10:
@MarkMMullin, thank you for the help and opening my eyes to the grossness of a LALR(1) (IIRC) parse of the down coming fifo stream.  Hopes @ArduCAM will have some information of how the sample code mostly avoids this trap...

aldobranti, 2016-08-10:
@ArduCAM , which segment markers does the Arduchip push into the fifo ? like 0xFFDA?

aldobranti, 2016-08-15:
So, I am thinking that I need some jpeg parsing code
here is a cut, though I cannot get this to work ? any jpeg wizards out there ?

`//Parse JPEG data from FIFO
  uint16_t skipCount;
  boolean collect1Skip = false, collect2Skip = false, inScanData = false;
  while ( (temp != 0xD9) | (temp_last != 0xFF) | ( skipCount != 0) ) {
    temp_last = temp;
    temp = myCAM.read_fifo();
    if ( skipCount > 0) skipCount --;
    //Write image data to buffer if not full
    if (i < 256)
      buf[i++] = temp;
    else {
      //Write 256 bytes image data to file
      //Serial.print("*");
      outFile.write(buf, 256);
      i = 0;
      buf[i++] = temp;
    }
    if ( inScanData & ( temp_last == 0xFF ) & (temp != 0x00 ) & ( temp < 0xD0 ) & ( temp > 0xD8 )) inScanData = false ;
    
    if ( ( temp_last == 0xFF ) & (skipCount == 0 ) & !inScanData) {
      if ( temp == 0xDA ) inScanData = true;
      else if (( temp == 0x00 ) | ( temp == 0x01 ) | /*( temp == 0xFF ) |*/ (( temp >= 0xD0 ) & ( temp <= 0xD9 ))) skipCount = 0;
      else collect1Skip = true;
    } else if ( collect1Skip ) {
      skipCount = temp ; collect1Skip = false ; collect2Skip = true;
    } else if ( collect2Skip ) {
      skipCount = skipCount * 256 + temp - 2 ; collect2Skip = false;
    }
  }
  Serial.print("finished (?) temp=0x"); Serial.print(temp, HEX); Serial.print(" temp_last=0x"); Serial.println(temp_last, HEX);
`

aldobranti, 2016-08-15:
@arducam, to continue where we left off on #74 
yes I do share the i2c bus with hmc5883 mag compass
here is another corrupt jpeg
[image]

NetForces, 2016-08-15:
FYI @aldobranti the hex pattern on that file is very very different than what I have.

aldobranti, 2016-08-15:
@NetForces It's not that easy to reach into the aether and pluck random bytes :)

JoeSouthern, 2016-08-15:
I'm not positive, but I think the Arducam only sends out a single FF D8 and FF D9. So I don't think that's the issue.

In some applications If there are more than one FF D8 and FF D9 combo then they are for thumbnails.

You have a single FF D8 in your file  (but not at the beginning) and a single FF D9. Your file is missing other jpg header info though.

aldobranti, 2016-08-15:
I had thought that a problem was finding a random FFD9 in the byte stream,  leading to a premature termination and hence my attempt to write a proper parser. When the arducam does create a good file (in the sample code) the file looks like this
0000000    ff  d8  ff  e0  00  10  4a  46  49  46  00  01  01  01  00  00
0000020    00  00  00  00  ff  db  00  43  00  08  06  06  07  06  05  08
thus a ffd8, ffe0 and ffdb segments for starters. So I think it's down to noise and build issues

JoeSouthern, 2016-08-16:
check my post on https://github.com/ArduCAM/Arduino/issues/75   - I fixed my issue. Perhaps this will work for you too.

ArduCAM, 2016-08-16:
@JoeSouthern, It seems that you are using the Arduino DUE board, the SD library changed the SPI interface speed, so the camera can't work at that high SCLK.
I have already written the post about DUE from http://www.arducam.com/arducam-arduino-due-board/

ArduCAM, 2016-08-16:
@aldobranti, You can use this tool to determine where the JPEG data is corrupted www.arducam.com/downloads/tools/JPEGsnoop.exe
If you come across any issue, please use proved example from the github and make sure the hardware is working before moving to modify the code.

ArduCAM, 2016-08-16:
@aldobranti and @NetForces, I agree with NetForce, the JPEG file aldobranti is not actually a JPEG file.
There is no JPEG header, and coincidental occurrence of 0xFF 0xD9 in the data stream.
I suspect the sensor is not well initialized to stream correct JPEG data, so please remove other device on I2C bus to isolate the problem.
Use this tool to analysis JPEG data: www.arducam.com/downloads/tools/JPEGsnoop.exe

aldobranti, 2016-08-16:
Thanks @ArduCAM , I am running on a Mac (10.10) so the .exe is a non-starter

aldobranti, 2016-08-16:
here is the hardware, nb nothing attached to p6
[image]
here is a bare_bones code with all the peripherals stripped out, this does not write a full length jpeg
[had to save as a pdf]
[bare_bones.pdf](https://github.com/ArduCAM/Arduino/files/419855/bare_bones.pdf)

here is a short length file 
[image]
surprisingly, these files seem to start the same way each time with an 8f a7 pair, just a thought...

and just to confirm , this physical rig executes the distributed example code , image attached 
[image]

@Arducam, pl;ease look at my bare_bones.ino -- it is full of debug code but your more practiced eye will spot something crucial

ArduCAM, 2016-08-16:
@aldobranti, I update the debug tool for Mac, please try it from www.arducam.com/downloads/app/ArduCAM_Host_V2.0_Mac.app.zip
I will look into your code when I get time, be patient.

JoeSouthern, 2016-08-16:
"Arducam - @JoeSouthern, It seems that you are using the Arduino DUE board, the SD library changed the SPI interface speed, so the camera can't work at that high SCLK."

No, I'm using the mega 2560 but also had been using the uno.

ArduCAM, 2016-08-16:
@JoeSouthern, Thanks for your reminder, would you please let me know which Arduino IDE version you are using.
I'm using 1.5.8 , 1.6.8 without any issue with UNO R3 and MEGA2560 R3 boards here.

aldobranti, 2016-08-16:
@ArduCAM . thanks I down loaded the Host tool 
I tried running it and checking if there were any instructions at Arducam.com
I don't think anything runs
[image]

from the testing folder in the distro the SPI BUS TEST code 
[image]
seems to print UCU ( I checked the Serial speed )
and the i2C test just says
[image]

ArduCAM, 2016-08-16:
@aldobranti, you have to use example sketch here to work with the host application https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_OV5642_Video_Streaming

aldobranti, 2016-08-16:
Thanks : The 2560+Shield is now free standing of any other connection, powered off the USB . I compiled  and uploaded ArduCAM_Mini_OV5642_Video_Streaming
I then start the host app but per screenshot very little actually happens 
[image]
how do I get this going ?
[image]

ArduCAM, 2016-08-17:
@aldobranti, I'm not sure if you can run 921600bps on your Mac, there should be "ArduCAM Start!" "OV5642 detected." when you open the COM port.
Would you please try to change the baudrate to 115200bps in example sketch and try it again?

aldobranti, 2016-08-17:
> BWb18ks5qglb4gaJpZM4Jg7wp> .
> ,"name":"GitHub"},"entity":{"external_key":"github/ArduCAM/Arduino","title":"A
> rduCAM/Arduino","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/assets/14341
> 8/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"https
> ://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
> b52498112777.png","action":{"name":"Open in
> GitHub","url":"https://github.com/ArduCAM/Arduino"}},"updates":{"snippets":[{"
> icon":"PERSON","message":"@ArduCAM in #72: @aldobranti, I'm not sure if you
> detected.\" when you open the COM port.\r\nWould you please try to change the
> again?\r\n\r\n"}],"action":{"name":"View
> Issue","url":"https://github.com/ArduCAM/Arduino/issues/72#issuecomment-240281

Thank you Lee, that is just the problem  it is a very old (2009 Mac mini)
I can now see the captured images at 115200.

If I plug in my i2c device and the gps, I can still recover images using the
host tool.

Now back to understanding why my code doesn¹t succeed in capturing :o

.a.

aldobranti, 2016-08-17:
By a process of stealing magic from working code I changed my setup() to include the following init for the 5642

`  //Check if the camera module type is OV5642
  myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
  if ((vid != 0x56) || (pid != 0x42)) {
    Serial.print("Can't find OV5642 module! vid=0x"); Serial.print(vid, HEX); Serial.print(" pid=0x"); Serial.println(pid, HEX);
  } else
    Serial.println("OV5642 detected");

  myCAM.set_format(JPEG);
  Serial.println("Changed to jpeg capture mode");
  myCAM.InitCAM();
  Serial.println("initialized the OV5642 module ");

  myCAM.OV5642_set_JPEG_size(OV5642_2592x1944);
  
  myCAM.set_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
  myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);`

and in particular the last 3 lines seemed to be the really potent bit of this magic

closed 2016-08-17 by aldobranti

JoeSouthern, 2016-08-17:
>>@JoeSouthern, Thanks for your reminder, would you please let me know which Arduino IDE version you are using.
I'm using 1.5.8 , 1.6.8 without any issue with UNO R3 and MEGA2560 R3 boards here.>>

@ArduCAM - I am using 1.6.9 but I used 1.65 also in my testing.
It was the only change I made was as I said -
from 
return card.init(SPI_HALF_SPEED, csPin) &&
to 
return card.init(SPI_FULL_SPEED, csPin) &&

I thought I was on the right track when SPI_QUARTER_SPEED gave me more random bytes. 

@aldobranti  should also look at this.

My files now from my initial testing are perfect. 

Basically, I would get a random extra byte(s) - and generally in the same place in the file.

This fixed it.

ArduCAM, 2016-08-17:
@aldobranti,
The OV5642's VSYNC signal is active low after the JPEG settings, so it is very important to reverse the VSYNC polarity in the ArduCAM register to make it work like the following like does.
myCAM.set_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);
And you have to clear the capture done flag using the following code before take a new image:
myCAM.clear_fifo_flag();
The last line of code is less important, you can use it as is or just ignore of it.
myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);