ArduCAM/Arduino #73: Arducam Mini w/ 2 MP OV2640 on Arduino Uno: Can't find OV2640 module

gohai, 2016-08-11:
I am trying to get an "ArduCAM Mini 2MP Rev. B" to work with an Arduino Uno running version 1.6.5 of the Software, and the latest ArduCAM library from GitHub.

Oddly I am seeing the following error with the `ArduCAM_Mini_OV2640_LowPowerMode` example

    Can't find OV2640 module!

I've verified that that both `vid` and `pid` aren't updated by the two calls to `rdSensorReg8_8`.

Anything I am doing wrong, or things I could try?

Much apprechiated!

gohai, 2016-08-11:
Here is the output from a logic analyzer, including timing information. Looks like the address (0x60 and 0x61) matches what I see on the OV2640 datasheet, but I I am receiving a NACK. Any ideas?

[image]
[image]

gohai, 2016-08-11:
I briefly tried the "ArduCAM Mini 2MP Rev. B" with a Raspberry Pi, which runs off 3.3V instead of the 5V of the Arduino Uno.

On the Pi, I am seeing the camera show up on `i2cdetect`, with 7-bit address 0x30, which seems correct.

Can you confirm that this model is perhaps not compatible with 5V levels?

ArduCAM, 2016-08-12:
Have you tried the ArduCAM_Mini_OV2640_Video_Streaming example yet? The low power mode will shutdown the camera so the I2C may not work when it is shutdown. If you use the camera with 3.3V system, please power the camera using 3.3V power supply as well.

gohai, 2016-08-12:
I have the same problem with the `ArduCAM_Mini_OV2640_Video_Streaming`. The problem occurs very early on, though, at the first I2C transmission in `setup`, so I think both examples should be equivalent at this point.

It seems to work on a 3.3V system (Raspberry Pi), but it doesn't seem to work on a 5V system (Arduino Uno). Can you confirm that Rev. B of that board work with a 5V system, such as the Arduino Uno?

At this point I think there is either something wrong with the board, or Rev. B doesn't work with 5V Arduino Uno at all. (OV2640 not pulling down SDA low enough?)

ArduCAM, 2016-08-17:
If you Arduino UNO board with camera, please power the camera with 5V, or else if you use the camera with 3.3V system like Raspberry Pi or BBB, please power the camera with 3.3V. The I2C interface on camera has internal pull up resistors, you do not need extra pull up resistors on your backend processor. Can you show me something that you said the OV2640 not pulling down SDA low enough?

gohai, 2016-08-17:
@ArduCAM 
I am already powering the camera with 5V when using the 5V Arduino. I also tried disabling the Atmega's built-in pull-up resistors with these lines, after `Wire.begin()`, but it doesn't help.

    PORTC &= ~ (1<<4);
    PORTC &= ~ (1<<5);

If you look at the logic analyzer output above, you'll see that SDA doesn't get low (enough) for the ACK to be received. I don't have a Digital Storage Oscilloscope to to say exactly.

My guess is that either the board is broken, or the design/revision isn't compatible with (5V) Arduino Uno.

ArduCAM, 2016-08-17:
@Gottfried, I'm totally understand what you are talking about. If there is any issue with hardware, please contact us for replacement.

spowndark, 2016-08-17:
hi friend i have the same error :( !!
[image]

gohai, 2016-09-12:
I've received a replacement board (same Rev. B), and unfortunately it still exhibits the problem when I try to get it to work with my 5V Arduino Uno. To be fair: I got it to work exactly *once*, when I got a `OV2640 detected` message over Serial, but all other times the I2C transaction just doesn't get acknowledged in the eye of the Atmega.

Giving up on 5V Arduino for now :(

ArduCAM, 2016-09-13:
@Gottfried, I'm sorry to hear that the replacement is not working either. But it is impossible that the ArduCAM-Mini-2MP doesn't work with UNO.
We test our all of our cameras using UNO R3 here. Can you show me your hardware setup, maybe I can help you figure out where the problem is.

gohai, 2016-09-17:
Dear Lee @ArduCAM, I just tied again, with the latest code from Git, latest Arduino version, and the `ArduCAM_Mini_OV2640_Video_Streaming`, and at least up until now everything seems to work. I'll close this bug and re-open it when I hit the issue next time. Thanks so much meanwhile.

closed 2016-09-17 by gohai

ArduCAM, 2016-09-18:
@Gottfried, Would you please let me know how did you resolve the issue, and the older camera works or not now?

gohai, 2016-09-19:
@ArduCAM It's hard for me to say - it used to be that the OV2640 module wouldn't be detected after startup more that 50%, but at the moment it seems to be working (did update to the latest Arduino software and the latest ArduCAM git). Not sure yet what made a difference or whether it is fixed for good.

Can I PayPal you the $$ for the second camera? I might not have time in the nearby future to investigate this further, or test the first camera again.

ArduCAM, 2016-09-19:
@Gottfried, I don't need your payment, I just want to make sure where the problem is at the very beginning, and make sure that other users will not come across the same issues.
Double check that if the first camera work or not right now?

gohai, 2016-09-19:
@ArduCAM I'll test Wednesday and send you a writeup! Best

gohai, 2016-09-26:
@ArduCAM Hi Lee, I did some testing and I believe I now know what happened:

I first tested the `ArduCAM_Mini_OV2640_LowPowerMode` example, because I was interested in using the module in low-power scenarios. This probably worked the first time, but starting with the second time it must have failed with `Can't find OV2640 module!`.

Why? Because the module, when idle, is in low-power mode, which persists even after resetting the Arduino Uno (or uploading a new sketch). Because the module is still in sleep at the beginning of setup(), the I2C communication via `myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid)` fails, and I never saw an `ACK` on the logic analyzer. (I sent a pull-request to fix that https://github.com/ArduCAM/Arduino/pull/91)

Then I tested `ArduCAM_Mini_OV2640_Video_Streaming`, but I must not have removed the module from power in the mean time, so - when the sketch was running the module must have still in low-power mode, and I was seeing the same error messages.

I tested with both modules. They both seem to work. Must have been this software issue!

ArduCAM, 2016-09-29:
@Gottfried
Thanks for finding our bugs in the example sketch. You are right, once the camera go into low-power state, it will not come back if you accidently reset the Arduino.
I have update the example according to your suggestions.

Nongsai, 2016-10-20:
Dear Lee,

I am having a similar problem - I get the "Can't find OV2640 module" error message. However, I have not launched the LowPowerMode and have only supplied the camera and the Arduino Mega board with 5V (USB connection). The ArduCam SPI bus  is fine (checked with ARDUCHIP_TEST1). If I print the vid and pid on the serial, I get vid = 33 and pid = 78. Could this be the problem? What does this mean, and what should I do to solve this problem? Best thanks!

ArduCAM, 2016-10-25:
@Nongsal,  we have update our libraries, please download the latest and try example https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_Video_Streaming first.