ArduCAM/Arduino #74: Stuck at "Start Capture" in ArduCAM_Mini_5MP_TimeElapse2SD

NetForces, 2016-08-12:
Just got a 5MP Mini RevB and I'm trying to at least get a picture on an SD card using a Arduino UNO.

When the Arduino starts I get:

```
ArduCAM Start!
OV5642 detected
SD Card detected!
Start Capture
```

But it stays there. never reach the `Capture Done!`.

Any clue ?

NetForces, 2016-08-12:
Looks like it may have been power related. Powering the UNO, Camera and SD from the USB port may not have been enough current.

It now goes through the "Start Capture" phase using an external 5Vdc power supply.

I'm not getting anything on my SD card, but I'll close this ticket for now and open a new one if I have other trouble.

NetForces, 2016-08-12:
Ok, I'm re-opening this, it was not power, I have no clue what is wrong, none of the example work.

NetForces, 2016-08-12:
Now I can get files on my SD card, but I have to kinda ground the ArduCam heatsink by touching it with my finger. If I remove my finger it stops taking captures.

The JPEG saved to the SD are also corrupted it seems as I can not open them.

NetForces, 2016-08-14:
Ok, I triple check my wiring:

[image]

Block diagram for the connection is:

[image]

 I have what I think is the most simple sketch:

```
// ArduCAM demo (C)2015 Lee
// web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules, and can run on any Arduino platform.
//
// This demo was made for Omnivision OV5642 5MP sensor.
// It will run the ArduCAM Mini 5MP as a real 5MP digital camera, provide JPEG capture.
// The demo sketch will do the following tasks:
// 1. Set the sensor to JPEG mode.
// 2. Capture and buffer the image to FIFO every 5 seconds
// 3. Store the image to Micro SD/TF card with JPEG format in sequential.
// 4. Resolution can be changed by myCAM.OV5642_set_JPEG_size() function.
// This program requires the ArduCAM V3.4.0 (or later) library and ArduCAM Mini 5MP shield
// and use Arduino IDE 1.5.2 compiler or above

#include <SD.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include <ov5642_regs.h>
#include "memorysaver.h"

#if defined(__arm__)
#include <itoa.h>
#endif

#define SD_CS 8
// set pin 10 as the slave select for SPI:
const int SPI_CS = 10;

ArduCAM myCAM(OV5642, SPI_CS);
boolean isShowFlag = true;

void setup()
{
  uint8_t vid, pid;
  uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
#else
  Wire.begin();
#endif
  Serial.begin(115200);
  Serial.println("ArduCAM Start!");
  // set the SPI_CS as an output:
  pinMode(SPI_CS, OUTPUT);

  // initialize SPI:
  SPI.begin();
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55)
  {
    Serial.println("SPI interface Error!");
    while (1);
  }

  //Check if the camera module type is OV5642
  myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
  if ((vid != 0x56) || (pid != 0x42))
    Serial.println("Can't find OV5642 module!");
  else
    Serial.println("OV5642 detected");

 
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);    //VSYNC is active HIGH
  //myCAM.OV5642_set_JPEG_size(OV5642_320x240);
         myCAM.OV5642_set_JPEG_size(OV5642_640x480);
  //       myCAM.OV5642_set_JPEG_size(OV5642_1280x720);
  //       myCAM.OV5642_set_JPEG_size(OV5642_1920x1080);
  //       myCAM.OV5642_set_JPEG_size(OV5642_2048x1563);
  //       myCAM.OV5642_set_JPEG_size(OV5642_2592x1944);
  //Initialize SD Card
  if (!SD.begin(SD_CS))
  {
    while (1);    //If failed, stop here
    Serial.println("SD Card Error");
  }
  else
    Serial.println("SD Card detected!");
}

void loop()
{
  char str[8];
  File outFile;
  byte buf[256];
  static int i = 0;
  static int k = 0;
  static int n = 0;
  uint8_t temp, temp_last;
  uint8_t start_capture = 0;
  int total_time = 0;

  start_capture = 1;
  delay(5000);

  if (start_capture)
  {
    //Flush the FIFO
    myCAM.flush_fifo();
    //Clear the capture done flag
    myCAM.clear_fifo_flag();
    //Start capture
    myCAM.start_capture();
    Serial.println("Start Capture");
  }

  while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));

  Serial.println("Capture Done!");
  //Construct a file name
  k = k + 1;
  itoa(k, str, 10);
  strcat(str, ".jpg");
  //Open the new file
  outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
  if (! outFile)
  {
    Serial.println("open file failed");
    return;
  }
  total_time = millis();

  i = 0;
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();
  temp = SPI.transfer(0x00);

  //Read JPEG data from FIFO
  while ( (temp != 0xD9) | (temp_last != 0xFF) )
  {
    temp_last = temp;
    temp = SPI.transfer(0x00);;
    //Write image data to buffer if not full
    if (i < 256)
      buf[i++] = temp;
    else
    {
      //Write 256 bytes image data to file
      myCAM.CS_HIGH();
      outFile.write(buf, 256);
      i = 0;
      buf[i++] = temp;
      myCAM.CS_LOW();
      myCAM.set_fifo_burst();
    }
  }
  //Write the remain bytes in the buffer
  if (i > 0)
  {
    myCAM.CS_HIGH();
    outFile.write(buf, i);
  }
  //Close the file
  outFile.close();
  total_time = millis() - total_time;
  Serial.print("Total time used:");
  Serial.print(total_time, DEC);
  Serial.println(" millisecond");
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Clear the start capture flag
  start_capture = 0;

}
```

The sketch runs and does same file to the SD card but only if I put my finger on the camera module heatsink... If I don't it stay stuck at "Start Capture":

```
ArduCAM Start!
OV5642 detected
SD Card detected!
Start Capture
Capture Done!
Total time used:224 millisecond
Start Capture
Capture Done!
Total time used:149 millisecond
Start Capture
Capture Done!
Total time used:155 millisecond
Start Capture
Capture Done!
Total time used:344 millisecond
Start Capture
Capture Done!
Total time used:324 millisecond
Start Capture
Capture Done!
Total time used:228 millisecond
```

But the files on the SDcard and not recongnize as JPEG and if I look at the hex dump of the file it makes no sense:

[image]

aldobranti, 2016-08-15:
I too find problems with my Arducam unless I can "earth" the kit with with a finger tip.

NetForces, 2016-08-15:
@aldobranti If you "earth" the heatsink, do you get readable JPEG ?

@ArduCAM anything for us to try ?

aldobranti, 2016-08-15:
No, by earthing the kit - it just gets that much further through the startup sequence. Arbitrary routines like set jpeg size or init cam just hang. On an issue I raised (#72) I just (today) posted some code to try and avoid a premature end to the jpeg file but it looks as if the whole structure in the fifo is badly out and the parsing code doesnt stand a chance

NetForces, 2016-08-15:
@aldobranti looks exactly the same as me.

aldobranti, 2016-08-15:
which makes me think : the real annoying thing for me is that @arducam's demonstration code seems to run correctly and generates a jpeg ( most of the time) -- my arducam is by now wrapped up nicely in its mesh faraday cage ( to stop it messing with a GPS ). So I have to press the shutter button on the shield _through_ the mesh, leaning on the mesh which of course makes like an earthing action. Stresses that no DC electrical action passes to the shutter button. So maybe his code is not so tight and perfect and it's just when I try to package up the arducam with all the other trickery that I'm running into these EMI / RFI problems

ArduCAM, 2016-08-15:
@aldobranti and @NetForces
The ArduCAM shield doesn't need any earthing, but make sure your SPI bus should be handle carefully with shielding or grounding.
Because with long cables which will cause glitches on SPI and corrupt the JPEG image data via transfer from the shield to Arduino board.
So it is highly recommend connect the shield to Arduino directly or with short and better jumping wires.

aldobranti, 2016-08-15:
@Arducam, the revC shield straps directly onto the mega 2560, so I cannot make it any shorter ;) I do not program the SPI , Do you have any beef with running this on a 2560 ?

ArduCAM, 2016-08-15:
@aldobranti, Yes I have tried the Rev.C shield with Meage2560 R3 board here. Can you show me the photo of your setup, is there any other device shareing the I2C or SPI bus?
What kind of example sketch you are using? Did you save any files that has corrupt JPEG images? I can help you analysis the image tomorrow.

aldobranti, 2016-08-15:
I put one up on #72 -- not wishing to hijack this thread #74 ( total apologies @NetForces )

NetForces, 2016-08-15:
@ArduCAM As you can see in the picture my module is connected directly on my Uno. My power and GND to the modules as well as my SD card wires could be shorter. I'll make something with a proto shield and they will be really short to see if it has an impact.

Attached is one of my bad JPEG file.

[image]

@aldobranti no probem, but @ArduCAM please check my issue as well ;-)

JoeSouthern, 2016-08-15:
I'm having related but not exactly the same issues. I'm further than you are but have my own issues that you too may run into later. I'm making  a post, but ran across yours.

A few things that I did when I was having initial problems.

One is you aren't getting the FF D8 at the beginning nor the FF D9 at the end - so clearly your files won't work. (See below - I did find you do have the FF D9).

1.  In your example - I don't see a   pinMode(cameraChipSelect, OUTPUT);  I have that in mine.

2.  Also, I changed these  buf[i++] = temp;  to  buf[i] = temp;   i++;  I honestly think that was ok, but I was testing relating to my issue. I tested this in C# and it works, but .. anyway..i just haven't written it like that before.

3.  Your use of  myCAM1.CS_HIGH(); and  myCAM1.CS_LOW(); prior to your writes - I 'think' are unnecessary. I tried using them, but it didn't seem to make any difference. In my struggles, they actually didn't help in some of my tests.

4.  I found there were two grounds on my sd card and I grounded them both. That was an issue. You don't seem to have that on yours.

5. In using an external 5V - make sure all parts ground together. Even when you think they should already be grounded - I took no chances and then it would work. So something funky going on. I didn't see a diagram on the 5V external integration, but that tripped me up for awhile.

6. I looked at your file. If you scan it you'll see there is an FF D8 in it. And the file actually ends with an FF D9. So, that part is right. I removed the part before the FF D8 but you don't have the rest of the headers in there so some issue still ... but the things above may help you.

I'm going to do my own post, but basically I get files being written to the SD card just fine. If it should be bout 20K I get 20K. The start with FF D8 and end with FF D9 and most of the header stuff seems fine. I can capture the camera via a PC Serial capture and the files are fine. My issue is that the files on my SD card have something happening in the middle, but something very very subtle as the rest of the file looks ok.

Anyway, I hope my setup stuff helps you although as I said, I'm having my own issues. You can look for my post if you want more detail on mine.

aldobranti, 2016-08-15:
@JoeSouthern thanks I nearly always get ffD9 at the end as the loop is set to terminate that way. 

 ArduCAM_Mini_5MP_TimeElapse2SD (#74)

https://github.com/ArduCAM/Arduino/issues/74#issuecomment-239858236

NetForces, 2016-08-17:
@JoeSouthern Thanks for the share I'll try what you suggested.

NetForces, 2016-08-17:
@ArduCAM I just tried the OSX capture application with the proper sketch, but it also locks at "CAM start single shoot."

[image]

Any hints ?

NetForces, 2016-08-19:
@ArduCAM is this project still maintained/supported ? I find that waiting multiple days for any feedback discouraging. Anyway to get better direct support ?

ArduCAM, 2016-08-20:
@NetForce, Sorry for my late response. the camera locks at "CAM start single shoot." means there is no proper VSYNC signal from the sensor.
It may not well initialized before you issue a capture command.
Would you please let me know what kind of example and arduino board you are using?

NetForces, 2016-08-20:
As explained above I tried pretty much all OV5642 examples, mainly the ones to save to SD and the one to capture from the computer. I'm using a genuine Arduino Uno R3. I even tried 2 boards fresh out of their boxes.

ArduCAM, 2016-08-20:
@NetForce, As I mentioned in other threads, the host application only work with https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_OV5642_Video_Streaming example.
The SD card example can't work with host application, it is stand alone example.

NetForces, 2016-08-20:
I copied that exact link code in a new Arduino sketch and it still does not work. Does the same and hangs at "CAM start single shoot."

Note that I do have to add the line:

```
#include <ov5642_regs.h>
```

Or else the Arduino compiler does not find the equates.

Also I'm using Arduino IDE 1.6.9 if that makes any difference.

NetForces, 2016-08-20:
Latest attempt:

[image]

ArduCAM, 2016-08-20:
@NetForces, The this no completion of the capture process means the hardware can not detect a complete image frame from the sensor.
Have you ever tried in Windows with windows host app V1.0? If problem still exsit please contact for replacement.

ArduCAM, 2016-08-20:
This might be the probem, you need to modify the memorysaver.h manually in the library to uncomment the line #define OV5642_CAM instead of add the header file yourself.

NetForces, 2016-08-20:
@ArduCAM It works ! 

Why didn't I see that you needed to edit memorysaver.h anywhere in the documentation ?

ArduCAM, 2016-08-20:
@NetForce, Congratulations! I think I need to update the user guide soon.

closed 2016-08-22 by NetForces

diegoto, 2017-05-04:
buenas, NetForces, tengo el mismo problema que tu, lo solucionastes?

diegoto, 2017-05-04:
Good, I need to take photos with my arducam ov 5642 and save the photos in a sd, can someone tell me the steps, connection and code? I would be very grateful. Thank you

NetForces, 2017-05-04:
@diegoto you need to change the file `memorysaver.h` of the library and uncomment the `#define OV5642_CAM` line

diegoto, 2017-05-04:
gracias NetForces, voy a probar y te digo!!!

diegoto, 2017-05-04:
no he podido borrando lo que me has dicho, si realizo el conexionado de tu foto, deberia de funcionarme NetForces?

diegoto, 2017-05-04:
podrias mandarme tu direccion de email, y hablamos porai mejor, sino te importa claro esta!!

diegoto, 2017-05-04:
EN ESTA LINEA ME DA EL PRIMER ERROR

while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));

ESTOS SON LOS ERRORES DE LA VENTANA INFERIOR DEL PROGRAMA ARDUINO

C:\Users\Diego-PC\Desktop\myCAM\myCAM.ino: In function 'void loop()':

myCAM:116: error: 'class ArduCAM' has no member named 'get_bit'

   while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));

                 ^

myCAM:133: error: 'class ArduCAM' has no member named 'CS_LOW'

   myCAM.CS_LOW();

         ^

myCAM:134: error: 'class ArduCAM' has no member named 'set_fifo_burst'

   myCAM.set_fifo_burst();

         ^

myCAM:148: error: 'class ArduCAM' has no member named 'CS_HIGH'

       myCAM.CS_HIGH();

             ^

myCAM:152: error: 'class ArduCAM' has no member named 'CS_LOW'

       myCAM.CS_LOW();

             ^

myCAM:153: error: 'class ArduCAM' has no member named 'set_fifo_burst'

       myCAM.set_fifo_burst();

             ^

myCAM:159: error: 'class ArduCAM' has no member named 'CS_HIGH'

     myCAM.CS_HIGH();

           ^

exit status 1
'class ArduCAM' has no member named 'get_bit'

diegoto, 2017-05-04:
ESTOS SON LOS ERRORES QUE ME APARECEN, SABRIAS DECIRME A QUE SE DEBEN?

C:\Users\Diego-PC\Desktop\myCAM\myCAM.ino: In function 'void loop()':

myCAM:116: error: 'class ArduCAM' has no member named 'get_bit'

   while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));

                 ^

myCAM:133: error: 'class ArduCAM' has no member named 'CS_LOW'

   myCAM.CS_LOW();

         ^

myCAM:134: error: 'class ArduCAM' has no member named 'set_fifo_burst'

   myCAM.set_fifo_burst();

         ^

myCAM:148: error: 'class ArduCAM' has no member named 'CS_HIGH'

       myCAM.CS_HIGH();

             ^

myCAM:152: error: 'class ArduCAM' has no member named 'CS_LOW'

       myCAM.CS_LOW();

             ^

myCAM:153: error: 'class ArduCAM' has no member named 'set_fifo_burst'

       myCAM.set_fifo_burst();

             ^

myCAM:159: error: 'class ArduCAM' has no member named 'CS_HIGH'

     myCAM.CS_HIGH();

           ^

exit status 1
'class ArduCAM' has no member named 'get_bit'

NetForces, 2017-05-04:
@diegoto you will have to write in english cause I don't speak/read spanish.

diegoto, 2017-05-04:
NO PROBLEM
 I TRANSLATE IT

NetForces, 2017-05-04:
Also all capital letters is considered screaming which will get you in trouble.

diegoto, 2017-05-04:
ok, sorry

diegoto, 2017-05-04:
Previously, I commented in Spanish, that I did what you told me about the library, and my problem starts in the next line of code, with the following errors, you know to tell me that these errors are due?

You managed to take photos and save them in the SD?
-

diegoto, 2017-05-04:
This is the line of your code from which, gives me the error, followed by the following errors that I will send below, which appear in the lower window of the arduino program

while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));

NetForces, 2017-05-04:
Been a while but yes. Used the code and wiring I posted on Aug 14th above in this ticket.
Are you sure your using the latest version of the ArduCam library ?

diegoto, 2017-05-04:
These are the errors that appear to me

C:\Users\Diego-PC\Desktop\myCAM\myCAM.ino: In function 'void loop()':

myCAM:116: error: 'class ArduCAM' has no member named 'get_bit'

   while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));

                 ^

myCAM:133: error: 'class ArduCAM' has no member named 'CS_LOW'

   myCAM.CS_LOW();

         ^

myCAM:134: error: 'class ArduCAM' has no member named 'set_fifo_burst'

   myCAM.set_fifo_burst();

         ^

myCAM:148: error: 'class ArduCAM' has no member named 'CS_HIGH'

       myCAM.CS_HIGH();

             ^

myCAM:152: error: 'class ArduCAM' has no member named 'CS_LOW'

       myCAM.CS_LOW();

             ^

myCAM:153: error: 'class ArduCAM' has no member named 'set_fifo_burst'

       myCAM.set_fifo_burst();

             ^

myCAM:159: error: 'class ArduCAM' has no member named 'CS_HIGH'

     myCAM.CS_HIGH();

           ^

exit status 1
'class ArduCAM' has no member named 'get_bit'

diegoto, 2017-05-04:
I am honestly not sure of using the latest version, where could you download it to me?

diegoto, 2017-05-04:
is the v3.4.0?

NetForces, 2017-05-04:
I'm using v4.0.2 and the latest version is v4.1.0 (You can see the version in the `ArduCAM.h` file.

diegoto, 2017-05-04:
ok, thank you

diegoto, 2017-05-04:
It seems that the other problems were taken away with what you told me netforces, but now this other appears to me, you know to tell me what is due?
Thanks in advance

C:\Users\Diego-PC\Documents\Arduino\libraries\ArduCAM/memorysaver.h:1:3: error: invalid preprocessing directive #Ifndef

 # Ifndef _MEMORYSAVER_

   ^

C:\Users\Diego-PC\Documents\Arduino\libraries\ArduCAM/memorysaver.h:2:3: error: invalid preprocessing directive #Definir

 # Definir  _MEMORYSAVER_

   ^

In file included from C:\Users\Diego-PC\Desktop\myCAM\myCAM.ino:22:0:

C:\Users\Diego-PC\Documents\Arduino\libraries\ArduCAM/memorysaver.h:25:3: error: invalid preprocessing directive #Si

 # Si (definida (ARDUCAM_SHIELD_REVC) || definido (ARDUCAM_SHIELD_V2))

   ^

C:\Users\Diego-PC\Documents\Arduino\libraries\ArduCAM/memorysaver.h:42:3: error: #endif without #if

 # endif 

   ^

C:\Users\Diego-PC\Documents\Arduino\libraries\ArduCAM/memorysaver.h:44:3: error: invalid preprocessing directive #Endif

 # Endif  // _MEMORYSAVER_

   ^

exit status 1
Error compilación en tarjeta Arduino/Genuino Uno.

diegoto, 2017-05-04:
I do not know because I get some lyrics like that, sorry

NetForces, 2017-05-04:
Use the "Insert code" button in the editor when your pasting stuff like code or compiler output or else it's really hard to read.

NetForces, 2017-05-04:
Copy or attach the content of your `memorysaver.h` file here.

diegoto, 2017-05-04:
# Ifndef _MEMORYSAVER_
# Definir  _MEMORYSAVER_

// Sólo cuando se utiliza frambuesa, habilitarla
// RASPBERRY_PI #define

// Hay dos pasos es necesario modificar en este archivo antes de la compilación normales
// Sólo plataforma de la serie ArduCAM Escudo necesita seleccionar módulo de la cámara, ArduCAM Mini-plataforma de la serie no lo hace

// Paso 1: seleccionar la plataforma de hardware, sólo uno a la vez

//# Definir  OV2640_MINI_2MP
// #define OV3640_MINI_3MP
#define OV5642_MINI_5MP
// #define OV5642_MINI_5MP_BIT_ROTATION_FIXED
// OV5642_MINI_5MP_PLUS #define
// OV5640_MINI_5MP_PLUS #define

// #define ARDUCAM_SHIELD_REVC	
// ARDUCAM_SHIELD_V2 #define

// Paso 2: Seleccione uno de los módulo de la cámara, sólo uno a la vez
# Si (definida (ARDUCAM_SHIELD_REVC) || definido (ARDUCAM_SHIELD_V2))
	// OV7660_CAM #define
	// OV7725_CAM #define
	// OV7670_CAM #define
	// OV7675_CAM #define
  // OV2640_CAM #define
// 	OV3640_CAM #define
	 OV5642_CAM #define
	// OV5640_CAM #define
	
	// MT9D111A_CAM #define
	// MT9D111B_CAM #define
	// MT9M112_CAM #define
	// MT9V111_CAM #define	
	// MT9M001_CAM #define	
	// MT9T112_CAM #define
	// MT9D112_CAM #define
# endif 

# Endif 	// _MEMORYSAVER_

NetForces, 2017-05-04:
Did you run the code in a translator ?

`Definir` is not something valid and all pre-processor keywords in C should start with `#`

NetForces, 2017-05-04:
@ArduCAM Please use tags when you make a release !

diegoto, 2017-05-04:
Okay, I'm going to modify it and tell me, if it's okay with you

diegoto, 2017-05-04:
# Ifndef _MEMORYSAVER_
# Define  _MEMORYSAVER_

// Sólo cuando se utiliza frambuesa, habilitarla
// RASPBERRY_PI #define

// Hay dos pasos es necesario modificar en este archivo antes de la compilación normales
// Sólo plataforma de la serie ArduCAM Escudo necesita seleccionar módulo de la cámara, ArduCAM Mini-plataforma de la serie no lo hace//

// Paso 1: seleccionar la plataforma de hardware, sólo uno a la vez

//# Define  OV2640_MINI_2MP
// #define OV3640_MINI_3MP
#define OV5642_MINI_5MP
// #define OV5642_MINI_5MP_BIT_ROTATION_FIXED
// #defineOV5642_MINI_5MP_PLUS #define
// #defineOV5640_MINI_5MP_PLUS #define

// #define ARDUCAM_SHIELD_REVC	
// #define ARDUCAM_SHIELD_V2 

// Paso 2: Seleccione uno de los módulo de la cámara, sólo uno a la vez

# Si (definida (ARDUCAM_SHIELD_REVC) || definido (ARDUCAM_SHIELD_V2))

//#define OV7660_CAM 
//#define OV7725_CAM 

// #define OV7670_CAM
// #define OV7675_CAM
// #define OV2640_CAM #define
// #define OV3640_CAM 
#define OV5642_CAM 
// #define OV5640_CAM 	
// #define MT9D111A_CAM 
// #define MT9D111B_CAM 
// #define MT9M112_CAM 
// #define MT9V111_CAM 	
// #define MT9M001_CAM 
// #define MT9T112_CAM 
// #define MT9D112_CAM 
# endif 
# Endif 	// _MEMORYSAVER_

diegoto, 2017-05-04:
now?

NetForces, 2017-05-04:
No... It should be like:

```
#ifndef _MEMORYSAVER_
#define _MEMORYSAVER_

//Only when using raspberry,enable it
//#define RASPBERRY_PI

//There are two steps you need to modify in this file before normal compilation
//Only ArduCAM Shield series platform need to select camera module, ArduCAM-Mini series platform doesn't

//Step 1: select the hardware platform, only one at a time
//#define OV2640_MINI_2MP
//#define OV3640_MINI_3MP
#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
//#define OV5640_MINI_5MP_PLUS

//#define ARDUCAM_SHIELD_REVC	
//#define ARDUCAM_SHIELD_V2

//Step 2: Select one of the camera module, only one at a time
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
  //#define OV2640_CAM
//	#define OV3640_CAM
	#define OV5642_CAM
	//#define OV5640_CAM 
	
	//#define MT9D111A_CAM
	//#define MT9D111B_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM	
	//#define MT9M001_CAM	
	//#define MT9T112_CAM
	//#define MT9D112_CAM
#endif 

#endif	//_MEMORYSAVER_
```

NetForces, 2017-05-04:
Or you can just remove everything that is commented-out that is useless:

```
#ifndef _MEMORYSAVER_
#define _MEMORYSAVER_

//There are two steps you need to modify in this file before normal compilation
//Only ArduCAM Shield series platform need to select camera module, ArduCAM-Mini series platform doesn't

//Step 1: select the hardware platform, only one at a time
#define OV5642_MINI_5MP

//Step 2: Select one of the camera module, only one at a time
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
	#define OV5642_CAM
#endif 

#endif	//_MEMORYSAVER_
```

[27 more replies]